zwave-js/zwave-js #2081: [V7, BETA 0] protocolVersion is not retained after a restart

marcus-j-davies, 2021-03-15:
- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**
After the initial interview, the `protocolVersion` of the ZWaveNode is available. but after a restart, that property is `undefined`.
after re-interviewing, the `protocolVersion` property is once again available

**Device information**
All devices

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, have only paid attention to testing V7 BETA 0
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [x] Other: Node-Red (using ZWJS V7, BETA 0)

**To Reproduce**
1 - After initial interview, you should find `protocolVersion` has been provided
2 - Restart the driver, it has become `undefined`

**Logfile:**
Line 247 - is the re-interview
[AfterRestart.txt](https://github.com/zwave-js/node-zwave-js/files/6141698/AfterRestart.txt)

marcus-j-davies, 2021-03-15:
Oddley, line 246 is...
`Beginning interview - last completed stage: None`

not sure it makes a difference, but the `interviewStage `is `Complete` prior to re-interview

AlCalzone, 2021-03-15:
> Beginning interview - last completed stage: None

Re-interview resets all the information, including the interview stage.

marcus-j-davies, 2021-03-15:
Yup, thought it be something like that.

PR #2084 (2021-03-15): fix: parse protocol version correctly
fixes: #2081

And once again, TypeScript knew better - and I should have trusted it...
closed 2021-03-15 by AlCalzone