zwave-js/zwave-js #2088: ZWaveError: The config has not been loaded yet!

MartinHjelmare, 2021-03-15:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is **not** described there.
- [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

We recently updated the Home Assistant official add-on to 0.1.11 with a new zwave-js server version. A user reported a crash of zwave-js when starting the add-on. See below log.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

- Add-on changelog: https://github.com/home-assistant/addons/blob/master/zwave_js/CHANGELOG.md#0111
- Original issue: https://github.com/home-assistant/addons/issues/1922
- Home Assistant add-on log

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] config.sh: executing... 
[cont-init.d] config.sh: exited 0.
[cont-init.d] structure.sh: executing... 
[cont-init.d] structure.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:22:58] INFO: Successfully send discovery information to Home Assistant.
22:23:00.381 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
22:23:00.390 DRIVER   version 6.5.0
22:23:00.391 DRIVER   
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
ZWaveError: The config has not been loaded yet!
    at ConfigManager.lookupMeter (/usr/lib/node_modules/zwave-js/node_modules/@zwave-js/config/src/ConfigManager.ts:326:10)
    at ConfigManager.lookupMeterScale (/usr/lib/node_modules/zwave-js/node_modules/@zwave-js/config/src/ConfigManager.ts:342:22)
    at new MeterCCReport (/usr/lib/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:559:43)
    at new CommandClass (/usr/lib/node_modules/zwave-js/src/lib/commandclass/CommandClass.ts:108:13)
    at new MeterCC (/usr/lib/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:340:1)
    at Function.from (/usr/lib/node_modules/zwave-js/src/lib/commandclass/CommandClass.ts:331:15)
    at new ApplicationCommandRequest (/usr/lib/node_modules/zwave-js/src/lib/controller/ApplicationCommandRequest.ts:88:32)
    at Function.from (/usr/lib/node_modules/zwave-js/src/lib/message/Message.ts:222:15)
    at Driver.serialport_onData (/usr/lib/node_modules/zwave-js/src/lib/driver/Driver.ts:1395:18)
    at SerialAPIParser.emit (events.js:315:20)
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
ZWaveError: The config has not been loaded yet!
    at ConfigManager.lookupMeter (/usr/lib/node_modules/zwave-js/node_modules/@zwave-js/config/src/ConfigManager.ts:326:10)
    at ConfigManager.lookupMeterScale (/usr/lib/node_modules/zwave-js/node_modules/@zwave-js/config/src/ConfigManager.ts:342:22)
    at new MeterCCReport (/usr/lib/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:559:43)
    at new CommandClass (/usr/lib/node_modules/zwave-js/src/lib/commandclass/CommandClass.ts:108:13)
    at new MeterCC (/usr/lib/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:340:1)
    at Function.from (/usr/lib/node_modules/zwave-js/src/lib/commandclass/CommandClass.ts:331:15)
    at new ApplicationCommandRequest (/usr/lib/node_modules/zwave-js/src/lib/controller/ApplicationCommandRequest.ts:88:32)
    at Function.from (/usr/lib/node_modules/zwave-js/src/lib/message/Message.ts:222:15)
    at Driver.serialport_onData (/usr/lib/node_modules/zwave-js/src/lib/driver/Driver.ts:1395:18)
    at SerialAPIParser.emit (events.js:315:20)
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

**Logfile:**

marcelhey, 2021-03-16:
Same problem here

AlCalzone, 2021-03-16:
Looks like this is the result of mixed dependency versions (which was fixed in 6.6.0). HA is preparing to upgrade to 6.6.x - hopefully this fixes the problem.

AlCalzone, 2021-03-17:
solved downstream

closed 2021-03-17 by AlCalzone