zwave-js/zwave-js #2093: [bug] major delays in my zwave system

haavardNO, 2021-03-17:
### Version

**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 2.3.0
zwave-js version: 6.6.2

### Describe the bug

I am facing some major delays in my zwave system. Have been running it for many years with different sw and no issues. I have also been running with both zwave2mqtt and zwavejs2mqtt previously versions with no delay issues.
To narrow down what might be causing it, and to follow what seams like the development direction, I have moved to use WS Server. It is minimal configuration, so should not be much that can be configured wrong there. 

So my setup now is:
Rpi 4 running zwavejs2mqtt connected to Home Assistant with WS Server.

My issue is simple to explain. Sometimes (not all the time) a light change can take several seconds (5-10) to update. And often, if I turn on a light in Home Assistant, the switch will return to off before it jumps to on. Probably due to delay.
If anyone has any tips on how to proceed it wold be appreciated, since I am running out of ideas 😊. I have attached the logs that are mentioned in the documentation.  

[zwavejs2mqtt-store-3.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6151360/zwavejs2mqtt-store-3.zip)

AlCalzone, 2021-03-17:
I'm seeing multiple instances (28 to be exact) of 
```
18:58:15.866 CNTRLR   [Node 024] Timed out while waiting for a response from the node
```
from different nodes in your log. When the controller sends a GET-type command to a node, it waits 10 seconds for a response, then gives up and moves on. During this time, it won't send any more commands so it doesn't disturb the ongoing transaction (the sent out command could prevent it from receiving the response it was waiting for).

Some examples:
* Node 21: Alarm Sensor queries multiple non-existant sensors, each query times out. Not sure where they came from, but you should **re-interview that node** to clean up
* Node 24: Querying the endpoints times out (which is okay for some devices) and it tries to query a (probably) non-existant config paramater 255.
* Node 20 does not respond when we query which Alarm Sensors it offers (on both endpoints)
* Nodes 22 and 28 are queried for a config parameter 100 they don't have
* Node 37 seems not to have config parameter 61
* Node 8 does not respond when querying config parameter 255

The timeouts could mean that the devices have a bad connectivity and the responses don't reach us, but since that only happens on specific requests it looks more like there is just no response to some requests.
The config parameters should be removed from our config files if they don't exist, but we'd need to know the device IDs (manufacturer/product type/product ID) and firmware version of each.

In the end, this means you're controlling things while interviews are still going strong. This is normally not a problem unless the messages time out, like they do for you. Version 7 does reduce this effect because it no longer re-queries all values on startup. However, HA will take a bit until they can upgrade.

haavardNO, 2021-03-17:
Thank you @AlCalzone for a detailed explanation.
One thing I noticed yesterday is that after posting the logs. I disabled all logging in the zwavejs2mqtt UI. After that it seams like things were quicker. At least not the 5-10 seconds delay. Still testing on this. It could just be me looking for positive answers. Is it known that heavy logging can induce latency in the system?

node 21:
This is a fibaro FGBS001, a multisensor that has 4 one wire temp sensors connected. I have done a re-interview again now.
 
nodes 22 & 28:
I have the EU version of the one in the manual below. According to this manual there should not be a parameter 100.
Is this fault there because the JS database needs cleaning? 
http://manuals-backend.z-wave.info/make.php?lang=en&sku=ZW075-A02&cert=ZC10-14060003

nodes 8 & 24:
These are both energy meters, same type just with different amount of clamps. According to manual below, parameter 255 is related to reset of the device...
[image]
https://www.intin.no/wp-content/uploads/2017/07/aeoteczw095.pdf

AlCalzone, 2021-03-17:
Node 22/28 is probably this config file:
https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x004b:0.0
/cc @blhoward2 in case you still have parameter 100 in there.

nodes 8 & 24
That parameter are writeonly. We shouldn't be querying those. I've raised #2095 to track that.

> It could just be me looking for positive answers. Is it known that heavy logging can induce latency in the system?

I think that's confirmation bias. The interviews are probably finished so the driver isn't waiting for responses anymore.

AlCalzone, 2021-03-17:
> These parameters are writeonly. We shouldn't be querying those.

I've checked and we weren't querying those in most circumstances. This could only happen if these devices support Configuration CC V3+ (which I doubt) or there was something wrong during the initial interview.

#2096 should solve the last edge case, but I'd like to see your cache file `<homeid>.metadata.jsonl` to verify.

haavardNO, 2021-03-17:
See logs below.
Also I have added some more info regarding your comments.
[zwavejs2mqtt-store-4.zip](https://github.com/zwave-js/node-zwave-js/files/6157013/zwavejs2mqtt-store-4.zip)

Node 20 is another mulitsensor / binaro sensor with one onewire temp sensor connected.
[node_20.json.zip](https://github.com/zwave-js/node-zwave-js/files/6156990/node_20.json.zip)

Node 37 is a fibaro wall plug. According to manual, parameter 61 controls LED ring illumination colour.
[node_37.json 2.zip](https://github.com/zwave-js/node-zwave-js/files/6156997/node_37.json.2.zip)

[image]

AlCalzone, 2021-03-17:
This is our config file for the ZW095: https://github.com/zwave-js/node-zwave-js/blob/5290e4ea23ad1bc26ac41517c404bbe2194a35b6/packages/config/config/devices/0x0086/zw095.json#L262-L286

This is the metadata you have for node 8:
```json
{
      "type": "number",
      "readable": true,
      "writeable": true,
      "valueSize": 1,
      "min": 0,
      "max": 1,
      "default": 0,
      "format": 0,
      "allowManualEntry": false,
      "states": {
        "0": "Do nothing",
        "1": "Reset to Factory Defaults"
      },
      "label": "Factory Reset",
      "description": "Reset all settings to factory defaults",
      "isFromConfig": true
    }
```
That does not match. Of course I didn't ask for the values... Could it be that you have a DSB28 instead or that it is misidentified as one? This looks more like your settings and also explains why the param 255 is getting queried.
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0086/dsb28.json

AlCalzone, 2021-03-17:
Node 20: I'd need to see a log where you re-interview it to figure out why the alarm sensor query is messed up.
Node 37 seems to have a value for param 61 now - maybe it came through after all?

haavardNO, 2021-03-17:
You are correct.
I have two DSB28. One with 1 output, and one with 3.
[image]
[image]

Did re re-interview of node 37. And at least parameter 61 is visible in the UI... Seams as I am able to chagne it also...
Node 20: Same log as yesterday? The big one, or just node export?

AlCalzone, 2021-03-17:
> Node 20: Same log as yesterday? The big one, or just node export?

A zwave-js logfile, but you'll have to re-interview the node. Just restarting does not do all the communication.

> I have two DSB28

Ok then a little patience please. @blhoward2 is in the process of reworking **ALL** Aeotec files

haavardNO, 2021-03-17:
> Ok then a little patience please. @blhoward2 is in the process of reworking ALL Aeotec files

No problem just happy someone is looking at it 😊 

See log below. I have re-interviewed serveral of the devices. Including all the fibaro wall plugs and node 20.
[zwavejs2mqtt-store-5.zip](https://github.com/zwave-js/node-zwave-js/files/6158301/zwavejs2mqtt-store-5.zip)

[image]

AlCalzone, 2021-03-17:
OK, Node 20:
```
18:15:05.509 CNTRLR « [Node 020] received response for endpoint capabilities (#1):
                      generic device class:  Binary Sensor
                      specific device class: Routing Binary Sensor
                      is dynamic end point:  false
                      supported CCs:
                        · Binary Sensor
                        · Alarm Sensor
```
but no response to Alarm Sensor queries.

Node 21 supports Alarm Sensor, but incorrectly encodes the supported sensor types (uses one value per type instead of a bitmask).

We can work around both by disabling the Alarm Sensor report.
Node 20:
```
manufacturer id: 0x010f
product type:    0x0501
product id:      0x1002
```

Node 21:
```
manufacturer id: 0x010f
product type:    0x0501
product id:      0x0102
```

haavardNO, 2021-03-17:
I see.
Will that be implemented in the configuration or in the back end by you?

[image]

AlCalzone, 2021-03-17:
That's a compatibility flag which needs to go into the configuration files.

AlCalzone, 2021-03-18:
@haavardNO This turned out to be many issues in once...
* Alarm Sensor has been disabled for FGBS001
* Node 37, Param 61 came back
* Other config parameters are being worked on

Anything else I forgot?

haavardNO, 2021-03-18:
That should be all for the devices above @AlCalzone. Thanks 😊

AlCalzone, 2021-03-18:
Alright, closing here then. Let me know if there's anything else.

closed 2021-03-18 by AlCalzone

haavardNO, 2021-03-31:
> You are correct.
> I have two DSB28. One with 1 output, and one with 3.
> [image]
> [image]
> 
> Did re re-interview of node 37. And at least parameter 61 is visible in the UI... Seams as I am able to chagne it also...
> Node 20: Same log as yesterday? The big one, or just node export?

@AlCalzone or @blhoward2  has there been any changes with this device as you know? I am not able to get data from the individual clamps any more... 
I have placed these clamps on different equipment to monitor status. So in my case I dont use the total calculated values, only the individual.
[image]

AlCalzone, 2021-03-31:
Log please

haavardNO, 2021-04-04:
@AlCalzone please see below. The meter is now node 99.
[zwavejs2mqtt-store-7.zip](https://github.com/zwave-js/node-zwave-js/files/6254532/zwavejs2mqtt-store-7.zip)
[node_99.json.zip](https://github.com/zwave-js/node-zwave-js/files/6254533/node_99.json.zip)

AlCalzone, 2021-04-04:
@haavardNO check your device configuration. You have only enabled
> Automatic Report: Group 1 - Power (Whole HEM)

but not the automatic reports for the single clamps.

haavardNO, 2021-04-05:
Thanks for the check @AlCalzone. Looks as there has been a change in how this is configured. Before I would fill out a excel sheet from aeotech and set a number that defined the reporting in a group. Now there is a selection for each value.
But seams as there is still something wrong since I am missing the settings for clamp 3.
[image]

blhoward2, 2021-04-05:
It would help if you pointed out which settings you think are missing. Also clamp and group don't mean the same thing.

haavardNO, 2021-04-05:
Yes, I see that I could be clearer :)
 I am missing values for clamp 3. As you can see from the image above, I only have threshold settings for clamp 3. No settings for clamp 3 group settings.

haavardNO, 2021-04-14:
@blhoward2 any updates? Something more you need from my end?