zwave-js/zwave-js #2099: [bug] Qubino Flush Dimmer - using I2 status updates are not triggered after restart

aveseli, 2021-03-17:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 2.3.0
zwave-js version: 6.6.2

### Describe the bug

#### Context / Setup

I am using a Qubino Flush Dimmer. It has 3 different inputs for physical switches. I1 is used to control the dimmer itself. I have hooked up I2 to a second switch. In the configuration I have set up `[10-112-0-100] Enable / Disable Endpoints I2` to `[9] Sensor binary`. 

[image]

This gives me another `Basic` report `[10-32-2-currentValue] Current value` which toggles from the values 0 to 255 and vice versa whenever I press the mono-stable switch on I2. 

[image]

#### The Problem
Whenever I restart zwavejs2mqtt it doesnt report anymore. I can see in the debug log that the zwave event is coming in, but I dont see the `Basic` section in the control panel nor it isn't reported to mqtt. When I re-interview the node everything works fine after that.

### To Reproduce

Steps to reproduce the behavior:

1. restart zwavejs2mqtt
2. Click the switch connected on I2, no mqtt message nor update on control panel.
3. re-interview the node and the problem is gone.

### Expected behavior

After restarting zwavejs2mqtt it should still report the toggle on I2 without re-interviewing the node.

AlCalzone, 2021-03-17:
Found it... the endpoint has a different device class than the node itself, causing the mapping to fail.
https://github.com/zwave-js/node-zwave-js/issues/2100

AlCalzone, 2021-03-17:
Potential fix at #2101

AlCalzone, 2021-03-19:
@aveseli Piecing together the response from Qubino and your log:
You have configured the Basic CC association group 5 (under root device), which is why the Basic CC commands are being sent:
[image]

If you remove that one and add Group 7 instead, you should receive the correct Binary Sensor commands instead. You might also need to configure the switch as a bistable switch (Parameter 2) to avoid getting the `false` report after a few seconds.

aveseli, 2021-03-19:
With Group 7 I have the same issue. No matter if I configure it as a bistable switch or not.

When I toggle the switch, it takes like 3 to 4 seconds until the event is reaching the controller.

With Group 5 it is much faster.

aveseli, 2021-03-19:
In the "old" version of zwavejs2mqtt I noticed that, when ever I press the switch on I2, two messages will arrive on mqtt:
- zwave/eg_kueche_dimmer/32/2/currentValue
- zwave/eg_kueche_dimmer/48/2/Any (some seconds later)

I dont rellay know where and why the second message ist send. Probably with the bugfix the first and the second message are mapped to the same target, which is 48/2/Any and therefore I see the messages arriving twice.

AlCalzone, 2021-03-23:
The first message is a Basic Report, the 2nd message is a Binary Switch report. With the bugfix, the Basic Report is now (expectedly) interpreted as a Binary Switch report, like you said.

I don't understand why the Binary Switch reports behave as they do in your case. I suggest you ask the Qubino support about that - they seem to be pretty responsive.

AlCalzone, 2021-03-24:
@aveseli from Qubino:
> The delay in the sent notification/sensor reports are unfortunately normal and a common complaint issue from our users. According to my colleagues, it is just the way the code was written almost 5 years ago and the delay might be even bigger

Seems like you're stuck with the Basic CC if you want instant reports. With https://github.com/zwave-js/node-zwave-js/pull/2152, the Basic CC Sets are no longer mapped to other CCs. We might have to also add a compat flag to keep it from being hidden. Can you try the linked PR please?

aveseli, 2021-03-24:
I've tested the PR. Events are send to `zwave/eg_kueche_dimmer/32/2/currentValue`. It seems to work as you exptected.

When using the currently released node-zwave-js 7 without #2152 the events are sent to `zwave/eg_kueche_dimmer/48/2/Any` in roghly the same amount of time.

AlCalzone, 2021-03-24:
Does `zwave/eg_kueche_dimmer/32/2/currentValue` stay now after a restart?

aveseli, 2021-03-24:
I have a fibaro installed somewhere else. I will switch the qubino with the fibaro. I think this is the best solution for me, since I am using the I2 only in one place.

I have asked qubino if they could instruct me to do a firmware update on myself. But they unfortunately wont do it. They also don't support OTA updates. So I am stuck with that old firmware. Kinda disappointing.

@AlCalzone thanks a lot for your support so far. I really like zwave-js and zwavejs2mqtt. You guys are doing an amazing job! Best zwave implementation so far. Thanks!

aveseli, 2021-03-24:
> Does `zwave/eg_kueche_dimmer/32/2/currentValue` stay now after a restart?

Yes it does.

AlCalzone, 2021-03-24:
Nice! Then I'll update the PR so it closes this issue :)

PR #2152 (2021-03-24): fix(cc): do not map BasicCCSet to other CCs
fixes: #2143
fixes: #2099
closed 2021-03-24 by AlCalzone