zwave-js/zwave-js #2103: Missing attributes for all nodes in 7.0.0

raman325, 2021-03-18:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

`supportsBeaming`, `supportsSecurity`, and `nodeType` do not get included in my state dump for any nodes in `zwave-js-server` when using `zwave-js 7.0.0-beta1`. You can see here where these properties are set: https://github.com/zwave-js/zwave-js-server/blob/master/src/lib/state.ts#L329-L331 . I can see properties like `supportedDataRates` and `maxDataRate` as well as `zwavePlusNodeType` and `zwavePlusRoleType` which makes me think these properties stay `undefined` after being initialized here: https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/node/Node.ts#L933-L935

**Device information**

Which device(s) is/are affected (make/model)?  all nodes
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: `zwave-js 6.6.3` and below
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [x] Other: running `zwave-js-server` (on this branch until my PR gets merged: https://github.com/raman325/zwave-js-server/tree/more_schema_3)

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

I don't have the log file from `zwave-js` at the moment because I am running this branch in a Home Assistant addon (https://github.com/marcelveldt/hassio-addons-repo/tree/master/zwave-js-server) which doesn't support logging at the moment. I have included my state dump though.
[schema_1.json.txt](https://github.com/zwave-js/node-zwave-js/files/6161632/schema_1.json.txt)

AlCalzone, 2021-03-18:
Did you interview the node prior to 7.0.0? Have you tried re-interviewing it?
Some of that information wasn't available before, so it might not be in your cache.

marcus-j-davies, 2021-03-18:
Also using Beta 1,
and I have the info (after being initialized, and after an interview)

[image]

raman325, 2021-03-18:
> Did you interview the node prior to 7.0.0? Have you tried re-interviewing it?
> 
> Some of that information wasn't available before, so it might not be in your cache.

Ah, I did not try reinterviewing. Will give that a shot and reopen this if I'm still missing the data.

closed 2021-03-18 by raman325