zwave-js/zwave-js #2108: Request missing config for POPP Flow Stop 2 (POPE701479)

jmgiaever, 2021-03-18:
**Which device is missing?**
POPP Flow Stop 2 - POPE701479

**What are the IDs?**

Manufacturer ID: 0x0154  
Product type: 0x0005
Product ID: 0x0003

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
- [Manual](http://manual.zwave.eu/backend/make.php?lang=EN&sku=POPE701479&type=popp)
- [Product page](https://www.popp.eu/products/flow-stop-2/#1463610695891-082b0cba-747b)
- [Flyer](https://www.popp.eu//wp-content/uploads/2021/01/EN_Flow-Stop-2.pdf)
- [Data sheet](https://www.popp.eu//wp-content/uploads/2021/01/EN_Datenblatt_Flow-Stop-2.pdf)

marcus-j-davies, 2021-03-18:
A mismatch in label potentially, but this is listed.

https://devices.zwave-js.io/?jumpTo=0x0154:0x0005:0x0003:1.1

jmgiaever, 2021-03-18:
Ah, looks like it the [manual for Flow Stop](https://manuals.zwave.eu/backend/make.php?lang=en&sku=POPE009501&type=popp) says its ID is `0x0154.0x0003.0x0512` and the [manual for Flow Stop **2**](https://manual.zwave.eu/backend/make.php?lang=EN&sku=POPE701479&type=popp) says its ID is `0x0154.0x0005.0x0003`

So there's some mixing going on. :) I'm not sure how the configs are loaded though, but shouldnt my device (`0x0154.0x0005.0x0003`) match this config, instead of saying
```
Manufacturer:				| Product 		| Product code
Unknown manufacturer 0x0154		| 0x0005 		| Unknown product 0x0003	
```

jmgiaever, 2021-03-18:
I'm using the latest build of zwjs2mqtt (2.4.0) and zwave-js 6.6.3.

marcus-j-davies, 2021-03-18:
I'm not overly experienced, but what firmware version is your device?
I'm unsure if this is the reason, but the config states 1.1 - 1.2

jmgiaever, 2021-03-18:
Hmmm, the version tab says:

```
Library type	| Z-Wave protocol version 	| Z-Wave chip firmware versions | Z-Wave chip hardware version
6 		| 4.5 				| ["1.6"] 			| undefined
```

jmgiaever, 2021-03-18:
Did a re-interview:
```
2021-03-18 19:40:45.041 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
19:40:45.051 CNTRLR « [Node 039] received response for protocol info:
basic device class: Routing Slave
generic device class: Binary Switch
specific device class: Valve
is a listening device: true
is frequent listening: false
is a routing device: true
is a secure device: unknown
is a beaming device: true
maximum baud rate: 40000 kbps
version: 4
19:40:45.053 CNTRLR [Node 039] Interview stage completed: ProtocolInfo
19:40:45.054 CNTRLR » [Node 039] pinging the node...
2021-03-18 19:40:45.100 INFO ZWAVE: Node 39 is alive
19:40:45.100 CNTRLR [Node 039] The node is alive.
19:40:45.102 CNTRLR « [Node 039] ping successful
19:40:45.103 CNTRLR » [Node 039] querying node info...
19:40:45.161 CNTRLR « [Node 039] node info received
supported CCs:
· Z-Wave Plus Info
· Version
· Manufacturer Specific
· Device Reset Locally
· Powerlevel
· Association
· Association Group Information
· Binary Switch
· All Switch
· Basic
controlled CCs:
19:40:45.164 CNTRLR [Node 039] Interview stage completed: NodeInfo
19:40:45.170 CNTRLR [Node 039] ManufacturerSpecificCC: doing a complete interview...
19:40:45.170 CNTRLR » [Node 039] querying manufacturer information...
2021-03-18 19:40:45.255 INFO ZWAVE: Node 39: value added: 114-0-manufacturerId => 340
2021-03-18 19:40:45.257 INFO ZWAVE: Node 39: value added: 114-0-productType => 5
2021-03-18 19:40:45.258 INFO ZWAVE: Node 39: value added: 114-0-productId => 3
19:40:45.264 CNTRLR « [Node 039] received response for manufacturer information:
manufacturer: Popp & Co (0x0154)
product type: 0x05
product id: 0x03
19:40:45.266 CNTRLR [Node 039] VersionCC: doing a complete interview...
19:40:45.267 CNTRLR » [Node 039] querying node versions...
2021-03-18 19:40:45.315 INFO ZWAVE: Node 39: value added: 134-0-libraryType => 6
2021-03-18 19:40:45.317 INFO ZWAVE: Node 39: value added: 134-0-protocolVersion => 4.5
2021-03-18 19:40:45.318 INFO ZWAVE: Node 39: value added: 134-0-firmwareVersions => 1.6
19:40:45.326 CNTRLR « [Node 039] received response for node versions:
library type: Routing Slave (0x06)
protocol version: 4.5
firmware versions: 1.6
19:40:45.328 CNTRLR » [Node 039] querying CC versions...
19:40:45.329 CNTRLR » [Node 039] querying the CC version for Binary Switch...
19:40:45.379 CNTRLR [Node 039] supports CC Binary Switch (0x25) in version 1
19:40:45.380 CNTRLR » [Node 039] querying the CC version for Association...
19:40:45.478 CNTRLR [Node 039] supports CC Association (0x85) in version 2
19:40:45.479 CNTRLR » [Node 039] querying the CC version for Association Group Information...
19:40:45.527 CNTRLR [Node 039] supports CC Association Group Information (0x59) in version 1
19:40:45.529 CNTRLR » [Node 039] querying the CC version for Battery...
19:40:45.576 CNTRLR [Node 039] does NOT support CC Battery (0x80)
19:40:45.577 CNTRLR [Node 039] skipping query for Device Reset Locally (0x5a) because max implem
ented version is 1
19:40:45.578 CNTRLR » [Node 039] querying the CC version for Manufacturer Specific...
19:40:45.632 CNTRLR [Node 039] supports CC Manufacturer Specific (0x72) in version 2
19:40:45.633 CNTRLR [Node 039] skipping query for Powerlevel (0x73) because max implemented vers
ion is 0
19:40:45.633 CNTRLR [Node 039] skipping query for Security 2 (0x9f) because max implemented vers
ion is 0
19:40:45.634 CNTRLR [Node 039] skipping query for Supervision (0x6c) because max implemented ver
[... 35 lines cut ...]
19:40:46.130 CNTRLR » [Node 039] Association group #1: Querying info...
19:40:46.181 CNTRLR « [Node 039] Received info for association group #1:
info is dynamic: false
profile: General: Lifeline
19:40:46.182 CNTRLR » [Node 039] Association group #1: Querying command list...
19:40:46.269 CNTRLR [Node 039] Interview stage completed: CommandClasses
19:40:46.270 CNTRLR [Node 039] Interview stage completed: OverwriteConfig
19:40:46.271 CNTRLR » [Node 039] requesting node neighbors...
19:40:46.287 CNTRLR « [Node 039] node neighbors received: 1, 2, 3, 4, 6, 7, 10, 11, 12, 13, 14, 15
, 16, 17, 18, 19, 20, 21, 22, 23, 24, 27, 29, 30, 31, 32, 36, 37, 41, 42
19:40:46.288 CNTRLR [Node 039] Interview stage completed: Neighbors
19:40:46.288 CNTRLR [Node 039] Interview completed
2021-03-18 19:40:46.298 INFO ZWAVE: Node 39: value added 39-37-0-currentValue => false
2021-03-18 19:40:46.299 INFO ZWAVE: Node 39: value added 39-37-0-targetValue => undefined
2021-03-18 19:40:46.300 INFO ZWAVE: Node 39: value added 39-114-0-manufacturerId => 340
2021-03-18 19:40:46.300 INFO ZWAVE: Node 39: value added 39-114-0-productType => 5
2021-03-18 19:40:46.301 INFO ZWAVE: Node 39: value added 39-114-0-productId => 3
2021-03-18 19:40:46.301 INFO ZWAVE: Node 39: value added 39-134-0-libraryType => 6
2021-03-18 19:40:46.302 INFO ZWAVE: Node 39: value added 39-134-0-protocolVersion => 4.5
2021-03-18 19:40:46.302 INFO ZWAVE: Node 39: value added 39-134-0-firmwareVersions => 1.6
2021-03-18 19:40:46.303 INFO ZWAVE: Node 39: value added 39-134-0-hardwareVersion => undefined
2021-03-18 19:40:46.304 INFO ZWAVE: Node 39 ready: Unknown manufacturer 0x0154 - Unknown product 0x0003 (0x0005)
19:40:46.304 CNTRLR [Node 039] The node is ready to be used
2021-03-18 19:40:46.311 INFO ZWAVE: Node 39: interview completed, all values are updated
```

marcus-j-davies, 2021-03-18:
Ok,

The only thing that is looking at me is the firmware version support on the config (given you have V2), it makes sense.
I could be wrong of course, so it might be best for someone with more knowledge to back up my thought (or shoot it down  😄)

@blhoward2 ?

blhoward2, 2021-03-18:
This won't match if it exceeds the fw version in the file. Usually we don't supply a max like that, though. Can you try to dig out a change log or manual showing the parameters yours supports versus what we have?

Brian

jmgiaever, 2021-03-18:
Hi,

Just to clearigy: fw version = `["1.6"]` ? 

Could you give me some guidance on what you're asking for and how I can find it? (ref: «change log or manual showing»).

The only differences that I can see in the manual, is that this (newer one = v2) one do not support assoc groups except from lifeline, while the other one does. However, the config does not mention any association groups.

marcus-j-davies, 2021-03-18:
Is there any difference in parameters/metadata between your device version, and the one we have listed?

https://devices.zwave-js.io/?jumpTo=0x0154:0x0005:0x0003:1.1

jmgiaever, 2021-03-18:
From what I can see in the manual; there's no configuration. The only difference is that the 1st version support assoc.group `Valve` (up to 10 nodes)  while the 2nd version do not support it. (Strange that the newer version don't?)

But other than this it seems to be the same device. But I guess that users with the 1st version can't use the provided config, as it the ID of the device dont match the one in the config.

blhoward2, 2021-03-18:
We can fix the id and version so both are included. We just need to verify that both should be covered by the file.

jmgiaever, 2021-03-18:
> From what I can see in the manual; there's no configuration. The only difference is that the 1st version support assoc.group `Valve` (up to 10 nodes) while the 2nd version do not support it. [...]

I can only quote myself, as the only difference I can see in the manual is the assoc.groups. One have `Valve` the other don't. And the config do not have any configuration params either.

marcus-j-davies, 2021-03-18:
@blhoward2 

it does look as if V2 has no config params.
More than happy to PR the changes with a condition for the parameters (if its not on your list)

blhoward2, 2021-03-18:
Go for it

closed 2021-03-19 by AlCalzone