zwave-js/zwave-js #2109: zwave.reset_node_meters command needed

mattbruman, 2021-03-18:
**Is your feature request related to a problem? Please describe.**
On zwave 1.4 there was a service command that allowed you to reset the accumulated totals on a meter, this was used on energy meters to reset the total at the end of the billing cycle so that sensors could be created to track monthly costs. The sensors don't know when your monthly cycle is up since many companies use a rolling 30 days that usually starts on service activation like mine is the 17th so I have to reset my totals on the 17th. I don't really know of any other way to track this.

**Describe the solution you'd like**
A method to reset a total on a entity like that could be done with zwave.reset_node_meters

**Describe alternatives you've considered**
I have tried in the past to figure out another way to have it tracked and never found a more elegant solution that just resetting the total every month with a automation.

**Additional context**
None

AlCalzone, 2021-03-18:
Did you mean to post this in the HomeAssistant repo?
zwave-js has two ways to do this (assuming the device supports reset):
1. https://zwave-js.github.io/node-zwave-js/#/api/CCs/Meter?id=reset
2. https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue with the correct value ID

mattbruman, 2021-03-18:
> Did you mean to post this in the HomeAssistant repo?
> zwave-js has two ways to do this (assuming the device supports reset):
> 
> 1. https://zwave-js.github.io/node-zwave-js/#/api/CCs/Meter?id=reset
> 2. https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue with the correct value ID

> Did you mean to post this in the HomeAssistant repo?
> zwave-js has two ways to do this (assuming the device supports reset):
> 
> 1. https://zwave-js.github.io/node-zwave-js/#/api/CCs/Meter?id=reset
> 2. https://zwave-js.github.io/node-zwave-js/#/api/node?id=setvalue with the correct value ID

Ahh well apparently HA hasn't implemented this yet as there is no way to send this to the driver. Should this request go to HA core feature request then?

AlCalzone, 2021-03-19:
Yes

closed 2021-03-19 by AlCalzone