zwave-js/zwave-js #2111: ZW3104 Smoothly Dim Config Option

mmeyer1081, 2021-03-18:
I have a couple of these GE/Jasco ZW3104 dimmers and the experience is much improved with ZwaveJS. The only issue is that they instantly pop to the set dimming value instead of smoothly changing like an Inovelli dimmer switch. I read that there is an undocumented option that can change this behavior in newer firmwares, but I'm unsure how to edit my device configuration to manually add this option similarly as explained in the below article with OpenHAB.

https://community.openhab.org/t/jasco-ge-honeywell-z-wave-outlet-dimming-problem-resolution/109387

Does this link give enough information to implement this config option for the ZW3104 devices?

28166/28167/ZW3104 I believe are all the same device.

marcus-j-davies, 2021-03-18:
If you know the firmware version, that this parameter became available, it can be added, and apply to that firmware and greater.

mmeyer1081, 2021-03-18:
It appears to be any firmware version v5.21 and newer. I have one device on 5.21 and one on 5.25.

marcus-j-davies, 2021-03-18:
If you can, and feel safe to do so, edit  the config file for your device, add this option (6) under `paramInformation` (along with the rest) if it works, then it can be included

```
"paramInformation": {
  "4": {...},
  "5": {...},
  "6": {
    "$if": "firmwareVersion >= 5.21",
    "label": "Light dimming mode",
    "description": "Controls whether to fade to next light level or jump straight there",
    "valueSize": 1,
    "minValue": 0,
    "maxValue": 1,
    "defaultValue": 0,
    "readOnly": false,
    "writeOnly": false,
    "allowManualEntry": false,
    "options": [
      {
        "label": "Smooth Level Changes"
        "value": 1,
      },
      {
        "label": "Instant Level Changes"
        "value": 0,
      }
    ]
  },
  "7": {...},
}
```

mmeyer1081, 2021-03-18:
Ok, I figured out how to edit the config files inside of zwavejs2mqtt and then added your lines to both of these files, then did a re-interview, which worked great and the smooth dimming works just as I thought! Both devices, I have a 28166 and 28167.

0x0063/28166_zw3104.json
0x0063/ge_28167_zw3104.json

Thanks so much for your help! Really great.

closed 2021-03-19 by AlCalzone