zwave-js/zwave-js #2112: Add versioning to cache so cache can be refreshed when node/value changes

raman325, 2021-03-19:
**Is your feature request related to a problem? Please describe.**
In the upgrade to `zwave-js 7.0.0`, I learned that I had to clear my cache so that `zwave-js` could rebuild it with the v7 schema since we no longer reinterview every node on startup.

**Describe the solution you'd like**
It would be great if there was something like a cache version, where if you install a zwave-js version that uses a different cache version from the existing cache, the cache gets wiped and recreated. If it doesn't exist, assume version 0, and make `zwave-js 7.0.0` use version 1.

**Describe alternatives you've considered**
Manual rebuild is the only option I am aware of currently

**Additional context**
Add any other context or screenshots about the feature request here. If you want to propose an API, share your thoughts how it could be used (syntax).

AlCalzone, 2021-03-19:
I'm not sure this is necessary often enough - and as a user I'd really hate this.
Wiping the cache means you have to re-interview the entire network in order to be able to use it.

raman325, 2021-03-19:
Regarding the re-interview, if the cache is old and leads to undesired behavior, that seems like a necessary evil, no?

Regarding your point about how often this would be needed, that is fair. Hopefully we don't reuse existing properties for new uses often (e.g. moving `nodeType` to `zwavePlusNodeType` and then using `nodeType` for something else)

AlCalzone, 2021-03-19:
Let's see:
* `zwavePlusNodeType` is just the name of the node property which is internally stored as a value ID on the `Z-Wave+ CC`, so no cache change here.
* `procotolVersion` gets migrated from the old `version`
* `supportedDataRates` gets migrated from the old `maxBaudRate`. The lower data rates might be missing, but IMO that is purely informational
* `maxDataRate` is dynamic and depends on supportedDataRates
* `supportsSecurity` is almost always wrong anyways
* `isSecure` gets migrated if possible
* `nodeType` and its predecessors didn't get saved before
* `supportsBeaming` doesn't get migrated, I give you that

I don't think that anything critical and especially nothing that should affect the behavior of zwave-js. So IMO nothing that warrants a forced cache reset and the resulting major inconvenience for users.

raman325, 2021-03-19:
OK fair enough. Just an FYI, `protocolVersion` got migrated but it reflected the old `version` (in my case it was 4 which is invalid for the enum) until I wiped the cache and started fresh.

closed 2021-03-19 by raman325

AlCalzone, 2021-03-19:
That's because it was migrated before #2107 :)

PR #2116 (2021-03-19): fix: try to parse supportsBeaming from legacy cache file
closes: #2112