zwave-js/zwave-js #2114: [bug] Heal on a single node seems not to work

Madelaide, 2021-03-19:
## System Health

version | core-2021.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.99
arch | x86_64
timezone | Australia/Adelaide

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | beta
supervisor_version | supervisor-2021.03.4
docker_version | 19.03.13
disk_total | 78.0 GB
disk_used | 23.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.1), Visual Studio Code (3.1.1), Grafana (6.1.3), InfluxDB (4.0.3), Glances (0.11.1), Log Viewer (0.9.1), Terminal & SSH (9.0.2), Mosquitto broker (5.1.1), Z-Wave JS to MQTT (0.6.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 2
views | 9
mode | storage

</details>

1. select either a node with listed neighbours or a node with no listed neighbours (but in the same proximity)
2. select Heal Node
3. note minimal log message
4. wait
5. no change, extant node with neighbours listed does not change, node with no listed neighbours (new node) also no change
6. same issue for aeotec smart switch6, aeotec 6n1 sensor, fibaro dimmer 211
7. 
screenshot below

[image]

[image]

PS: I tried to include the zwave log at DEBUG level, but I have 4 logs and there is no date on them, the highest number log doesnt seem to have the heal command in it, so it's unclear if it is today's log or not (with no date showing).

But this snippet shows the node interviews okay and finds its nodes fine.

22:46:45.501 CNTRLR   [Node 006] Interview stage completed: CommandClasses
22:46:45.501 CNTRLR   [Node 006] Interview stage completed: OverwriteConfig
22:46:45.502 CNTRLR » [Node 006] requesting node neighbors...
22:46:45.508 CNTRLR « [Node 005]   node neighbors received: 6, 7, 8, 9, 10, 15, 18, 19, 20, 21, 23, 
                      27, 31, 36, 37, 41, 42, 44, 45, 49, 50, 51, 53, 55, 56, 57, 58, 59, 60, 62, 63
                      , 64, 75, 80, 82, 87, 88, 89, 93, 94, 102, 107, 109, 111, 112, 113, 117, 118, 
                      119, 124, 125, 130
22:46:45.509 CNTRLR   [Node 005] Interview stage completed: Neighbors
22:46:45.509 CNTRLR   [Node 005] Interview completed
22:46:45.520 CNTRLR « [Node 006]   node neighbors received: 5, 7, 8, 9, 10, 15, 18, 19, 21, 23, 31, 
                      36, 37, 38, 41, 42, 44, 50, 51, 53, 55, 56, 57, 59, 60, 62, 63, 64, 75, 82, 87
                      , 88, 89, 92, 93, 94, 102, 107, 109, 110, 111, 112, 113, 118, 119, 124, 126, 1
                      30, 131
22:46:45.521 CNTRLR   [Node 006] Interview stage completed: Neighbors
22:46:45.521 CNTRLR   [Node 006] Interview completed

Madelaide, 2021-03-19:
[zwavejs2mqtt-store (11) 2.zip](https://github.com/zwave-js/node-zwave-js/files/6170600/zwavejs2mqtt-store.11.2.zip)

In this example, I ran a Heal on node 114 (fibaros pick up many nodes in my house) around 21:20, and then I turned off node 23 at around 21;22 - the latter is in the zwjs log, but node 114 has nothing in the last hour, and there is no Heal in the log

Madelaide, 2021-03-19:
Oh I see you found the issue, great, well maybe the second log(s) help anyway.

Madelaide, 2021-03-19:
PS: I put this on node-zwave-js it seems; should it have been on the other repo?

AlCalzone, 2021-03-19:
I moved it here, since the problem lies here.

closed 2021-03-19 by AlCalzone

Madelaide, 2021-03-23:
UPDATE: note I installed 2.4.1 today, the Heal node still behaves the same, zero heal information in the zwave log, no nodes updated.

Not sure if this is related...

Error: Not Found
at /opt/app.js:936:15
at Layer.handle [as handle_request] (/opt/node_modules/express/lib/router/layer.js:95:5)
at trim_prefix (/opt/node_modules/express/lib/router/index.js:317:13)
at /opt/node_modules/express/lib/router/index.js:284:7
at Function.process_params (/opt/node_modules/express/lib/router/index.js:335:12)
at next (/opt/node_modules/express/lib/router/index.js:275:10)
at /opt/node_modules/express-session/index.js:502:7
at readCallback (/opt/node_modules/session-file-store/lib/session-file-helpers.js:193:20)
at /opt/node_modules/graceful-fs/graceful-fs.js:123:16
at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:63:3)

AlCalzone, 2021-03-23:
What's the zwave-js version in 2.4.1?

Madelaide, 2021-03-24:
There are many version numbers floating around the ui, hopefully this is one you requested:
Z-Wave JS to MQTT
Current version: 0.8.1 ( Changelog )

Changelog 
What’s changed

⬆️ Dependency updates

Update database to ff8e4088321dc6490efde18a31ec893939de4de7 @frenck (#92)
⬆️ Upgrades ZwaveJS2Mqtt to v2.4.1 @frenck (#91)
⬆️ Bump frenck/action-addon-linter from v2.0.0 to v2.1 @dependabot (#93)

AlCalzone, 2021-03-24:
> Driver Version: 6.6.3

This is only fixed in v7.0.0, so you'll need to wait for that update.

Madelaide, 2021-03-31:
I have installed the public update, and server version is 7.0.0.

Heal on node 124 does this and no more...

2021-04-01 08:00:23.662 INFO ZWAVE: Success zwave api call healNode { success: true, message: 'Success zwave api call', result: false }

PS: i have THREE zwave debug logs zwavejs_357/8/9  and NONE of them have a date in the logs, and none of them have any entries for this morning that I can see.

PLEASE add the date to the log!

Madelaide, 2021-03-31:
PPS: these logs are 30, 70 and 15 MB each

Madelaide, 2021-04-01:
I update HA today, the update with the yaml debugging, and now the log has a date, time looks to UTC and not local (ie the time now is 15:04)

2021-04-01T04:31:58.999Z CNTRLR   [Node 023] [~] [Multilevel Switch] currentValue: 0 => 0           [Endpoint 0]
2021-04-01T04:31:59.010Z SERIAL « [ACK]                                                                   (0x06)
2021-04-01T04:31:59.013Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-01T04:31:59.013Z SERIAL » [ACK]                                                                   (0x06)
2021-04-01T04:31:59.013Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-01T04:32:03.175Z SERIAL « 0x01070013430101a109                                                 (9 bytes)
2021-04-01T04:32:03.176Z SERIAL » [ACK]                                                                   (0x06)
2021-04-01T04:32:03.176Z DRIVER « [REQ] [SendData]
                                    callback id:     67
                                    transmit status: NoAck
2021-04-01T04:32:04.016Z SERIAL » 0x01090013170226022544b5                                            (11 bytes)
2021-04-01T04:32:04.020Z SERIAL « [ACK]                                                                   (0x06)
2021-04-01T04:32:04.026Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-04-01T04:32:04.027Z SERIAL » [ACK]                                                                   (0x06)
2021-04-01T04:32:04.027Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-04-01T04:32:04.045Z SERIAL « 0x0107001344000003ac                                                 (9 bytes)
2021-04-01T04:32:04.046Z SERIAL » [ACK]                                                                   (0x06)
2021-04-01T04:32:04.047Z DRIVER « [REQ] [SendData]
                                    callback id:     68
                                    transmit status: OK
2021-04-01T04:32:04.055Z SERIAL « 0x01090004001703260300c3                                            (11 bytes)

AlCalzone, 2021-04-01:
I suggest you delete the old logs and try again. That way the logs should be smaller too.

PR #2115 (2021-03-19): fix: allow healing single nodes
With this PR, we now treat healing a single node similar to healing the entire network. Healing a single node now remembers that a heal process is active, so the method that does the actual work does not just exit.
As a result, it won't be possible to perform a heal on another node or the network until the single-node healing process was completed or aborted.

fixes: #2114