zwave-js/zwave-js #2123: Interview Zooz ZEN25 MultiSensor crashes zwave-js.

cthierman, 2021-03-21:
**Checklist:**

- [ x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x ] I have checked the troubleshooting section and my problem is **not** described there.
- [x ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I have a running set of devices using node-red module node-red-contrib-zwave-js dev fork, version 3.0.0.  using zwave-js version 7.0.0-beta.2

Total count of devices are 4 excluding the Aeotec Zwave Stick. the nodes are 5,6,7 and 10.

All were up, running and alive. I initiated a Interview on a working node, node 7, which is a Zooz ZEN25 double plug/switch. Shortly there after the node-red application restarted. A view of the node-red log indicated the following (I have bolded the most relative message):

Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]: 20 Mar 23:33:27 - [info] [debug:unknown ClassName ]
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]: {
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   payload: {
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:     node: 7,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:     event: 'VALUE_UPDATED',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:     timestamp: 2021-03-21T05:33:27.053Z,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:     object: {
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       commandClassName: 'Version',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       commandClass: 134,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       endpoint: 0,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       property: 'applicationBuildNumber',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       newValue: 52445,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:       propertyName: 'applicationBuildNumber'
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:     }
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   },
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   _msgid: '366142ea.b5861e',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   topic: 'homie/status/plug/zwave/Test-Kitchen Zooz Plugin',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   _event: 'node:777c646f.ce812c',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   dontgraph: true,
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   instance_name: 'All Plugs',
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]:   textname: 'Test-Kitchen Zooz Plugin'
Mar 20 23:33:27 zwave2-mqtt Node-RED[13939]: }
**Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]: TypeError: ep1Caps is not iterable**
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]:     at MultiChannelCC.interview (/home/pi/.node-red/node_modules/zwave-js/src/lib/commandclass/MultiChannelCC.ts:434:9)
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]:     at interviewEndpoint (/home/pi/.node-red/node_modules/zwave-js/src/lib/node/Node.ts:1339:5)
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]:     at ZWaveNode.interviewCCs (/home/pi/.node-red/node_modules/zwave-js/src/lib/node/Node.ts:1458:19)
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]:     at ZWaveNode.interview (/home/pi/.node-red/node_modules/zwave-js/src/lib/node/Node.ts:1041:8)
Mar 20 23:33:40 zwave2-mqtt Node-RED[13939]:     at Driver.interviewNode (/home/pi/.node-red/node_modules/zwave-js/src/lib/driver/Driver.ts:912:10)
Mar 20 23:33:47 zwave2-mqtt systemd[1]: nodered.service: Main process exited, code=exited, status=1/FAILURE
Mar 20 23:33:47 zwave2-mqtt systemd[1]: nodered.service: Failed with result 'exit-code'.

I have included a zwave-js.log.txt file with some activity that was happening at the time. Hopefully it's enough to provide enough details.

**Device information**

Which device(s) is/are affected (make/model)? Zooz ZEN25 
What are the node IDs? Nodeid is 7.

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ x] Other: 
- Node-red module, node-red-contrib-zwave-js version 3.0.0 from dev track.

**To Reproduce**

Steps to reproduce the behavior:
1. Go to 'in node-red console, goto zwave-js menu and select the node in question, followed by the interview button.'
2. after short delay, node-red console will be unavailable while node-red restarts from crash.

**Additional context**

Add any other context about the problem here.

**Logfile:**
attached.
[zwave-js-crash-report.txt](https://github.com/zwave-js/node-zwave-js/files/6176920/zwave-js-crash-report.txt)

AlCalzone, 2021-03-23:
It looks like there's something missing in the log. Can you make a new one where you re-interview the device? 

And can you upload the raw logfile? Yours seems to have some encoding issues.
@marcus-j-davies is that possible?

marcus-j-davies, 2021-03-23:
Yup,

The log generated, is the native logging output to the file transport.

@cthierman,
 - Set logging to Debug.
 - Publish
 - Delete the current file `%home%/.node-red/zwave-js-log.txt`. (unless you  overwrite the default).
 - Restart Red.
 - Do the interview (obviously this will fail according to the report)
 - Attach log here as a file `%home%/.node-red/zwave-js-log.txt`. (unless you  overwrite the default).
 
 this will give us a clean log of everything,

cthierman, 2021-03-24:
Seems now when I do the interview it doesn't crash...

 But I did generate the log file with debug with the same instructions provided.

Perhaps, in trying to shrink the file, I delete important stuff. Since it was a log file for a number of hours. I will attach the full log file here. 

My apologies for that. 

Because I can't seem to recreate it now, maybe this may be more intermittent than I first thought. 

I'll let you decide what best to do... but  Here is the full log.. Let me know if it's still corrupted. It's long. a good 6 hours of logging.
[saved-crash-log.txt](https://github.com/zwave-js/node-zwave-js/files/6194672/saved-crash-log.txt)

AlCalzone, 2021-03-24:
There's something happening in the logfile that looks like an interview, but a part of the messages (the INFO level commentary) are still missing.

@marcus-j-davies looks like this is happening on your end. The other logs I'm getting are complete 🤷🏻‍♂️
Any idea?

marcus-j-davies, 2021-03-24:
As this is not a fresh log (still from 21/03),  may not have been set to `Debug` (5) maybe?
this is my `Debug` (5) log during an interview

[zwave-js-log.txt](https://github.com/zwave-js/node-zwave-js/files/6195432/zwave-js-log.txt)

AlCalzone, 2021-03-24:
Yeah, that one looks normal 🤷🏻‍♂️

marcus-j-davies, 2021-03-24:
@cthierman ,

the log level must not have been set to Debug during the original capture. No other reason I can think of.

AlCalzone, 2021-03-24:
Anyways, I think I know what caused this.

PR #2141 (2021-03-24): fix: abort interview attempt when endpoint query times out
fixes: #2123
closed 2021-03-24 by AlCalzone