zwave-js/zwave-js #2125: ZEN17 endpoints improperly mapped

blhoward2, 2021-03-21:
**Describe the bug**

The new Zooz ZEN17 Universal Relay allows the two inputs (S1/S2) to be decoupled from control of the two relays (R1/R2). The prototypical use case for this is a garage door connected to R1 and a reed switch connected to S2, which is templated in HA to be the open/close state of the door.

At present, when S2 is decoupled from R2, a change to S2 causes R1 to switch. The log file shows:

```
14:10:17.365 CNTRLR   [Node 140] Mapping unsolicited report from root device to first supporting end
                      point #1
```

Adding the `preserveRootApplicationCCValueIDs` compat flag to the device file fixes this aspect and a third entity is created for the root endpoint. I've merged a draft PR to this effect.

However, S2 changing still causes R2 to physically change. I'm unsure if this is a device bug (not respecting the parameter setting) or we're commanding the R2 endpoint to on:

```
14:25:40.237 DRIVER « [Node 140] [REQ] [ApplicationCommand]
           └─[BinarySwitchCCReport]
             current value: true
             target value: true
             duration:   [Duration: 0seconds]
14:25:40.312 DRIVER « [Node 140] [REQ] [ApplicationCommand]
           └─[MultiChannelCCCommandEncapsulation]
            │ source:   2
            │ destination: 0
            └─[BinarySwitchCCReport]
              current value: true
              target value: true
              duration:   [Duration: 0seconds]
```

Additionally, commanding S1 to on causes the root endpoint to also turn on, so this may confuse HA/users. (It does technically work because the root endpoint stays on if either R1 or R2 is on, so what happens at present is you command S1 on, R1 triggers door (then goes back off), root endpoint turns on with S1, reed sensor opens, report is received from S2, R2 physically turns on (shouldn't happen), S2 goes off, R2 goes off, root endpoint goes off.)

```
14:34:20.307 DRIVER » [Node 140] [REQ] [SendData]
           │ transmit options: 0x25
           │ callback id:   210
           └─[MultiChannelCCCommandEncapsulation]
            │ source:   0
            │ destination: 1
            └─[BinarySwitchCCSet]
              target value: true
14:34:20.316 DRIVER « [RES] [SendData]
            was sent: true
14:34:20.390 DRIVER « [REQ] [SendData]
            callback id:   210
            transmit status: OK
14:34:20.814 DRIVER « [Node 140] [REQ] [ApplicationCommand]
           └─[BinarySwitchCCReport]
             current value: true
             target value: true
             duration:   [Duration: 0seconds]
14:34:20.884 DRIVER « [Node 140] [REQ] [ApplicationCommand]
           └─[MultiChannelCCCommandEncapsulation]
            │ source:   1
            │ destination: 0
            └─[BinarySwitchCCReport]
              current value: true
              target value: true
              duration:   [Duration: 0seconds]
14:34:21.444 DRIVER » [Node 140] [REQ] [SendData]
           │ transmit options: 0x25
           │ callback id:   211
           └─[MultiChannelCCCommandEncapsulation]
            │ source:   0
            │ destination: 1
            └─[BinarySwitchCCGet]
14:34:21.452 DRIVER « [RES] [SendData]
            was sent: true
14:34:21.536 DRIVER « [REQ] [SendData]
            callback id:   211
            transmit status: OK
14:34:21.728 DRIVER « [Node 140] [REQ] [ApplicationCommand]
           └─[MultiChannelCCCommandEncapsulation]
            │ source:   1
            │ destination: 0
            └─[BinarySwitchCCReport]
              current value: true
              target value: true
              duration:   [Duration: 0seconds]
```

What should happen is: S1 triggers R1 to trigger door (then goes off), reed sensor opens, report is received from S2, S2 turns on, S2 goes off.) If our endpoint is tracking R2 and not S2 then fixing that would actually cause a different issue I think as we would no longer have an entity to template to track the door open/close state (unless the root endpoint) continues to turn on with S2.

blhoward2, 2021-03-21:
[node_interview.txt](https://github.com/zwave-js/node-zwave-js/files/6178568/node_interview.txt)

[node_140.json.zip](https://github.com/zwave-js/node-zwave-js/files/6178570/node_140.json.zip)

AlCalzone, 2021-03-23:
> The new Zooz ZEN17 Universal Relay allows the two inputs (S1/S2) to be decoupled from control of the two relays (R1/R2)

Did Zooz say how this is supposed to work in terms of Z-Wave? What you describe (or at least what's happening in the logs) seems like its not conforming with the specifications. Since it is a new device, I'm 99.9% sure we shouldn't need to use the compat flag to preserve root endpoint values:

[image]

blhoward2, 2021-03-23:
I haven't reached out to them yet as I wanted to understand if it was us or them. The issue seems to be that we always map the message on the root endpoint to endpoint 1, so when a message comes through for 2 we turn on 1.

AlCalzone, 2021-03-23:
> Their original email does say to exclude and re-include so a new entity is created.

That's required by the Z-Wave specs for actions which change the capabilities of a device. So it might just be that the device is only halfway there. It's a bit unfortunate, but you'll actually have to re-include without hard-resetting the device.

blhoward2, 2021-03-23:
Ok, I'll try it.

masonevans, 2021-03-23:
FWIW, I am seeing the exact same thing with my Zooz ZEN16 (originally opened as [an issue in HA Core](https://github.com/home-assistant/core/issues/48035)).  I hadn't gotten around to creating a PR to add the `preserveRootApplicationCCValueIDs` flag to the ZEN16 yet, so I guess I will first try excluding and re-including to see if that works.

blhoward2, 2021-03-23:
It didn't make any difference on mine. I don't have any switches connected to my zen16 to test it.

blhoward2, 2021-03-24:
Ok, after more info from Zooz. The root endpoint thing is something they're looking at but we need the compat flag for the ZEN17. Based on the description, we also need this for the ZEN16.

The relay controlling despite parameter 10/11 being enabled is because this only works with certain switch types. I've edited our description to make this clear.

Healing nearby nodes seems to have fixed my interview issues, at least the parameter values are now loading up on interview.

I think my PR closes these issues and renders this relay good-to-go.

PR #2126 (2021-03-24): fix: add compat flag to zen17 and zen16; fix zen17 config
Closes #2125
closed 2021-03-24 by AlCalzone