zwave-js/zwave-js #2127: Metadata for Notification CC V1 alarmLevel/Type has changed

kpine, 2021-03-21:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Since v6.5.1, the metadata for alarmLevel/Type values in Notification CC V1 has changed:
* In 6.5.0, type was "number" and "writeable" was false.
* In 6.5.1+, type is "any" and "writeable" is true.

The change has caused the corresponding entities in Home Assistant to become unavailable. The discovery schema assumes the notification cc-based sensor values will be of type "number", which was true from v6.5.0. Since that has changed, the entities are no longer discovered. I did not see anything in the changelog that would indicate a change in value type, although there look to be some notification cc-related changes.

The "writeable" property, while mostly benign, does cause zwavejs2mqtt to show it as an editable field, and setting it  obviously fails since these values aren't actually writable. In 6.5.0 it is a read-only field.

6.5.0:

[image]

6.5.1:

[image]

**Device information**

Which device(s) is/are affected (make/model)?  ZCOMBO smoke alarm (gen 1)
What are the node IDs? 10

**Did you change anything?**
- [X] **Yes**: *(please describe)* Upgraded zwave-js from v6.5.0 to a newer version (any version from 6.5.1).
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *(specify application with versions)* zwavejs2mqtt 2.0.0 with node-zwave-js 6.5.0.
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image: I used tagged versions, 2.0.0 shows this problem, 2.0.1 and later does not.
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Start the software
2. If HA is installed, observe the entities are unavailable, see metadata type of values is "any"
3. If using zwavejs2mqtt, see that the alarmLevel/Type are editable, sending the value fails

I was able to reproduce by simply downgrading to 6.5.0 using the existing cache files.

**Additional context**

None.

**Logfile:**
Logs/cache/dumps from 6.5.0: [6.5.0.zip](https://github.com/zwave-js/node-zwave-js/files/6178455/6.5.0.zip)
Logs/cache/dumps from 6.5.1: [6.5.1.zip](https://github.com/zwave-js/node-zwave-js/files/6178457/6.5.1.zip)
Logs/cache/dumps from 6.6.3: [6.6.3.zip](https://github.com/zwave-js/node-zwave-js/files/6178458/6.6.3.zip)

AlCalzone, 2021-03-24:
I'll need to see a log which includes a re-interview of the affected node and the receipt of a notification report.

kpine, 2021-03-24:
Which version do you need to see an interview of, and notification report, 6.5.0, 6.6.3, or both? The device was originally included in 6.5.0 or earlier. The type provided is based on whichever version is installed, no interview necessary to see it change.

AlCalzone, 2021-03-24:
One where it is broken.

kpine, 2021-03-24:
Re-interview seems to have fixed that, type is number and writeable is false now. 

A side effect of the re-interview: my node's manufacturer and product information are all unknown now.

[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/6200264/zwavejs.log)
[node_10.json.txt](https://github.com/zwave-js/node-zwave-js/files/6200267/node_10.json.txt)

AlCalzone, 2021-03-24:
> Re-interview seems to have fixed that, type is number and writeable is false now.

Probably because I changed how this metadata is defined (static -> dynamic)

> A side effect of the re-interview: my node's manufacturer and product information are all unknown now.

You had two interviews going on in parallel which broke each other. 7.0.0 fixes that (hopefully).

closed 2021-03-24 by AlCalzone