zwave-js/zwave-js #2128: Aeotec ZW098 LED Bulb - MultilevelSwitchCCSet/ColorSwitchCCSet: deserialization not implemented

andrewpc, 2021-03-21:
**Checklist:**

Issue does not appear to be related to any clients of this software. Confirmed seen whether using Home Assistant or ZwaveJS2MQTT

**Describe the bug**

I have 5 of the above mentioned bulbs ZW098. 

Whenever I switch these bulbs on or off I received an error in the log:
```
20:44:27.668 DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet:
deserialization not implemented
```
If I adjust the colour, I receive the following
```
20:48:50.465 DRIVER Dropping message because it could not be deserialized: ColorSwitchCCSet: deser
ialization not implemented
```

**Device information**

Which device(s) is/are affected (make/model): ZW098 LED Bulb - Firmware 1.5
What are the node IDs?: 11,12,17,18,19

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x ] **No**

**Did this use to work before?**
It worked in OZW. The error happens in both zwavejs2mqtt and zwavejs server within home assistant
    
**How are you using `node-zwave-js`**
I have tried both the native zwavejs implementation for home assistant and also zwavejs2mqtt. I have also tried setting values from both the Home Assistant UI and also from the control panel interface. The same issue is seen.

**To Reproduce**

Adjust the colour brightness of these bulbs and the error will occur.

**Additional context**

Add any other context about the problem here.

I will attach a log file if needed, but a short excerpt is below. Please let me know what you require

Brightness change

```
2021-03-22 10:00:16.863 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 50 => 0
2021-03-22 10:00:16.871 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 0 => 0
10:00:16.963 DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet:
deserialization not implemented
2021-03-22 10:00:21.908 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 0 => 0

```

Turn on and set color

```
2021-03-22 10:03:14.691 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 0 => 50
10:03:14.763 DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet:
deserialization not implemented
2021-03-22 10:03:17.723 INFO ZWAVE: Node 17: value updated: 51-0-currentColor-0 212 => 0
2021-03-22 10:03:17.725 INFO ZWAVE: Node 17: value updated: 51-0-currentColor-1 43 => 0
2021-03-22 10:03:17.725 INFO ZWAVE: Node 17: value updated: 51-0-currentColor-2 0 => 63
2021-03-22 10:03:17.726 INFO ZWAVE: Node 17: value updated: 51-0-currentColor-3 0 => 0
2021-03-22 10:03:17.727 INFO ZWAVE: Node 17: value updated: 51-0-currentColor-4 0 => 255
2021-03-22 10:03:17.728 INFO ZWAVE: Node 17: value updated: 51-0-currentColor [object Object] => [object Object]
2021-03-22 10:03:17.729 INFO ZWAVE: Node 17: value updated: 51-0-targetColor [object Object] => [object Object]
2021-03-22 10:03:17.729 INFO ZWAVE: Node 17: value updated: 51-0-hexColor 000000 => 3f00ff
2021-03-22 10:03:17.766 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 50 => 50
10:03:17.863 DRIVER Dropping message because it could not be deserialized: ColorSwitchCCSet: deser
ialization not implemented
10:03:17.961 DRIVER Dropping message because it could not be deserialized: MultilevelSwitchCCSet:
deserialization not implemented
2021-03-22 10:03:19.714 INFO ZWAVE: Node 17: value updated: 38-0-currentValue 50 => 50
```

-->

AlCalzone, 2021-03-22:
Can you show how your associations are configured? It seems like the controller is in a group it's not supposed to be in.

andrewpc, 2021-03-23:
You were right. There was a problem although I didn't set things up this way. The bulbs have two association groups. Both Group 1 (Lifeline) and Group 2 (Retransmit) had Node 1 as the target. When I removed Node 1 from the Group 2 association, the error did not appear in the logs. How would this have occurred? Maybe OZW which is what I was using before zwave js did this incorrectly? Thanks

AlCalzone, 2021-03-23:
It's likely that OZW had a weird association set up, this wouldn't be the first time.

andrewpc, 2021-03-23:
Unfortunately I need to re-open this. I removed all of the Group 2 associations earlier today, but I just noticed the error again and went and check and saw that they have come back:

[image]

I didnt do anything manual to add these in.

andrewpc, 2021-03-23:
Would it have something to do with the config which currently contains isLifeline:true for both Group 1 and Group 2?

```
	"associations": {
		"1": {
			"label": "Group 1",
			"maxNodes": 5,
			"isLifeline": true
		},
		"2": {
			"label": "Group 2",
			"maxNodes": 5,
			"isLifeline": true
		}
	}
```

robertsLando, 2021-03-23:
It could be a device configuration error, @AlCalzone ?

AlCalzone, 2021-03-23:
Yup, that second isLifeline seems incorrect:

> In this group, the [...] commands would be sent to the associated nodes in association group 2 when the Bulb receives [...] from the main controller.

@blhoward2 FYI in case you haven't touched these associations.

blhoward2, 2021-03-23:
> Yup, that second isLifeline seems incorrect:
> 
> 
> 
> > In this group, the [...] commands would be sent to the associated nodes in association group 2 when the Bulb receives [...] from the main controller.
> 
> 
> 
> @blhoward2 FYI in case you haven't touched these associations.

Done

PR #2413 (2021-04-21): fix(config): remove Group 2 Lifeline attribute for Aeotec ZW098
fixes #2128 

Current value is erroneous and causes problem outlined in above issue
closed 2021-04-21 by AlCalzone