zwave-js/zwave-js #2129: Aeotec ZW111/ZW116 Config - Basic Set Compat Flag needed

andrewpc, 2021-03-22:
The Aeotec ZW111 nano single dimmer allows for two physical switches to be connected. They can be configured to either control the direct load, send an event or do both. This is configured in config parameter 22-112-0-81 and 22-112-0-82 and then completed by adding the destination node (in my case Node 1) to either Group - Control:Key 1 or Group - Control:Key 2 depending on what you would like each switch to do.

If configured to send an event only, the send a Basic Set (only option) which by default with zwavejs gets mapped into a report. In order to use this event to perform a secondary function it needs to be an event so that Home Assistant can be triggered. I was able to get this working by making a local change to the zw111_2.3.json device config adding the compat flag.

```
"compat": {
		"treatBasicSetAsEvent": true
	}
```

Can this be added to this configuration? This will likely also be needed for ZW116 nano switch which supports the same functionality.

Thanks

AlCalzone, 2021-03-22:
@blhoward2 got any insight here?

blhoward2, 2021-03-22:
This sounds correct if it's sending a basic set. The question is when you set it to do both, does it send a status update and send the event? You'd need both to come through.

andrewpc, 2021-03-22:
If you configure it to send both, then it uses the Group 1 config item 22-112-0-80 for the report. Image attached:

[image]

blhoward2, 2021-03-22:
I understand that's what it seems to do, but can you confirm it in the log file?

andrewpc, 2021-03-22:
Yes I will do later today.

andrewpc, 2021-03-23:
I set Switch 2 to "Control itself and other nodes" and the set the Group 1 notification to Basic and the following log was output when pressing the switch

```
11:53:03.575 CNTRLR [Node 021] treating BasicCC Set as a value event
2021-03-23 11:53:03.576 INFO ZWAVE: Node 21: value notification: 32-0-event 99
2021-03-23 11:53:04.175 INFO ZWAVE: Node 21: value updated: 38-0-currentValue 0 => 99 
```

Setting Group 1 notification to Send a multilevel switch report and then pressing again showed the following:

```
11:55:23.978 CNTRLR [Node 021] treating BasicCC Set as a value event
2021-03-23 11:55:23.980 INFO ZWAVE: Node 21: value notification: 32-0-event 0
2021-03-23 11:55:24.475 INFO ZWAVE: Node 21: value updated: 38-0-currentValue 99 => 0
```

So as I understand it, both are working correctly there?

blhoward2, 2021-03-23:
Ok. I can add this to the file I'm working on.

andrewpc, 2021-03-23:
Thank you. I also have some ZW132 Dual Nanos which seem to implement this the same way. It hasn't affected me because I use the second switch to control a separate load, but in theory the same event for a switch could be used in an automation so it might be worth adding to these as well?

blhoward2, 2021-03-23:
Can you test it? I'm generally leery to add a compat flag like this one without actual testing on the device.

andrewpc, 2021-03-23:
Yes Results below. Confirming ZW132 and ZW116 (Dual and Single Nano Switch) both operate in the same way as the Dimmer:

ZW132

```
12:44:35.265 CNTRLR [Node 024] treating BasicCC Set as a report
2021-03-23 12:44:35.267 INFO ZWAVE: Node 24: value updated: 37-0-currentValue true => false
2021-03-23 12:44:36.448 INFO ZWAVE: Node 24: value updated: 37-1-currentValue true => false
```

ZW116

```
12:59:26.754 CNTRLR [Node 004] treating BasicCC Set as a report
2021-03-23 12:59:26.755 INFO ZWAVE: Node 4: value updated: 37-0-currentValue true => false
2021-03-23 12:59:28.323 INFO ZWAVE: Node 4: value updated: 50-0-value-66049 145.656 => 5.975
2021-03-23 12:59:28.323 INFO ZWAVE: Node 4: value updated: 50-0-deltaTime-66049 0 => 0
2021-03-23 12:59:29.223 INFO ZWAVE: Node 4: value updated: 37-0-currentValue false => false
2021-03-23 12:59:31.323 INFO ZWAVE: Node 4: value updated: 50-0-value-66049 5.975 => 0
2021-03-23 12:59:31.324 INFO ZWAVE: Node 4: value updated: 50-0-deltaTime-66049 0 => 0
```

AlCalzone, 2021-03-23:
> ```
> 11:53:03.575 CNTRLR [Node 021] treating BasicCC Set as a value event
> 2021-03-23 11:53:03.576 INFO ZWAVE: Node 21: value notification: 32-0-event 99
> 2021-03-23 11:53:04.175 INFO ZWAVE: Node 21: [...]

To be honest that looks like treating the Basic Set as an event is redundant here. That flag was added for devices that can send Basic Sets without changing the current value, e.g. the GE dimmers with the double tap function.

Maybe I'm misunderstanding the intent, but why would someone need the event in addition to the value report?

andrewpc, 2021-03-23:
@AlCalzone These devices do support basic set without changing the current value but @blhoward2 asked me to make sure that if we did so then it would not prevent status reports coming through (that was this log). When I use this functionality on the second attached switch, it doesnt control the connected light but instead generates an event that I use in home assistant to turn on my lamps.

AlCalzone, 2021-03-23:
> These devices do support basic set without changing the current value

Can you share a log where this is the case (with the flag if it must be)?

andrewpc, 2021-03-23:
> > These devices do support basic set without changing the current value
> 
> Can you share a log where this is the case (with the flag if it must be)?

```
21:09:45.436 CNTRLR [Node 021] treating BasicCC Set as a value event
2021-03-23 21:09:45.437 INFO ZWAVE: Node 21: value notification: 32-0-event 255
```

AlCalzone, 2021-03-23:
Is it the same for OFF (0)? And there's nothing after that like
> 2021-03-23 11:55:24.475 INFO ZWAVE: Node 21: value updated: 38-0-currentValue 99 => 0

andrewpc, 2021-03-23:
Here is me pressing the button four times. Note this switch (s2) is configured per the image 4 comments up with "control the other nodes" selected and node #1 in group 3. When configured this way, it does not control the connected light circuit, but only send the Basic set event. This allows you to connect a second switch to a single nano and use it to control something else.

```
- 23:09:00.308 CNTRLR [Node 021] treating BasicCC Set as a value event
- 2021-03-23 23:09:00.310 INFO ZWAVE: Node 21: value notification: 32-0-event 255
- 23:09:02.809 CNTRLR [Node 021] treating BasicCC Set as a value event
- 2021-03-23 23:09:02.811 INFO ZWAVE: Node 21: value notification: 32-0-event 0
- 23:09:04.707 CNTRLR [Node 021] treating BasicCC Set as a value event
- 2021-03-23 23:09:04.708 INFO ZWAVE: Node 21: value notification: 32-0-event 255
- 23:09:06.507 CNTRLR [Node 021] treating BasicCC Set as a value event
- 2021-03-23 23:09:06.507 INFO ZWAVE: Node 21: value notification: 32-0-event 0
```

AlCalzone, 2021-03-23:
Okay, that makes sense. Might be a bit confusing though that the event sometimes uses 99 and sometimes 255 as the value, depending on the config. And I think you won't be able to distinguish S1 (group 3) from S2 (group 4).

I'm on board with adding the flag @blhoward2. I think it's worth adding a comment in the file that the controller needs to be added to group 3 or 4 to make the event work.

blhoward2, 2021-03-26:
So, query whether we should add the device to group 3-4 by default? Does it hurt anything if the switches aren't decoupled?

Also, we're presently pulling the names for these groups from the device itself. @andrewpc Are there better names we could be using for these groups?

andrewpc, 2021-03-26:
If you add the controller to groups 3 and 4 automatically and a switch is connected then a basic set will be sent in addition to whatever type of report is sent using config item 80 for group 1. In theory it shouldn't cause any issues, unless there is confusion from users regarding why the extra event is sent? I think thats what you were asking?

As for the second question, the extract from the engineering sheet is below:

[image]

From reading that, I think it makes sense to have the group name be something that references the fact that this is related to one of the connected switches. Maybe something like "S1 External Switch - Group 3" or whatever you think seems the most consistent?

blhoward2, 2021-03-26:
Right, this is what we do for the GE switches. They get an extra event in exchange for users don't have to configure the groups themselves.

andrewpc, 2021-03-26:
Sounds good to me. There are 4 groups. Engineering sheet excerpt below. Naming might be as follows?:

Group 1 - Lifeline
Group 2 - Retransmit
Group 3 - External Switch S1 (Event | Trigger | Push )
Group 4 - External Switch S2 (Event | Trigger | Push )

One of those optional above for Group 3 and 4 might help identify what they are used for but optionally just the first part might work?

[image]

andrewpc, 2021-04-13:
Hey @blhoward2 How is the Aeotec refactor going? Just wondering when I might see the updated configs come through for  this issue and also #2128?

closed 2021-04-15 by AlCalzone

luke-barnett, 2021-05-29:
@AlCalzone @blhoward2  it looks like the compat flag didn't end up being set for ZW132 (dual nano switch)? I've just updated zwavejs (I use zwavejs2mqtt to give me a UI) and lost the sensor I was getting for it in home assistant. Finding this I figured it had moved to an event but I'm not seeing any trigger through when I listen for them via developer tools.

I have the same use case mentioned earlier where I've got one of the switches detached from the relay but I can't use it without an appropriate event firing through.

AlCalzone, 2021-05-29:
Doesn't look like it:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0086/zw132.json#L341

@zwave-js-bot add compat flag to 0x0086:0x0003:0x0084
```
// The device can report S1/S2 switch operation via Basic Set to groups 3 and 4
treatBasicSetAsEvent: true
```

AlCalzone, 2021-05-29:
Hmm, need to do that manually...

AlCalzone, 2021-05-29:
@luke-barnett can you post a driver log (loglevel "debug") where you operate the switch? Just want to make sure that the compat flag is actually needed.

luke-barnett, 2021-05-29:
@AlCalzone commented on the #2731 with logs.

PR #2349 (2021-04-15): fix(config): cleanup and template Aeotec configurations (part 2)
Aeotec templating group 2. Closes #2129