zwave-js/zwave-js #2131: Battery-powered Zooz Zen34 switch mysteriously reverts to ProtocolInfo

rccoleman, 2021-03-22:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

**Device information**

Which device(s) is/are affected (make/model)?  Zooz Zen34
What are the node IDs? 65 for the Zen34.  69 is my Kwikset lock, which is also part of the logging

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *(specify application with versions)*
I've been using the same manually build zwjs2mqtt container for a couple of weeks now, with versions as follows:

App Version
2.2.0+dd3e52e.dd3e52e

Zwavejs Version
6.6.1-922b3e45e

Zwavejs-server Version
1.1.1
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [x] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: master (pre-7.0, built roughly 2 weeks ago)
    - `zwavejs2mqtt` branch: master

- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

I have 2 Zen34 switches and both have have been working fine.  This evening, I noticed that one of them no longer worked, and I saw that it was stuck in ProtocolInfo stage in zwjs2mqtt.  I then saw the following sequence in the node-zwave-js logs that seems to have triggered it:

```
18:13:21.831 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.119 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.249 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.366 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.376 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.451 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.493 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.513 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.598 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.615 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.619 CNTRLR   [Node 065] Beginning interview - last completed stage: None
18:13:23.619 CNTRLR   [Node 065] new node, doing a full interview...
18:13:23.619 CNTRLR » [Node 065] querying protocol info...
18:13:23.703 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.704 CNTRLR   [Node 065] in the process of replying to a NonceGet, won't send another NonceR
                      eport
18:13:23.713 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:24.060 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:24.061 CNTRLR   [Node 065] in the process of replying to a NonceGet, won't send another NonceR
                      eport
18:13:24.070 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
```

Node 69 is my Kwikset lock, and it looks like node-zwave-js is repeatedly trying to get a nonce.  It then weirdly sends that same command to Node 65, my Zen34, which is not added securely.  After doing one more nonce request to the lock, it decides to start interviewing Node 65 (Zen34), and fails because it's battery-powered.  The entire log is attached.

Note that Node 69 (the lock) is added securely, but Node 65 (Zen34) is not:

[image]

Unfortunately, I only have Verbose logging enabled, and I've only had this happen once or twice before.  I'm using a manually-built container that I built 12 days ago and has been up for 8 days, and I haven't changed my ZWave network at all in that time.

**Additional context**

Add any other context about the problem here.

**Logfile:**

[zwavejs_1.log.zip](https://github.com/zwave-js/node-zwave-js/files/6179146/zwavejs_1.log.zip)

AlCalzone, 2021-03-24:
> Node 69 is my Kwikset lock, and it looks like node-zwave-js is repeatedly trying to get a nonce

You're misreading the log. Node 69 is repeatedly requesting a nonce so it can send an encrypted message. Similarly, Node 65 is also requesting one multiple times in short succession. Node 44 also seems to be such a candidate. You can also see earlier in the log that these devices spam their nonce responses too, although zwave-js requests only once.

Usually when these things happen, the devices have bad RF connectivity and/or bad routes. For bad routes (whereever they come from), healing the network can help.

As to why the interview suddenly restarts - I have no idea. It looks like it was a manual `refreshInfo` call, but I'm not sure. I'm seeing that a couple times in your log - maybe you have an automation that calls that?

rccoleman, 2021-03-26:
Thanks for clarifying that these are device requests and not node-zwave-js requests - that makes sense.  I can't imagine why node 65, a non-secure battery-operated Zooz Zen34 switch, would spontaneously be requesting a nonce.  Nothing in Home Assistant would be automatically refreshing the node values or initiating a re-interview.

So the time difference between this last exchange with the node and where it mysteriously tries to request a nonce and the reinterview starts is over an hour:

```
17:08:40.825 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[WakeUpCCWakeUpNotification]
17:08:40.827 CNTRLR « [Node 065] received wakeup notification
17:08:40.832 CNTRLR   [Node 065] The node is now awake.
17:08:40.837 DRIVER » [Node 065] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      189
                      └─[WakeUpCCIntervalGet]
17:08:40.845 DRIVER « [RES] [SendData]
                        was sent: true
17:08:40.860 DRIVER « [REQ] [SendData]
                        callback id:     189
                        transmit status: OK
17:08:40.876 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[WakeUpCCIntervalReport]
                          wake-up interval:   43200 seconds
                          controller node id: 1
17:08:41.137 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[BatteryCCReport]
                          level:  100
                          is low: false
17:08:41.877 CNTRLR » [Node 065] Sending node back to sleep...
17:08:41.881 DRIVER » [Node 065] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      190
                      └─[WakeUpCCNoMoreInformation]
17:08:41.889 DRIVER « [RES] [SendData]
                        was sent: true
17:08:41.904 DRIVER « [REQ] [SendData]
                        callback id:     190
                        transmit status: OK
17:08:41.908 CNTRLR   [Node 065] The node is now asleep.
```

```
18:13:23.598 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.615 DRIVER « [Node 069] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
18:13:23.619 CNTRLR   [Node 065] Beginning interview - last completed stage: None
18:13:23.619 CNTRLR   [Node 065] new node, doing a full interview...
18:13:23.619 CNTRLR » [Node 065] querying protocol info...
18:13:23.703 DRIVER « [Node 065] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
```

It looks like the weird exchange was triggered as I was leaving the house, which triggers an automation to lock all my doors, and it was during the attempt to lock my garage side door (node 69).  The motion sensor reports leading up to it (nodes 18, 39) & garage lights (node 24) all point to me leaving the house around 18:07, so there's no way that I was initiating a refresh or reinterview when all this happened.  69 (0x45) and 65 (0x41) are different by 1 bit, so I wonder if the nonce spamming caused a flipped bit and (somehow?) triggered a reinterview because a nonce request from a non-secure node wouldn't make sense.

I'll up the log level to debug and see I can catch anything else if it happens again.  I also healed those nodes to see if I can improve the comms.

AlCalzone, 2021-03-26:
> 69 (0x45) and 65 (0x41) are different by 1 bit, so I wonder if the nonce spamming caused a flipped bit and (somehow?) triggered a reinterview because a nonce request from a non-secure node wouldn't make sense.

I've seen before that users succeeded in having two nodes with the same ID in the network after `replaceFailedNode` when the old one wasn't really failed. Maybe that's the case here?

Z-Wave messages have a checksum to protect against single bits flipping, CRC8 for data rates 9.6kbps and 40kbps, CRC16-CCITT for data rate 100kbps. It is not entirely impossible but it seems odd to me that the bit would be reliably flipped multiple times in more than one location.

rccoleman, 2021-03-26:
I would think that'd see that in the zwavejs2mqtt, but I did have some trouble getting node 69 (the lock) added and I recall that I had to mark it "failed" and probably replace it with the function.  I'm definitely having some perplexing communications issues with that lock, so I'll work on that.

AlCalzone, 2021-03-27:
>  Perhaps we need to not only guard against whether a network key has been provided (as in the first `if` block), but also if we're just getting a nonsense nonce request for a non-secure node.

The entire point of that code you quoted is to detect if we mis-identified a device as unsecure because of a timeout during the initial interview and provide some kind of self-healing capacity. You are correct that there is one location where a new interview is forced when we receive a secure command from a node we thought to be insecure. 
-> driver.handleRequest, lines 1910ff.

Before I think about removing this, we need to make sure that there is no ghost lingering in your network.
It definitely has happened before and caused some really weird behavior where a different device reported different capabilities during the interview than the device that was supposed to be controlled.
The fact that it requests a nonce for Security S0 when the Zen34 only supports Security S2 makes me think that something like this is happening.

rccoleman, 2021-03-27:
Fair enough.  How would I find such a phantom device?  Reviewing the logs, I see that I did RemoveFailedNode on nodes 64 and 67 and ReplaceFailedNode on nodes 44 and 64.  Nothing that I see for node 65 other than the Zooz device, but it's a lot to go through.  There are no duplicate node IDs in HA or in the zwjs2mqtt web UI, and only the Zooz switch at node 65.

rccoleman, 2021-04-05:
I'm happy to help weed out any lingering ghost nodes, but I don't know where to look beyond the log perusal above.  It could also just be a firmware issue with these new Zen34 switches.

AlCalzone, 2021-04-06:
Unfortunately, the logs aren't much help figuring out if there are ghosts. From a `zwave-js` POV, devices using the same node ID are the same. You won't see any duplicates in the software.

You could start by making a physical inventory of all your devices and their node IDs - including battery powered ones in a drawer somewhere. Maybe something sticks out.

closed 2021-04-17 by zwave-js-assistant[bot]