zwave-js/zwave-js #2136: [bug]

atterdal, 2021-03-22:
Hi!

Non-updated Aeotec Nano Dimmers register as "Unknown product 0x006f" by "Unknown manufacturer 0x0086".

This makes it so that it's not possible to update firmware OTA.

[image]

AlCalzone, 2021-03-23:
@atterdal I need a zwave-js log to debug this. => https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs

SAABoholic, 2021-03-25:
FYI
I just successfully updated my US version ZW111 from 2.2 to 2.3 via the interface from ZWaveJS2MQTT 3.0 and ZWaveJS 6.6.0
using the .exe file in the FW package from Aotecs website.

closed 2021-04-05 by zwave-js-assistant[bot]