zwave-js/zwave-js #2155: Inovelli LZW36 - incorrect status reported when turning on fan

themitzman, 2021-03-25:
**Checklist:**

- [X ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X ] I have checked the troubleshooting section and my problem is **not** described there.
- [ X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**
When I turn on the fan on my fan/light dimmer switch, Zwave is reporting the light as turned on as well even though it is not.

**Device information**

Which device(s) is/are affected (make/model)?  Inovelli LZW36
What are the node IDs? 7

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [ X] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [X] **Yes**, in: *(specify application with versions)* It worked with openzwave beta
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [X] `HomeAssistant` version 2021.3.3
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...' in HA, go to devices and select "Master Bedroom"
2. Click on '...' Turn on "Master Bedroom (2)" this is the fan
3. Scroll down to '...'
4. See error - "Master Bedroom" indicator turns on as well. This is the light but the light is not actually on.

**Additional context**

Add any other context about the problem here.

**Logfile:**
[dc781aff_files.zip](https://github.com/zwave-js/node-zwave-js/files/6201414/dc781aff_files.zip)
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6201418/zwavejs_1.log)

blhoward2, 2021-03-25:
@AlCalzone I think this is the same issue as with Zooz:

```
01:25:48.055 CNTRLR   [Node 007] Mapping unsolicited report from root device to first supporting end
                      point #1
```

Endpoint 2 turning on is sending a root device message mapped to endpoint 1.

blhoward2, 2021-03-25:
User confirmed via discord that adding the compat flag to disable root mapping fixes this bug. I have reported this to Inovelli for their comment.

AlCalzone, 2021-03-25:
@themitzman it looks like your lifeline group contains two associations, one for Node 1, Endpoint 0 and one for Node 1, Endpoint 1.
Please remove the one to Node 1, Endpoint 1 and operate the fan afterwards. I want to make sure that this is not just the result of an incorrect association.

AlCalzone, 2021-03-25:
The main problem here is that we're getting a multi channel encapsulated report from endpoint 2 (correct) and a multicast (!) report from the root device. The latter not only seems incorrect but it is impossible to assign it to the correct endpoint when there are multiple possible sources it could have come from.

Unless this is just the result of the incorrect association, I might have to disable mapping altogether if it is ambiguous.

themitzman, 2021-03-25:
Could you explain that in dummy terms? I'm still fairly new to zwavejs so
I'm notn100% clear on what you mean by lifeline.

AlCalzone, 2021-03-25:
Z-Wave devices use associations to report changes to other devices and the controller. There is a dedicated group for the latter (normally group 1), which is called "Lifeline". Yours contains two associations, one of which might be the cause for the wrongly addressed reports.

themitzman, 2021-03-25:
Ok I see. When I get to my desk for work I'll take a look and reply back. I
appreciate the guidance. Maybe this happened when I migrated from ozw to
zwavejs.

AlCalzone, 2021-03-25:
It's likely that OZW set up the endpoint 1 target and didn't map the weird root device report to an endpoint. AFAIK OZW does not hide the unnecessary root device values by default.

Zwave-js sets up the endpoint 0 target during the device interview but now you have 2 associations. It would have cleaned up unnecessary node associations (without an endpoint), but the endpoint 1 association does not automatically get removed.

themitzman, 2021-03-25:
Got it. I am going to revert the compat change we did last night,
reinterview the node, then remove it from the lifeline. I'll let you know
how it goes.

AlCalzone, 2021-03-25:
Probably better if you first remove the unnecessary association. Not sure if that changes anything, but this way it should not leave any traces.

themitzman, 2021-03-25:
OK so I removed the compat flag from last night, restarted the container,
removed the second entry from the lifeline association (endpoint 1), then
reinterviewed the node.  The issue is now resolved!

So out of curiosity, do you think the migration from openzwave 1.4 was the
potential cause? Meaning, openzwave created that association when it
shouldn't have?

AlCalzone, 2021-03-25:
I think its a combination. OZW set up a weird association, which worked for OZW because it didn't do anything special with the root reports. And zwave-js doing over-eager mapping between endpoints.

themitzman, 2021-03-25:
Fair enough assessment. Appreciate the assistance and glad it's something
easily fixed.

closed 2021-03-25 by AlCalzone