zwave-js/zwave-js #2158: Fibaro FGRGBWM-441 - Missing state

0rsa, 2021-03-25:
Hello,

If I change the value of a module FGRGBWM-441 (class MultiLevelSwitch), it works but there is no event reporting this change, except for the power consumed.

I get only this event in node-red:
[image]

I have no problem using Fibaro roller shutter using the same class (MultiLevelSwitch).

What is expected? To get an event with the newValue of module MultiLevelSwitch

AlCalzone, 2021-03-25:
Please choose an appropriate issue template and fill it out completely when opening an issue. That helps us investigate your issue quicker because all the necessary information is there. Otherwise, the issue will go to the bottom of the pile - and that pile is growing quickly.

0rsa, 2021-03-25:
Ok it works if I specify a endpoint.

closed 2021-03-25 by 0rsa