zwave-js/zwave-js #2165: Changes to thermostat endpoint to value mapping in zwave-js 7.0.0

raman325, 2021-03-26:
**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ ] I have checked the troubleshooting section and my problem is **not** described there.
- [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

It seems that in some cases, upgrading to `zwave-js 7.0.0` results in different command classes being reported on different endpoints of the thermostat. In the example state dump provided below, the `Thermostat Setpoint CC` is on endpoint 0, but the `Thermostat Mode CC` and `Multilevel Sensor CC` values are on endpoints 1 and 2. Is this expected behavior?

Note that I was able to reproduce this when I upgraded but a re-interview resolved the issues I saw.

I don't think this is specific to `zwave-js-server` because in the server logic we simply dump the values that `zwave-js` provides.

**Device information**

Which device(s) is/are affected (make/model)?  Radio Thermostat CT100 and CT101 (dump is from CT101 and is from another user, I did not think to take a dump)
What are the node IDs? 5

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version 2021.3.4 with the zwave-js-server core addon v0.1.14 (https://github.com/home-assistant/addons/tree/master/zwave_js)
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Start with a version of zwave-js < 7.0.0 (must have a thermostat)
2. Upgrade to zwave-js 7.0.0 (don't wipe your existing cache before doing this)
3. Observe the value to endpoint mappings

**Additional context**

Add any other context about the problem here.

**Logfile:**

I wasn't able to get a logfile dump from the user but will ask. In the meantime I have the state dump of the node.
[thermostat.json.txt](https://github.com/zwave-js/node-zwave-js/files/6213484/thermostat.json.txt)

AlCalzone, 2021-03-27:
Any chance you could share zwave-js's cache files before and after the upgrade (without re-interview)?

I do have a thermostat but it has no endpoints I could use to reproduce.

raman325, 2021-03-27:
No promises but I'll try to revert and go through the update again

raman325, 2021-03-28:
here you go:
**6.6.3**
[d6aca8a3.json.txt](https://github.com/zwave-js/node-zwave-js/files/6216832/d6aca8a3.json.txt)
[d6aca8a3.metadata.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6216833/d6aca8a3.metadata.jsonl.txt)
[d6aca8a3.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6216834/d6aca8a3.values.jsonl.txt)
State dump: 
[thermostat_6.6.3.json.txt](https://github.com/zwave-js/node-zwave-js/files/6216835/thermostat_6.6.3.json.txt)

**7.0.0**
[d6aca8a3.json.txt](https://github.com/zwave-js/node-zwave-js/files/6216837/d6aca8a3.json.txt)
[d6aca8a3.metadata.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6216838/d6aca8a3.metadata.jsonl.txt)
[d6aca8a3.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/6216839/d6aca8a3.values.jsonl.txt)
State dump:
[thermostat_7.0.0.json.txt](https://github.com/zwave-js/node-zwave-js/files/6216840/thermostat_7.0.0.json.txt)

raman325, 2021-03-28:
BTW - this might be a device quirk because now reinterviewing does not fix the issue

AlCalzone, 2021-03-28:
I can't really tell because there is no zwave-js log - but could it be that some of the interview queries are sometimes timing out? This would explain why it sometimes works and sometimes doesn't.

I also don't see **any** reference to endpoint 1 in the cache files.

>  In the example state dump provided below, the Thermostat Setpoint CC is on endpoint 0, but the Thermostat Mode CC and Multilevel Sensor CC values are on endpoints 1 and 2. Is this expected behavior?

Ignoring endpoint 1 for now, this seems to be working as intended. In the files it looks like Thermostat Setpoint CC is only on endpoint 0, while the Multilevel Sensor CC exists on endpoint 2. Therefore having the values on different endpoints is expected.
Depending on how they were reported, it **could** be that prior to https://github.com/zwave-js/node-zwave-js/pull/2160 some reports were incorrectly mapped from the root to a different endpoint and now have values in the cache. This would have also been the case in 6.6.3 though.

That being said:
1. Make sure that parts of the interview aren't timing out due to reception problems
2. Make sure that the associations are set up correctly. OZW seems to create a lifeline to endpoint 1 which zwave-js does not yet clean up. It does look like they are okay (Lifeline to endpoint 0)
3. Make a zwave-js log of a re-interview.

raman325, 2021-03-28:
> I can't really tell because there is no zwave-js log - but could it be that some of the interview queries are sometimes timing out? This would explain why it sometimes works and sometimes doesn't. [...]

Huh that's interesting, I was looking at the state dump we get from `zwave-js-server` not the cache files. I see what you mean about no endpoint 1 in the cache files, so why does `zwave-js-server` get endpoint 1 from `node.getAllEndpoints` as well as when we retrieve some of the values?

raman325, 2021-03-28:
Just looking through the code, it looks like it's `node.getAllEndpoints` and `node.getDefinedValueIDs` that are returning endpoint 1

AlCalzone, 2021-03-28:
I really need to see a logfile to get a full picture. It may very well be expected that endpoint 1 **should** be existing.

AlCalzone, 2021-03-28:
Looking at the cache files again, this does look like the node has two static endpoints:
```
{"k":"{\"nodeId\":26,\"commandClass\":96,\"endpoint\":0,\"property\":\"individualCount\"}","v":2}
```
So either endpoint 1 should exist and respond or there should be another one that wasn't discovered either.

raman325, 2021-03-28:
I think you need more detailed logs than these right? Having a bear of a time trying to get debug logs enabled 
[logs.txt](https://github.com/zwave-js/node-zwave-js/files/6218014/logs.txt)

AlCalzone, 2021-03-28:
Only on mobile now, but it seems to be enough.

raman325, 2021-03-28:
Great, and just a note, I tried bumping the addon to 7.0.1 and it appears that now both the setpoint and mode CC's are on endpoint 1, which is more in line with what I expected. Could this be because of https://github.com/zwave-js/node-zwave-js/pull/2160?

Separately, is it expected that these used to be advertised on endpoint 0 but are now advertised on endpoint 1? We use the value ID to create unique IDs for the entities we create for values, so the net result of that is we end up with an entity that's orphaned (the one that was for the values on endpoint 0) and a new entity for the values that are on endpoint 1. If that's expected, then we will have to add some logic to clean up some of these orphaned entities.

AlCalzone, 2021-03-28:
If the same value exists on endpoint 0 and another one, the other one is preferred. So yes, that's intended.

But it should have always been that way, these recent changes should only affect unsolicited reports that are sent from endpoint 0 when they should have come from another one.

raman325, 2021-03-29:
Got it, will look into what we can do to clean up orphaned entities then.

So to my comment about 7.0.0 vs 7.0.1, would that PR explain the difference I saw? That is, on 7.0.0 Thermostat Setpoint was on endpoint 0 and Thermostat Mode was on endpoint 1, but in 7.0.1 both were on endpoint 1.

AlCalzone, 2021-03-29:
```
2021-03-28T20:32:43.584Z CNTRLR « [Node 026] received response for endpoint capabilities (#1):
                                  generic device class:  Thermostat
                                  specific device class: General Thermostat V2
                                  is dynamic end point:  false
                                  supported CCs:
                                    · Z-Wave Plus Info
                                    · Anti-Theft
                                    · Clock
                                    · Indicator
                                    · Association Group Information
                                    · Multilevel Sensor
                                    · Firmware Update Meta Data
                                    · Thermostat Mode
                                    · Thermostat Operating State
                                    · Thermostat Fan Mode
                                    · Thermostat Fan State
                                    · Thermostat Setpoint
                                    · Configuration
                                    · Battery
                                    · Association
```
Looks like the values should have always been on endpoint 1, not endpoint 0, since the CCs are explicitly listed and not only implied in the device class.

Endpoint 2 looks really strange though (but the device class should include Multilevel Sensor CC):
```
2021-03-28T20:32:43.630Z CNTRLR « [Node 026] received response for endpoint capabilities (#2):
                                  generic device class:  Multilevel Sensor
                                  specific device class: Unused
                                  is dynamic end point:  false
                                  supported CCs:
                                    · No Operation
                                    · No Operation
                                    · No Operation
                                    · No Operation
```

Like I said above, the PR is just for unsolicited reports. They were previously **always** mapped from root to a higher supporting endpoint. Now only if the mapping is unambiguous. So if anything, you would see values appear on endpoint 0 now.

My money is still on an incomplete interview due to a timeout, causing zwave-js to have missing information about endpoint 1 and did not map to that endpoint.

raman325, 2021-03-29:
Hmm ok. The odd thing about all of this is that I went down this path because of multiple user reports about this issue. I was not able to get logs from each user, but their experience mirrored mine in terms of what values ended up on what endpoints. They all seem to be using devices from the same manufacturer as me (Radio Thermostat of America) which makes me think it's a firmware issue of some sort. I've tried reaching out to them before but haven't had much success, wonder if it's worth another shot.

AlCalzone, 2021-03-29:
If you can reproduce, you should be able to use zwavejs2mqtt directly without HA and extract full logs for <7 and >=7 versions there.

raman325, 2021-03-29:
OK I should be able to try to reproduce later

raman325, 2021-03-30:
OK here we go:
[node_26_6.6.3.log](https://github.com/zwave-js/node-zwave-js/files/6225619/node_26_6.6.3.log)
[node_26_7.0.0.log](https://github.com/zwave-js/node-zwave-js/files/6225620/node_26_7.0.0.log)

raman325, 2021-03-30:
Here's a log when I upgrade and retain the cache: 
[node_26_7.0.0_from_existing_cache.log](https://github.com/zwave-js/node-zwave-js/files/6225675/node_26_7.0.0_from_existing_cache.log). I deleted the cache and restarted, and again values like `Air temperature` were not discovered

raman325, 2021-03-30:
Actually retaining the cache on upgrade doesn't always mean `Air temperature` is discovered - when I tried switching back to the addon (z2m 3.0.1, zwave-js 7.0.0) I couldn't get the `Air temperature` back

AlCalzone, 2021-03-30:
Hmm, your logs are truncated. I'll write you on discord so we don't pollute this issue.

raman325, 2021-04-02:
Adding logs here for visibility:
[6.6.3 (1).log](https://github.com/zwave-js/node-zwave-js/files/6247775/6.6.3.1.log)
first 7.0.0 run: [7.0.0 (1).log](https://github.com/zwave-js/node-zwave-js/files/6247776/7.0.0.1.log)
second 7.0.0 run: [zwavejs_1 (1).log](https://github.com/zwave-js/node-zwave-js/files/6247778/zwavejs_1.1.log)
third 7.0.0 run: [zwavejs_1 (2).log](https://github.com/zwave-js/node-zwave-js/files/6247780/zwavejs_1.2.log)

raman325, 2021-04-04:
@AlCalzone are you still waiting for any info from me? Not trying to rush you on this, just want to make sure you are unblocked.

AlCalzone, 2021-04-04:
I was away over the holidays. But I think I've found the culprit:
https://github.com/zwave-js/node-zwave-js/pull/2239

This caused no capabilities to be discovered, which in turn caused no values to be queried. Can you test the linked PR?

raman325, 2021-04-04:
hope you had a good holiday! I will test the PR as soon as I can figure out how to 🙂

AlCalzone, 2021-04-04:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/docker_custom_builds

raman325, 2021-04-05:
OK so it's hard to tell because it seems like the interview is currently inconsistent, but at least for now things seem to be more consistent with the linked PR - I have cleared my cache 3 times for a full interview and the multilevel sensors get discovered every time.

AlCalzone, 2021-04-05:
What do you mean by inconsistent?

closed 2021-04-05 by AlCalzone

raman325, 2021-04-05:
> Actually retaining the cache on upgrade doesn't always mean `Air temperature` is discovered - when I tried switching back to the addon (z2m 3.0.1, zwave-js 7.0.0) I couldn't get the `Air temperature` back

See my comment here. The list of values that were discovered would change between interviews

AlCalzone, 2021-04-05:
That's because the values weren't actually queried in 7.x, so they were not created if they didn't exist.

PR #2239 (2021-04-05): fix: also query capabilities in Multilevel Sensor V5 interview
fixes: #2165 
(I hope)