zwave-js/zwave-js #2166: Kwikset 888 alarmType and alarm

cppsse, 2021-03-26:
Kiwkset 888 Deadbolt is missing alarmType and alarmLevel attributes.  See issue #1713 ([specifically here](https://github.com/zwave-js/node-zwave-js/issues/1713#issuecomment-808504726))

Similar work was recently done for Kwikset 910, 912, 914 and Yale locks.  The existing merge for the 910, 912, and 914 models should work for the 888.  I used a generic Kwikset template sensor under OZW 1.4 to report on which keypad codes were used.

[image]
[image]

closed 2021-03-27 by AlCalzone

AlCalzone, 2021-03-27:
Just to be sure that you're not misunderstanding. You're not going to see the alarmType and alarmLevel values if the mapping is applied. Instead zwave-js will emit `"notification"` events and update the lock status for you.

acalcutt, 2021-04-01:
Hi @AlCalzone , I also have a Kwikset 888 and am trying to move my configuration over to Zwave JS. I am using Z-Wave JS 0.1.16. From what I can tell, the lock statuses are working properly, however "alarmType" and "alarmLevel" still list as unknown. Does you post above mean these values will still not get updated? 

One thing I used to do with a combination of "alarmType" and "alarmLevel" was figure out which user was entering. When a user entered it would give a alarmLevel of the user associated with the pin (1,2,3...30), and an alarmType of 19 (PIN Unlock). Using the combination of those I could see whos pin was used like this in node red.
[image]

With "alarmType" and "alarmLevel" still listed as unknown, I'm not sure I see any way to do something similar. If they were just exposed I could still use something like my node red setup to fill the gap. Unless there is something that can be done to catch the last user user the "alarmType" (19) and "alarmLevel" (number associated with the user)  and emit a motivation for that.

AlCalzone, 2021-04-01:
We have a mapping defined for the Kwikset 888 - you'll receive events (I think HA calls them zwavejs_event) which specify that a `RF (Un)lock` happened and which includes the user ID as a parameter.

acalcutt, 2021-04-01:
Thank, based on your comment I was able get something working with the "zwave_js_event" In node red I was able to listen to "zwave_js_event" with an "Events All" node. Once I had that I looked for a payload with "Keypad unlock operation" at $msg.payload.event.label and like you mentioned the UserID was there in the payload at $msg.payload.event.parameters.userId

My uploaded node red looks like this
[image]

 thanks for your help, I think watching "zwave_js_event" should be fine

acalcutt, 2021-04-01:
@AlCalzone I will say I found one issue with the lock status. This lock has a feature where the lock automatically locks after X seconds. When the lock automatically locks itself, it does not get updated in hass.io (it still shows "lock", like it is unlocked). 

I can see it does send a "Auto lock locked operation" event but the status doesn't change in hass. I had this problem with my old configuration, so right now I am just doing like I did then... if I see a "Auto lock locked operation" even I am just calling the lock.lock service (telling has to lock even though it already is locked, which updates the status...which is a hack but works)
[image]

AlCalzone, 2021-04-01:
I think that's already fixed on master but not released yet

cweakland, 2021-04-04:
Would it be possible to add alarmtype 161 "Bad Code Entered" to the list of notifications?

AlCalzone, 2021-04-04:
I'll need to check but AFAIK there is no direct translation to Notification events for this.

AlCalzone, 2021-04-04:
Yep, I remembered correctly. The V2 notifications distinguish between trying to lock and trying to unlock with an invalid user code. I assure you that if we map it to one of them, there will be reports that someone wants to have the other mapping.

So I'm not going to map either and let everyone decide themselves.

PR #2168 (2021-03-27): fix: add alarm value mapping for kwikset 888
Closes #2166