zwave-js/zwave-js #2174: Integration with device 012501 Popp & Co / Electric Strike Lock Control on FW 1.5 isn't working at all.

hubik77, 2021-03-27:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Integration with device 012501 Popp & Co / Electric Strike Lock Control on FW 1.5 isn't working at all. According to the docs it should be.

[image]

**Device information**

Which device(s) is/are affected (make/model)?  012501 Popp & Co / Electric Strike Lock Control
What are the node IDs?  
"id": 101,
  "deviceId": "340-1-5",
  "manufacturer": "Popp & Co",
  "manufacturerId": 340,
 "nodeId": 101,

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [x] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version core-2021.3.4
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Just try to use this device somehow.

**Additional context**

Device doesn't refresh it's state on home assistant and user is unable to control the device. 
Partly something works -> trying to open lock doesn't do anything before resetting whole HA then after reset lock opens.
The same goes for locking.

**Logfile:**

{
  "id": 101,
  "deviceId": "340-1-5",
  "manufacturer": "Popp & Co",
  "manufacturerId": 340,
  "productType": 5,
  "productId": 1,
  "name": "",
  "loc": "",
[... 307 lines cut ...]
  "keepAwake": false,
  "maxBaudRate": null,
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "status": "Alive",
  "interviewStage": "Complete",
  "hexId": "0x0154-0x0005-0x0001",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0154:0x0005:0x0001:0.0",
  "productLabel": "012501",
  "productDescription": "Electric Strike Lock Control",
  "zwaveVersion": 4,
  "zwavePlusVersion": 1,
  "nodeType": 0,
  "roleType": 5,
  "endpointsCount": 0,
  "isControllerNode": false,
  "deviceClass": {
    "basic": 4,
    "generic": 64,
    "specific": 1
  },
  "_name": "NodeID_101"
}

blhoward2, 2021-03-27:
Try excluding and reincluding as secure. Most locks won't work if included insecurely. Yours is insecure it seems (""isSecure": false")

AlCalzone, 2021-03-28:
And please **attach** logfiles and such as files. Having them inline makes these issues insanely hard to digest.

closed 2021-04-10 by zwave-js-assistant[bot]