zwave-js/zwave-js #2176: Replace Failed Node does not re-inteview the new device

steve28, 2021-03-27:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

When using "Replace Failed Node" in the control panel to replace a failed node with a different device (i.e., GE switch with an Inovelli switch), the new device is not interviewed upon inclusion.  As a result, it still appears as the old device in the node list and thus has incorrect paramters/config/etc. until a manual re-interview is performed.

**Device information**
I did two replacements of GE/Jasco 45609 on/off switches with Innovelli LZW31 dimmers.  I saw the same behavior both times.

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ x] Other: I am using the latest HA add-on, but I am interacting with zwave-js through the control panel, not HA.
**To Reproduce**

Steps to reproduce the behavior:
1. Have a node that's dead that you want to replace with a _different_ device
2. open the dead node from the node list
3. click advanced
4. Under Failed Nodes, click Replace, answer "no" to secure inclusion.
5. Include the new node
**Additional context**

Add any other context about the problem here.

**Logfile:** N/A

blhoward2, 2021-03-27:
Can you tell us what version of zwave-js is being used by the addon?

steve28, 2021-03-27:
> Can you tell us what version of zwave-js is being used by the addon? 
version 6.6.3.  I know 7.0.x is out, but they haven't updated the add-on yet.

I'll try to catch a log on my next switch update - hopefully tonight or tomorrow

steve28, 2021-03-27:
Here's the log - looks like it DOES interview:

```
16:28:43.186 CNTRLR » [Node 009] pinging the node...
16:28:47.316 CNTRLR   [Node 009] Node 9 did not respond after 1 attempts, it is presumed dead
16:28:47.317 CNTRLR   [Node 009] ping failed: Failed to send the command after 1 attempts (Status No
16:29:08.712 CNTRLR » [Node 009] Assigning return route to controller...
16:29:09.637 CNTRLR   [Node 009] Beginning interview - last completed stage: None
16:29:09.637 CNTRLR   [Node 009] new node, doing a full interview...
16:29:09.637 CNTRLR » [Node 009] querying protocol info...
16:29:09.654 CNTRLR « [Node 009] received response for protocol info:
16:29:09.681 CNTRLR   [Node 009] Interview stage completed: ProtocolInfo
16:29:09.681 CNTRLR » [Node 009] pinging the node...
16:29:09.732 CNTRLR « [Node 009] ping successful
16:29:09.733 CNTRLR » [Node 009] querying node info...
16:29:09.890 CNTRLR « [Node 009] node info received
16:29:09.917 CNTRLR   [Node 009] Interview stage completed: NodeInfo
16:29:09.918 CNTRLR » [Node 009] querying securely supported commands...
16:29:20.251 CNTRLR   [Node 009] Timed out while waiting for a response from the node
16:29:20.252 CNTRLR   [Node 009] The node is not included securely. Continuing interview non-securel
16:29:20.348 CNTRLR   [Node 009] ManufacturerSpecificCC: doing a complete interview...
16:29:20.349 CNTRLR » [Node 009] querying manufacturer information...
16:29:20.523 CNTRLR « [Node 009] received response for manufacturer information:
16:29:20.552 CNTRLR   [Node 009] VersionCC: doing a complete interview...
16:29:20.552 CNTRLR » [Node 009] querying node versions...
16:29:20.709 CNTRLR « [Node 009] received response for node versions:
16:29:20.709 CNTRLR » [Node 009] querying CC versions...
16:29:20.709 CNTRLR » [Node 009]   querying the CC version for Multilevel Switch...
16:29:20.890 CNTRLR   [Node 009]   supports CC Multilevel Switch (0x26) in version 2
16:29:20.891 CNTRLR   [Node 009]   skipping query for All Switch (0x27) because max implemented vers
16:29:20.891 CNTRLR » [Node 009]   querying the CC version for Z-Wave Plus Info...
16:29:21.080 CNTRLR   [Node 009]   supports CC Z-Wave Plus Info (0x5e) in version 2
16:29:21.080 CNTRLR » [Node 009]   querying the CC version for Configuration...
16:29:21.270 CNTRLR   [Node 009]   supports CC Configuration (0x70) in version 1
16:29:21.270 CNTRLR » [Node 009]   querying the CC version for Association...
16:29:21.460 CNTRLR   [Node 009]   supports CC Association (0x85) in version 2
16:29:21.460 CNTRLR » [Node 009]   querying the CC version for Association Group Information...
16:29:21.650 CNTRLR   [Node 009]   supports CC Association Group Information (0x59) in version 1
16:29:21.650 CNTRLR   [Node 009]   skipping query for Transport Service (0x55) because max implement
16:29:21.651 CNTRLR » [Node 009]   querying the CC version for Version...
16:29:21.840 CNTRLR   [Node 009]   supports CC Version (0x86) in version 2
16:29:21.840 CNTRLR » [Node 009]   querying the CC version for Manufacturer Specific...
16:29:22.089 CNTRLR   [Node 009]   supports CC Manufacturer Specific (0x72) in version 2
16:29:22.090 CNTRLR   [Node 009]   skipping query for Device Reset Locally (0x5a) because max implem
16:29:22.090 CNTRLR   [Node 009]   skipping query for Powerlevel (0x73) because max implemented vers
16:29:22.090 CNTRLR   [Node 009]   skipping query for Security (0x98) because max implemented versio
16:29:22.090 CNTRLR   [Node 009]   skipping query for Security 2 (0x9f) because max implemented vers
16:29:22.090 CNTRLR   [Node 009]   skipping query for Supervision (0x6c) because max implemented ver
16:29:22.090 CNTRLR » [Node 009]   querying the CC version for Protection...
16:29:22.379 CNTRLR   [Node 009]   supports CC Protection (0x75) in version 2
16:29:22.380 CNTRLR   [Node 009]   skipping query for Application Status (0x22) because max implemen
16:29:22.380 CNTRLR » [Node 009]   querying the CC version for Firmware Update Meta Data...
16:29:22.657 CNTRLR   [Node 009]   supports CC Firmware Update Meta Data (0x7a) in version 4
16:29:22.657 CNTRLR   [Node 009] trying to load device config
16:29:22.677 CNTRLR   [Node 009] device config loaded
16:29:22.697 CNTRLR   [Node 009] ZWavePlusCC: doing a complete interview...
16:29:22.697 CNTRLR » [Node 009] querying Z-Wave+ information...
16:29:22.842 CNTRLR « [Node 009] received response for Z-Wave+ information:
16:29:22.844 CNTRLR   [Node 009] MultilevelSwitchCC: doing a complete interview...
16:29:22.848 CNTRLR » [Node 009] requesting current switch state...
16:29:23.089 CNTRLR   [Node 009] ConfigurationCC: Loading configuration parameters from device confi
16:29:23.108 CNTRLR » [Node 009] querying parameter #1 value...
16:29:23.335 CNTRLR « [Node 009] parameter #1 has value: 3
16:29:23.336 CNTRLR » [Node 009] querying parameter #2 value...
16:29:23.549 CNTRLR « [Node 009] parameter #2 has value: 101
16:29:23.549 CNTRLR » [Node 009] querying parameter #3 value...
16:29:23.734 CNTRLR « [Node 009] parameter #3 has value: 101
16:29:23.735 CNTRLR » [Node 009] querying parameter #4 value...
16:29:23.924 CNTRLR « [Node 009] parameter #4 has value: 101
16:29:23.924 CNTRLR » [Node 009] querying parameter #5 value...
[... 55 lines cut ...]
16:29:59.421 CNTRLR » [Node 009] Association group #1: Querying command list...
16:29:59.732 CNTRLR » [Node 009] Association group #2: Querying name...
16:30:00.036 CNTRLR « [Node 009] Association group #2 has name "On/Off Control"
[... cut ...]
16:30:02.317 CNTRLR « [Node 009]   node neighbors received: 2, 4, 20, 34, 35, 43, 53, 67, 73
16:30:02.317 CNTRLR   [Node 009] Interview stage completed: Neighbors
16:30:02.317 CNTRLR   [Node 009] Interview completed
16:30:02.317 CNTRLR   [Node 009] The node is ready to be used
```
It must be some display issue where the CP does not update.  In this case, I had left the CP for a while (5-10 min) and came back in, then it had populated.  I know the first two I replace there was no amount of refresh/reload of the browser that would show me the new info or config params.

blhoward2, 2021-03-28:
@robertsLando Thoughts?

robertsLando, 2021-03-28:
The data are sent as soon as I receive them from Zwavejs so excluded any kind of network delays this could just be something on zwavejs side, maybe the interview takes a lot before reaching the ready status, this can be normal with some devices

AlCalzone, 2021-03-28:
Can you share the full logfile? I wonder why we try to ping the device first.

steve28, 2021-03-28:
Attached: [zwavejs_355 2.log](https://github.com/zwave-js/node-zwave-js/files/6217476/zwavejs_355.2.log)

The one with debug on starts at line 45061, where I replace Node 009.

AlCalzone, 2021-03-30:
Ok, the 2nd ping is purely "cosmetical" - shouldn't hurt. I was confused by the 1st one but that is done to ensure the node is really failed.

AlCalzone, 2021-03-30:
@robertsLando The interview takes about a minute here but looks pretty normal.