zwave-js/zwave-js #2179: Zwave network stall after pressing Zwave remote button

djaydev, 2021-03-28:
From original issue #1884 logs capture on 2021-02-20, as well logs showing continued problem on 2021-3-28.

Using a Zwave remote to send a "Central Scene" to Home Assistant.  After pressing any button on the remote, HA sees the scene and sends a command back to zwavejsmqtt to turn on or off a light.  The problem is the command to toggle the light is stalled until these response attempts messages timeout.  Sometimes it's stalled/delayed, sometimes the command is just never executed.  Is there a way to avoid trying to update these remote devices after receiving the wake up notification?

Log sample when the button is pressed, but no other command during and/or after can be executed:
```
2021-02-20 20:08:35.307 INFO ZWAVE: Node 52: value notification: 91-0-scene-001 0
20:08:35.309 CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x8a0001"
}
20:08:35.334 CNTRLR « [Node 052] received wakeup notification
2021-02-20 20:08:35.337 INFO ZWAVE: Node 52 is now awake
20:08:35.343 CNTRLR [Node 052] The node is now awake.
2021-02-20 20:08:35.399 INFO ZWAVE: Node 52: value updated: 132-0-wakeUpInterval 0 => 0
2021-02-20 20:08:35.401 INFO ZWAVE: Node 52: value updated: 132-0-controllerNodeId 1 => 1
2021-02-20 20:08:35.452 INFO ZWAVE: Node 52: value updated: 132-0-wakeUpInterval 0 => 0
2021-02-20 20:08:35.453 INFO ZWAVE: Node 52: value updated: 132-0-controllerNodeId 1 => 1
20:08:39.603 CNTRLR [Node 052] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-02-20 20:08:42.634 INFO ZWAVE: Node 52: value notification: 91-0-scene-001 0
20:08:42.636 CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x8b0001"
}
20:08:42.661 CNTRLR [Node 052] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
20:08:42.666 CNTRLR « [Node 052] received wakeup notification
20:08:47.280 CNTRLR [Node 052] The node did not respond after 3 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-02-20 20:08:47.282 INFO ZWAVE: Node 52 is now asleep
20:08:47.292 CNTRLR [Node 052] The node is now asleep.
```
[node_52.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6030557/node_52.json.txt)
[node_53.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6030558/node_53.json.txt)

Hello,

Unfortunately the issue still exist.  If I press the button on the zwave remote, I get the logs below.  When the "Failed to execute controller command after 1/3 attempts" comes up, no other zwave devices can be controlled the whole network stalls for several seconds every press.  On the dashboard all the remotes show as "sleep" and "complete" under status and interview stage.

```
2021-03-28 09:45:11.186 INFO ZWAVE: Node 52 is now awake
2021-03-28T13:45:11.189Z CNTRLR [Node 052] The node is now awake.
2021-03-28T13:45:11.195Z CNTRLR Failed to execute controller command after 1/3 attempts. Scheduling next try i
n 100 ms.
2021-03-28 09:45:11.348 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 12 => 32
2021-03-28 09:45:11.362 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 32 => 32
2021-03-28T13:45:12.191Z CNTRLR » [Node 052] Sending node back to sleep...
2021-03-28 09:45:15.935 INFO ZWAVE: Node 52: value notification: 91-0-scene-002 0
2021-03-28T13:45:15.937Z CNTRLR « [Node 052] received CentralScene notification {
"nodeId": 52,
"ccId": "Central Scene",
"ccCommand": "0x03",
"payload": "0x3c0002"
}
2021-03-28T13:45:15.960Z CNTRLR [Node 052] The node did not respond after 1 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-03-28 09:45:15.963 INFO ZWAVE: Node 52 is now asleep
2021-03-28T13:45:15.971Z CNTRLR [Node 052] The node is now asleep.
2021-03-28T13:45:15.990Z CNTRLR Failed to execute controller command after 1/3 attempts. Scheduling next try i
n 100 ms.
2021-03-28T13:45:15.999Z CNTRLR « [Node 052] received wakeup notification
2021-03-28 09:45:16.001 INFO ZWAVE: Node 52 is now awake
2021-03-28T13:45:16.005Z CNTRLR [Node 052] The node is now awake.
2021-03-28 09:45:16.128 INFO ZWAVE: Node 50: value updated: 38-0-currentValue 0 => 0
2021-03-28 09:45:16.183 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 32 => 52
2021-03-28 09:45:16.197 INFO ZWAVE: Node 20: value updated: 38-0-currentValue 52 => 52
2021-03-28T13:45:17.008Z CNTRLR » [Node 052] Sending node back to sleep...
2021-03-28T13:45:21.156Z CNTRLR [Node 052] The node did not respond after 1 attempts.
It is probably asleep, moving its messages to the wakeup queue.
2021-03-28 09:45:21.158 INFO ZWAVE: Node 52 is now asleep
2021-03-28T13:45:21.162Z CNTRLR [Node 052] The node is now asleep.
```

AlCalzone, 2021-03-28:
The error you're seeing is something that trips up the controller stick itself. Without the logs, all I can do is guess.

I **really** need the full logfile that zwave-js writes (loglevel `debug`) from startup until this happens. Please do not copy it from the web UI, because that messes up the formatting and includes logs from zwavejs2mqtt that I don't need.
AFAIK you can copy the logfiles out of the Docker container somehow but I'm not sure how.

djaydev, 2021-03-28:
Logs from both the docker container, and one from changing the settings in the UI to log level debug, log to file.

[_zwavejs2mqtt_logs.txt](https://github.com/zwave-js/node-zwave-js/files/6217785/_zwavejs2mqtt_logs.txt)
[zwavejs2mqtt.log.txt](https://github.com/zwave-js/node-zwave-js/files/6217786/zwavejs2mqtt.log.txt)

AlCalzone, 2021-03-28:
Sorry, but that's not the correct logfile. I need the one from `zwave-js`, not `zwavejs2mqtt`.
=> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs

djaydev, 2021-03-28:
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6217958/zwavejs2mqtt.log)
 new log file based on:
=> https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=export-logs

AlCalzone, 2021-03-28:
You did not read what I wrote.

> I need the one from **zwave-js**, not zwavejs2mqtt.

[image]

djaydev, 2021-03-28:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6218007/zwavejs_1.log)

new log file based on:
https://user-images.githubusercontent.com/17641229/112766759-83814e80-9013-11eb-9846-a4324989d86c.png

AlCalzone, 2021-03-29:
Ok it looks like this is caused by a really irritating communication pattern from your remote.
It tells the driver that it is awake and ready to accept commands, but **immediately** stops responding. This on its own wouldn't be so bad, but I do see 1-2 instances where this wakeup pattern causes zwave-js to re-query the device's wakeup interval (which times out after 3-5s each time).

In addition, it first sends a Central Scene notification to which your automation triggers a command to Node 50 which gets sent at the same time that node 52 sends its wakeup notification. This causes a collision, zwave-js backs off and tries to send again after 100ms. Which is what you're seeing in the log above, but that is correct behavior as per the specs and should not really be noticeable.

The first part we can work around by telling the driver not to send the device to sleep and not to query its wakeup interval on wakeup, but I'll need your cache files (`<hex-number>.json` and `.values.jsonl`) which should be in the same directory as the log.

djaydev, 2021-03-29:
[cachefiles.zip](https://github.com/zwave-js/node-zwave-js/files/6221829/cachefiles.zip)

AlCalzone, 2021-03-29:
I've disabled the main culprits for the behavior you're seeing for devices with no regular wake up. Should be in the next release.

closed 2021-03-29 by AlCalzone

djaydev, 2021-03-29:
fantastic work! thanks!

PR #2186 (2021-03-29): fix: minimize automatic interaction with manual wakeup nodes
fixes: #2179