zwave-js/zwave-js #2183: Issues with zen32 scene controller

jjlawren, 2021-03-28:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Including the zen32 creates a node but it doesn't seem to finish interviewing. Information for the node remains empty along with the device ID.

**Device information**

Which device(s) is/are affected (make/model)?  Zooz zen32
What are the node IDs? 30

**Did you change anything?**
- [x] **Yes**: *(please describe)* Added new device
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [x] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [x] `zwavejs2mqtt` (latest) docker image (currently still on 2.4.0)
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Attempt to include the device.

**Logfile:**

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6218134/zwavejs_1.log)

jjlawren, 2021-03-28:
Was able to get it fully recognized after excluding & re-including. Will test a bit more to see if anything seems off.

AlCalzone, 2021-03-29:
Your log is full of this:
```
16:14:41.463 DRIVER « [RES] [SendData]
                        was sent: false
16:14:41.464 CNTRLR   The controller response indicated failure after 1/3 attempts. Scheduling next 
                      try in 100 ms
```

This means the stick is stuck in a state where it cannot send messages. It seems to be caused by some delay, followed by zwave-js retrying a command.

AlCalzone, 2021-03-29:
If that happens again, pulling the stick, putting it back in and restarting zwave-js should do the trick.

jjlawren, 2021-03-29:
Let's call the initial inclusion problems a fluke, it's behaving as I'd expect now that it's fully joined.

The only obvious "issue" I see with this device is cosmetic where one of the "LED Indicator" labels seems incorrect:
[image]
It counts Relay->1->2->3->5 where I'd expect 4 in the last value. Is this a firmware oopsie?

AlCalzone, 2021-03-29:
Most likely a typo in our config file

AlCalzone, 2021-03-29:
==> https://github.com/zwave-js/node-zwave-js/pull/2189

closed 2021-03-29 by jjlawren