zwave-js/zwave-js #2190: Multicast commands from zwavejs2mqtt not working

mattstocum, 2021-03-29:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Sending a multicast command to zwavejs2mqtt results in a timeout of the command and all communication with the nodes included in the command getting queued until the timeout occurs.

**Device information**

Which device(s) is/are affected (make/model)?  Zooz Zen 22 and Nortek PD300Z-2
What are the node IDs? 4 and 5

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [X] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Publish following data to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/multicast/set

{
  "nodes": [
    4, 5
  ],
  "commandClass": 38,
  "property": "targetValue",
  "value": 33
}

**Additional context**

I thought the issue might be related to the one node supporting SupervisionCC, and the other not, but I tried sending commands to two nodes that both do not support SupervisionCC and saw the exact same behavior. I did change the light states individually on the two nodes in the log file so that can be seen working. 

**Logfile:**
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6223709/zwavejs_1.log)

blhoward2, 2021-03-29:
Can you please post your mqtt log too? Also at a silly level.

mattstocum, 2021-03-29:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6224499/zwavejs_1.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6224500/zwavejs2mqtt.log)
Two fresh logs both set to silly.

AlCalzone, 2021-03-29:
The command gets sent
```
2021-03-29T20:50:16.525Z DRIVER » [REQ] [SendDataMulticast]
                                  │ target nodes:     4, 5
                                  │ transmit options: 0x25
                                  │ callback id:      36
                                  └─[MultilevelSwitchCCSet]
                                      target value: 0
...
2021-03-29T20:50:16.536Z DRIVER « [RES] [SendDataMulticast]
                                    was sent: true
```
but we're not getting the callback whether it was received or not. Are both nodes in range of the controller?

mattstocum, 2021-03-29:
Both nodes have an almost literal line of sight to the controller, there's a single hollow core door in between them and the controller, and about 10 to 15 feet away from it. Non-multicast commands work perfectly, but when I try to do a multicast command it sends the command, then after a few minutes I get the timeout notification, for the second attempt, then again for the third. I just tried again using 5 nodes that are the same type of device and I'm seeing the exact same behavior. The controller I'm using is the Zooz 700 series stick.

AlCalzone, 2021-03-29:
Then it looks like the stick is not behaving as it should. You'll need to wait a few days though - I'm in contact with Silabs to verify, but that will likely take until after easter.

mattstocum, 2021-03-29:
Thanks. Let me know if there’s anything else I can test. I do have the pc z-wave controller software installed, so if there’s a way to send multicast commands from that I can try to determine if it’s the stick or Zwave js. 

Matt

AlCalzone, 2021-03-30:
The command is being sent and we get confirmation from the stick, so I'd say it's very unlikely that zwave-js is messing up here.

AlCalzone, 2021-03-30:
FYI I can reproduce with my 700-series stick. Looking at Zniffer, nothing gets sent. When I do the same with my 500-series RasPi hat, it works.

blhoward2, 2021-03-30:
Interesting. So did they change the command for the 700 series?

AlCalzone, 2021-03-30:
700-series is using the Bridge Controller API instead of Static Controller API. That one has a few new commands, including the Command Handler we already implemented and two new calls `SendData_Bridge` and `SendDataMulticast_Bridge`.

Apparently the current 700 firmware reports both `SendData` and `SendDataMulticast` as supported, but doesn't actually support `SendDataMulticast`. `SendData` seems to have a macro that uses `SendData_Bridge`, `SendDataMulticast` seems not to.

AlCalzone, 2021-03-30:
Fix is up at #2194

closed 2021-03-30 by AlCalzone