zwave-js/zwave-js #2196: Unable to power toggle GE Z-Wave Plus 40-Amp Indoor/Outdoor Smart Switch (14285)

tswyman, 2021-03-30:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here. @raman325
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

My Z-Wave controller connects to the GE/Jasco Smart Switch just fine. It reports the correct power state based on the manual switch, as well as other various power usage data.
The problem is when I try to power toggle the "binary switch" nothing happens.
It looks like Home Assistant Sends a command, but the Switch just does not respond.

**Device information**

Which device(s) is/are affected (make/model)?  
GE Z-Wave Plus 40-Amp Indoor/Outdoor Smart Switch (GE/Jasco-14285)
What are the node IDs?
7

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [X] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [X] `HomeAssistant` version core-2021.3.4
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to GE Z-Wave Plus 40-Amp Indoor/Outdoor Smart Switch in Devices
2. Click on Binary Switch slider button to toggle power state
4. Slider Button will toggle and command is sent
4. Slider Button will revert to previous state after 1 second

**Additional context**

**Logfile:**
[zwave_js_dump14285.json.txt](https://github.com/zwave-js/node-zwave-js/files/6230476/zwave_js_dump14285.json.txt)

raman325, 2021-03-30:
Just to note, this is not working via zwavejs2mqtt either which is why I suggested he come here

marcus-j-davies, 2021-03-30:
@tswyman,

Please could you attach the `DEBUG` logs, that are generated by the Z-Wave JS library?, I'm not sure how you do that with HomeAssistant or any middleware that maybe involved.

The node configuration as attached, is helpful, but it doesn't show the communication between the Controller/node/Z-Wave JS.

Thanks.

raman325, 2021-03-30:
If you are using zwavejs2mqtt @tswyman you can set the log level to debug in settings and then log it to a file so you can grab it

tswyman, 2021-04-27:
[zwavejs_361.log.txt](https://github.com/zwave-js/node-zwave-js/files/6381450/zwavejs_361.log.txt)
[zwavejs2mqtt.log.txt](https://github.com/zwave-js/node-zwave-js/files/6381451/zwavejs2mqtt.log.txt)

Here are my logs.  Sorry it's been so long for me to get this to you guys.
Thanks!

AlCalzone, 2021-04-27:
The log looks perfectly normal - except that the device doesn't react to the switch command. It doesn't support Protection CC and there doesn't seem to be a config param that blocks the manual interaction.

I think you'll have to contact the manufacturer support for this one.

tswyman, 2021-04-27:
Ok.  Thanks!

closed 2021-04-27 by tswyman