zwave-js/zwave-js #2205: Changing log level via Driver.updateLogConfig does not change the log level

raman325, 2021-03-31:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

This is a follow on to #1908. Now, changing `logToFile` does affect the driver (I can switch logging to file and console at will), but changing log level does not.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [x] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [x] Other: HomeAssistant Z-Wave JS Addon (installation method shouldn't matter, more importantly running `zwave-js 7.0.1`)

**To Reproduce**

Steps to reproduce the behavior:
1. Start the driver with a log level of `info`
2. Call the `driver.updateLogConfig` command and set the log level to `debug`
3. Check the logs

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-03-31:
Wtf? I specifically tested that 😕

raman325, 2021-03-31:
> Wtf? I specifically tested that 😕

I mean I suppose it could be something in `zwave-js-server` but I can't currently see how at the moment... I can confirm that calling `getLogConfig` shows debug as the log level but I am not seeing it in the console... Just let me know if it works on your end again when you test it and I'll check back through the server code

AlCalzone, 2021-03-31:
Ok looks like I broke it after the fix 🙈

PR #2243 (2021-04-05): fix: remove unnecessary caching for loglevel visibility, fix loglevel configuration
fixes: #2205
closed 2021-04-05 by AlCalzone