zwave-js/zwave-js #2208: Inovelli Scene Events Buffered?

legomind, 2021-03-31:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt: 3.0.2
zwave-js: 7.0.0

### Describe the bug

Scene scene events seem to be buffered and only emitted after the subsequent scene has been triggered.

### To Reproduce

Steps to reproduce the behavior:

Scene triggered (1) -> No Event
Scene triggered (2) -> Event for Scene (1)
Scene triggered (3) -> Event for Scene (2)

Basically scenes events are only emitted after another scene is triggered.

This behavior does not exist using ozw, therefore it is not likely the inovelli devices.

### Expected behavior

Appropriate scene events should be raised right away.

### Additional context

blhoward2, 2021-04-01:
We're going to need logs at the Silly level to begin to look into this. Please ensure they show the interview and the various scene events.

AlCalzone, 2021-04-01:
Zwave-js logs to be specific. The request is included in the issue template for a reason.

legomind, 2021-04-01:
Ok. I'll collect the logs shortly

legomind, 2021-04-02:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6251074/zwavejs_1.log)

Log attached.

Here is the sequence of scenes I triggered:

- up
- down
- up x2
- down
- up
- down
- up x2
- down
- up x2
- down
- up
- down

legomind, 2021-04-02:
It occurred to me that having home assistant events would be usefull, using the exact same sequence as above, here are the events as listed in home assistant events page.

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6251093/zwavejs_1.log)

```
Event 11 fired 4:01 PM:
{
    "event_type": "zwave_js_event",
    "data": {
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
        "device_id": "a45ea9b6ab5b08c96fdbcd13fa4b3d7f",
        "command_class": 91,
        "command_class_name": "Central Scene",
        "label": "Scene 001",
        "property": "scene",
        "property_name": "scene",
        "property_key": "001",
        "property_key_name": "001",
        "value": "KeyPressed",
        "value_raw": 0
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-02T20:01:18.242906+00:00",
    "context": {
        "id": "6d3fc09996acb449884cf736b0ecad87",
        "parent_id": null,
        "user_id": null
    }
}
Event 10 fired 4:01 PM:
{
    "event_type": "zwave_js_event",
    "data": {
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
        "device_id": "a45ea9b6ab5b08c96fdbcd13fa4b3d7f",
        "command_class": 91,
        "command_class_name": "Central Scene",
        "label": "Scene 002",
        "property": "scene",
        "property_name": "scene",
        "property_key": "002",
        "property_key_name": "002",
        "value": "KeyPressed2x",
        "value_raw": 3
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-02T20:01:16.845162+00:00",
    "context": {
        "id": "a5739ee4375d6ec2c5940288e3b4717a",
        "parent_id": null,
        "user_id": null
    }
}
Event 9 fired 4:01 PM:
{
    "event_type": "zwave_js_event",
    "data": {
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
[... 247 lines cut ...]
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
        "device_id": "a45ea9b6ab5b08c96fdbcd13fa4b3d7f",
        "command_class": 91,
        "command_class_name": "Central Scene",
        "label": "Scene 002",
        "property": "scene",
        "property_name": "scene",
        "property_key": "002",
        "property_key_name": "002",
        "value": "KeyPressed",
        "value_raw": 0
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-02T20:00:06.345369+00:00",
    "context": {
        "id": "9c2877cc95df8c01d1b9a6a66db3330a",
        "parent_id": null,
        "user_id": null
    }
}
```

AlCalzone, 2021-04-04:
> Here is the sequence of scenes I triggered:

This matches exactly what I'm seeing in the logs (scene number 2 seems to be up, 1 down). The commands are received as-is from the device.

<details>
<summary>Expand to see logs</summary>

```
2021-04-02T19:50:26.643Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 122
                                      key attribute:   KeyPressed
                                      scene number:    2

2021-04-02T19:50:29.539Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 123
                                      key attribute:   KeyPressed
                                      scene number:    1

2021-04-02T19:50:33.240Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 124
                                      key attribute:   KeyPressed2x
                                      scene number:    2

2021-04-02T19:50:35.641Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 125
                                      key attribute:   KeyPressed
                                      scene number:    1

2021-04-02T19:50:39.639Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 126
                                      key attribute:   KeyPressed
                                      scene number:    2

2021-04-02T19:50:41.439Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 127
                                      key attribute:   KeyPressed
                                      scene number:    1

2021-04-02T19:50:42.841Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 128
                                      key attribute:   KeyPressed2x
                                      scene number:    2

2021-04-02T19:50:44.439Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 129
                                      key attribute:   KeyPressed
                                      scene number:    1

2021-04-02T19:50:48.040Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 130
                                      key attribute:   KeyPressed2x
                                      scene number:    2

2021-04-02T19:50:49.650Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 131
                                      key attribute:   KeyPressed
                                      scene number:    1

2021-04-02T19:50:51.141Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 132
                                      key attribute:   KeyPressed
                                      scene number:    2

2021-04-02T19:50:52.640Z DRIVER « [Node 102] [REQ] [ApplicationCommand]
                                  └─[CentralSceneCCNotification]
                                      sequence number: 133
                                      key attribute:   KeyPressed
                                      scene number:    1

```

</details>

Of course I can't tell **when** you pressed the buttons and when the corresponding commands were received - so if there is buffering going on in the device, you won't see it here without commentary and some longer pauses between commands.

legomind, 2021-04-04:
Would an annotated log be of any help?

legomind, 2021-04-05:
Ok here is an annotated log:

After pressing up x2:

zwavejs log:
```
17:05:29.008 SERIAL « 0x010b00040066055b03ad030267                                        (13 bytes)
17:05:29.010 SERIAL » [ACK]                                                                   (0x06)
17:05:29.011 DRIVER « [Node 102] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 173
                          key attribute:   KeyPressed2x
                          scene number:    2
17:05:29.012 CNTRLR   [Node 102] [!] [Central Scene] scene[002]: 3
2021-04-04 17:05:29.013 INFO ZWAVE: Node 102: value notification: 91-0-scene-002 3
17:05:29.013 CNTRLR « [Node 102] received CentralScene notification {
                          "nodeId": 102,
                          "ccId": "Central Scene",
                          "ccCommand": "0x03",
                          "payload": "0xad0302"
                      }

```

home assistant event:
```
{
    "event_type": "zwave_js_event",
    "data": {
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
        "device_id": "a45ea9b6ab5b08c96fdbcd13fa4b3d7f",
        "command_class": 91,
        "command_class_name": "Central Scene",
        "label": "Scene 002",
        "property": "scene",
        "property_name": "scene",
        "property_key": "002",
        "property_key_name": "002",
        "value": "KeyPressed",
        "value_raw": 0
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-05T00:05:29.014646+00:00",
    "context": {
        "id": "2e92bb5a0ba4ec3a79b5d4be275d9ca2",
        "parent_id": null,
        "user_id": null
    }
}
```

Subsequent action was down:

zwavejs log:
```
17:09:50.307 SERIAL « 0x010b00040066055b03ae000164                                        (13 bytes)
17:09:50.308 SERIAL » [ACK]                                                                   (0x06)
17:09:50.309 DRIVER « [Node 102] [REQ] [ApplicationCommand]
                      └─[CentralSceneCCNotification]
                          sequence number: 174
                          key attribute:   KeyPressed
                          scene number:    1
17:09:50.312 CNTRLR   [Node 102] [!] [Central Scene] scene[001]: 0
2021-04-04 17:09:50.314 INFO ZWAVE: Node 102: value notification: 91-0-scene-001 0
17:09:50.315 CNTRLR « [Node 102] received CentralScene notification {
                          "nodeId": 102,
[... 219 lines cut ...]
        "type": "value_notification",
        "domain": "zwave_js",
        "node_id": 102,
        "home_id": 3569435897,
        "endpoint": 0,
        "device_id": "a45ea9b6ab5b08c96fdbcd13fa4b3d7f",
        "command_class": 91,
        "command_class_name": "Central Scene",
        "label": "Scene 002",
        "property": "scene",
        "property_name": "scene",
        "property_key": "002",
        "property_key_name": "002",
        "value": "KeyPressed",
        "value_raw": 0
    },
    "origin": "LOCAL",
    "time_fired": "2021-04-05T00:13:14.814050+00:00",
    "context": {
        "id": "c0c9cec095cc91bd82a95536fda3ecbb",
        "parent_id": null,
        "user_id": null
    }
}
```

AlCalzone, 2021-04-05:
Ok this looks like zwave-js picks up the correct events but HA doesn't always get the correct ones. I think this may be caused in zwave-js-, let me check with the devs of that.

legomind, 2021-04-05:
Ok, if i need to create this issue elsewhere let me know.

raman325, 2021-04-05:
What version of Home Assistant are you running @legomind ?

legomind, 2021-04-05:
core-2021.3.1 in docker container

kpine, 2021-04-05:
You should upgrade to the latest 2021.3.4. There was a fix in 2021.3.2.

raman325, 2021-04-05:
Generally speaking you should stay up to date with patch releases to avoid situations like this one 🙂

legomind, 2021-04-05:
wow i totally missed the patch releases. my bad; updating now. will close if the update resolves the issue.

legomind, 2021-04-05:
updating to 2021.3.4 fixed the issue. I feel like an idiot. :-P
sorry for the wasted time.

closed 2021-04-05 by legomind