zwave-js/zwave-js #2216: Schlage BE469 does not always reflect correct state/output CC notification

FuzzyMistborn, 2021-04-01:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

I'm trying to get the event notification for my door lock to determine what user code unlocked the lock.  However, the notification is intermittent in actually showing up, and oftentimes the devices state isn't accurately reflected in HASS (IE I'll manually unlock the door and 30 seconds later it'll show unlocked, sometimes a lot longer).  Other times it's instant and shows up right away.

**Device information**

Which device(s) is/are affected (make/model)?  Schlage BE469
What are the node IDs? Node 78

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [X] **No**

**Did this use to work before?**
- [X] **Don't know**, this is a new device (first time I'm really noticing the issue)
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [X] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [ ] `HomeAssistant` version XYZ
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6240701/zwavejs_1.log)

sstratoti, 2021-04-03:
I'm experiencing similar issues. I don't know if this helps, but I compared the Zwave-JS json file to the OZW equivalent.

In the OZW config file, it looks like it was updated from the ZWA specs in November 2020.

Paremeter 6 has the following:

```
<CommandClass id="113">
    <!-- These Door Locks don't send a DoorLockReport when the
		Lock Status is Changed, but instead send a Alarm Message -
		So we trigger a Refresh of the DoorLock Command Class when
		we recieve a Alarm Message Instead -->
    <TriggerRefreshValue Genre="user" Index="6" Instance="1">
      <RefreshClassValue CommandClass="98" Index="1" Instance="1" RequestFlags="0"/>
    </TriggerRefreshValue>
  </CommandClass>
```
For parameter 6 in Zwave-JS:

```
		"6": {
			"label": "User Slot Bit Field",
			"description": "Occupied slots for User Codes",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
```
HTH.

sstratoti, 2021-04-03:
Also, so you don't have to go hunting:
Original issue: https://github.com/OpenZWave/Zwave2Mqtt/issues/742
and revision in OZW: https://github.com/OpenZWave/open-zwave/commit/c9697ac2eb6278252675eb699ab8b6c7138015f8

So it looks like the original file that was added to OZW had this fix in it, but the parameter index was wrong (0 instead of 6).

However, parameter 6 IS different between the two projects, and I didn't have any issues when I was on OZW. I just migrated yesterday, and my front door lock is not updating/refreshing its values/status in homeassistant whenever someone unlocks the door.

AlCalzone, 2021-04-03:
I'll be able to look at the logs on Monday. Just a heads up, comparing **configuration** param 6 with whatever OZW's value index 6 for the **Notification** CC is, is like comparing apples and oranges.

sstratoti, 2021-04-03:
Totally get that. Was just trying to note that both files were generated by ZWA xml, and OZW seems to have a note indicating that the fixed a problem with that generation. 

Thanks for taking a look, and all the work you’re doing with this. Aside from this lock it’s been mostly smooth sailing!

AlCalzone, 2021-04-04:
@FuzzyMistborn 

The loglevel is a bit low, but I can see that the device is spamming requests for nonces (number used **once**) to encrypt the secure messages several times per second:
```
2021-04-01T02:46:04.539Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2021-04-01T02:46:04.549Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2021-04-01T02:46:04.588Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2021-04-01T02:46:04.689Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2021-04-01T02:46:04.706Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
2021-04-01T02:46:04.894Z CNTRLR   [Node 078] in the process of replying to a NonceGet, won't send another NonceR
                                  eport
```
As a result, `zwave-js` either does not respond to most of these requests and always invalidates all but the last one, because only one per device must be valid at a given time. When an encrypted commands with one of those invalidated onces is received, it gets discarded. `zwave-js` conforms with the Z-Wave specifications in its strict nonce handling whereas OZW did not - which is likely why you didn't notice in OZW which mess was going on behind the scenes.

Usually this spamming seems to be a result of a bad network health - messed up routes et cetera. I suggest you start by performing a network heal.

FuzzyMistborn, 2021-04-04:
Huh.  I swear I'd turned on "silly" logging.  I'll try a heal and report back.

FuzzyMistborn, 2021-04-06:
A heal seems to have vastly improved performance.  Likely cause was that I replaced a bunch of switches and probably forgot to heal to reset all the pathways. @sstratoti I'd suggest giving that a try.

closed 2021-04-06 by FuzzyMistborn

FoxxMD, 2021-04-07:
@FuzzyMistborn how exactly are you getting the user code from an event in HA? That value, as well as whether the lock event was manual (from inside) or with keypad, used to be attributes in `new_state` (in node-red payload from HA on lock event) but are no longer there.