zwave-js/zwave-js #2217: Aeotec ZW100 motion sensor undesired behaviour after 0.1.10

pixal-it, 2021-04-01:
For HomeAssistant, this is how you do it:
  Home Assistant -> settings -> Integrations -> Z-Wave JS -> Configure -> Create dump -> zip the json file and post it here.

- [ x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
Please see this issue that was raised for the issue as well:  https://github.com/home-assistant/core/issues/48412
- [ x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [ x] I have checked the troubleshooting section and my problem is **not** described there.
- [ x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**
The Aeotec ZW100 PIR "motion" entity, times out exactly after the configured time-out time in parameter 3, if you wouldn't know better, this sounds like everything works as designed. But what doesn't work as designed any longer, is that new or continuous motion during this countdown should reset or keep resetting that internal countdown until the last motion is detected.
Currently, the motion sensor no longer seems to communicate new detected motion to HA and things become very unreliable. As a side effect, as long as the physical MultiSensor detects or keeps detecting motion for a longer period of time (while the entity in HA is "clear", the entity in HA will not be updated. You need to leave the room, wait until the MultiSensors internal time-out has finished and only then motion will result in the HA entity changing to "detected",

The last known good version of zwaveJS is 0.1.10, to which I keep rolling back. I have tried all later versions including 0.1.16 today but the issue keeps appearing. I have two different setups with the same issue.

**Device information**

Which device(s) is/are affected (make/model)?
- Aeotec ZW100 MultiSensor 6

What are the node IDs?
- 63, 64, 65, 66, 70

**Did you change anything?**
No, but rolling back to the 0.1.10 snapshot instantly fixes the issues.

**Did this use to work before?**
Yes, Z-wave JS 0.1.10 and earlier.
    
**How are you using `node-zwave-js`**
- Home Assistant OS 5.12, core-2021.3.4  supervisor-2021.03.6
- Driver Version: 7.0.1
- Server Version: 1.3.1

**To Reproduce**

Steps to reproduce the behavior:
1. The issue is reproducable for me on both Raspberry 3 and when using the qcow2 OS on a NUC both supervised images.
2. Using an Aeotec Gen5 USB Controller ZW090 if relevant.
3. Include a factory reset ZW100 MultiSensor Module. Default time out, parameter 3, is 240 seconds.
4. Monitor the entity that represents the motion sensor. Once movement is detected, notice the time until no motion is detected.
   - if no other motion is detected, the entity should go to "clear" after 240 seconds as usual
   - if you create new motion within these 240 seconds, the entity should restart the 240 seconds counter, the entity should remain "Detected" until 240 seconds after the last motion. (this is the issue here)

[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6243781/zwave_js_dump.zip)

AlCalzone, 2021-04-01:
I need a zwave-js logfile (level "debug") to check if zwave-js is really doing something wrong here, if the device isn't sending the notification or if zwave-js-server/HA messes up.

pixal-it, 2021-04-01:
After seeing someone with very similar setup as I have, including the same Aeotec sensors and the described issue did not appear to be there I was absolutely confused but it's clear that it should work out of the box.

I took the following steps that seem to be the workaround for me for now.

1. (I'm not sure if this step was relevant) but I uninstalled the older but known good Z-Wave JS add-on. The Z-Wave JS integration immediately started installing the latest Z-Wave JS add-on again but this time it took much much longer to query all device parameters and interview them.

Afterwards I saw that next to the binary_sensor_xx_motion, every Multisensor now had a **binary_sensor_xx_any** next to it. This extra **any** entity doesn't have the motion device class by default but does fully function, including the expected motion=on as long as there is movement.

closed 2021-04-01 by pixal-it

AlCalzone, 2021-04-01:
What you're describing sounds like either the device wasn't interviewed yet or HA messes up somewhere.