zwave-js/zwave-js #2237: Add a device compat flag to allow Basic Set value mapping

kpine, 2021-04-04:
**Is your feature request related to a problem? Please describe.**
In short, since zwave-js v7.0.1, the Vision Security ZG8101 Tilt Sensor is no longer functional at all in HA (and in zwavejs2mqtt does not function the same way).

In more detail, this device doesn't work the way it used to prior versions: the Binary Sensor CC value never changes. I am sure this is due to PR #2152, as the device sends a Basic Set 0 when closed, and Basic Set 255 when open. Those previously mapped to the Binary Sensor CC (device generic type Binary Sensor), resulting in false/true values. With the referenced PR, the Sets have no effect.

I presume this device does not follow the Z-Wave specification, from what I see a Basic Set command is not allowed to be mapped to a Binary Sensor CC. However, the past behavior worked perfectly, both in ZWave-JS and OZW. The binary sensor value is very easy to use in Home Assistant without any workarounds.

**Describe the solution you'd like**
A device compat flag called `treatBasicSetAsReport`, or something similar, that enables the previous behavior of mapped values for Basic Set commands. For this device, the Basic Set would map to Binary Sensor CC. Coincidentally, this is something OZW [supports](https://github.com/OpenZWave/open-zwave/blob/99f10b5fe7f5511b2ffc1e889d390130c1b58cdc/config/vision/zg8101.xml#L12). Its default behavior is to not map Basic Sets, only Reports, and this compat flag allows the desired behavior. So looking for something very similar.

**Describe alternatives you've considered**
I could use the Basic value in HA, however there appears to be some new bug (related?) in the zwave-js-server and it doesn't even expose the Basic currentValue value anymore (it did previously). So I can't even use that right now.

If that did work, I'd still have to make a template sensor to bring back the previous behavior. Not the end of the world, but it would be more convenient to have the binary sensor working by default. I imagine most users expect this device to work this way, as it always has across different software.

This device also supports Notification V2 values, but they are buggy as well (surprise). The values are permanently set to non-zero values (tilt open sets the same value as tilt closed), and will auto-idle to 0 after 5 mins (by zwave-js). For a garage door tilt sensor this behavior is useless, the state will nearly always be wrong.

**Additional context**
The docs still indicate that basic mapping is the default: <https://zwave-js.github.io/node-zwave-js/#/config-files/file-format?id=disablebasicmapping>. I assume this must only apply to Basic Reports now. Would it be helpful to update the docs and be more specific (Basic Report) due to the change in behavior with 7.0.1?

Cache files and log (including re-interview) for completeness. The Tilt Sensor is Node 23.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6254022/zwavejs2mqtt-store.zip)

Hackashaq666, 2021-04-05:
In addition to the Nortek WAPIRZ-1 device mentioned by @kevingeorge, this issue seems to apply to the Nortek device WADWAZ-1 door/window sensor, and most likely the Vision and Monoprice variants of the same device.

closed 2021-04-05 by AlCalzone

kpine, 2021-04-05:
The fix works perfectly, thank you! No re-interview was necessary, for anyone else needing this.

olddawgpowers, 2021-04-05:
kpine- I am not clear what the fix is/was.  Is there something I need to do?

AlCalzone, 2021-04-05:
If the device really needs the flag (I'd need the device manual, device id and a log to confirm), the flag needs to be added to that device's configuration file.

olddawgpowers, 2021-04-05:
> If the device really needs the flag (I'd need the device manual, device id and a log to confirm), the flag needs to be added to that device's configuration file.

Thanks- I don't think I edit the device's configuration file in Home Assistant using Z-Wave js.  It may be something they add later.

AlCalzone, 2021-04-05:
I can do it but I need the stuff I asked for above as verification

kevincgeorge, 2021-04-05:
> If the device really needs the flag (I'd need the device manual, device id and a log to confirm), the flag needs to be added to that device's configuration file.
I can provide a log, device id and manuals, i'm using just the standard zwave-js integration in HA.  Can you let me know which log you need?

AlCalzone, 2021-04-05:
The one node-zwave-js writes. Should start with a large ascii art of "Z-WAVE"

kevincgeorge, 2021-04-05:
My apologies it looks like I spoke too soon on the logs point; I'm having a hard time figuring out how to get the whole log file out of zwave-js in HA supervised.
The devices are: 0x014f/wadwaz-1.json and 0x014f/wapirz-1.json
I've included the manuals.
[WADWAZ-1 WAPIRZ-1 manuals.zip](https://github.com/zwave-js/node-zwave-js/files/6260003/WADWAZ-1.WAPIRZ-1.manuals.zip)

AlCalzone, 2021-04-05:
I guess I'll need that log and your cache file after all. One of your manuals includes this table, which looks like the device is supposed to send Notification CC Reports:
[image]

Best if you ask the HA community how you can get these - I don't know.

kevincgeorge, 2021-04-05:
Here's the log.  Node 036 is the WADWAZ-1 and Node 007 is the WAPIRZ-1
[zwavejs_359.log](https://github.com/zwave-js/node-zwave-js/files/6261052/zwavejs_359.log)

kevincgeorge, 2021-04-06:
And I think this is the other file you were asking for.
[e4440334.zip](https://github.com/zwave-js/node-zwave-js/files/6261198/e4440334.zip)

PR #2244 (2021-04-05): feat: opt-in to map `Basic CC Set` to better CCs with compat flag `enableBasicSetMapping`
fixes: #2237