zwave-js/zwave-js #2252: Missing device config

AlCalzone, 2021-04-06:
**Which device is missing?**

Unsure, looks like this is something from Ring.

**What are the IDs?**

Manufacturer ID: 0x0346
Product type: 0x0701
Product ID: 0x0401
Firmware Version: 1.11

**Discovered info:**

associationGroups: [Lifeline]

parameters:
```json
{
    "1":{
        "defaultValue":70,
        "info":,
        "isAdvanced":true,
        "maxValue":70,
        "minValue":1,
        "name":"Heartbeat Interval",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "2":{
        "defaultValue":1,
        "info":,
        "isAdvanced":true,
        "maxValue":5,
        "minValue":0,
        "name":"Application Retries",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "3":{
        "defaultValue":5,
        "info":,
        "isAdvanced":true,
        "maxValue":60,
        "minValue":1,
        "name":"Application Level Retry Base Wait Time Period",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "4":{
        "defaultValue":25,
        "info":,
        "isAdvanced":true,
        "maxValue":100,
        "minValue":0,
        "name":"Battery level low threshold",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "5":{
        "defaultValue":10,
        "info":,
        "isAdvanced":true,
        "maxValue":100,
        "minValue":0,
        "name":"Battery level critical threshold",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "6":{
        "defaultValue":10000,
        "info":,
        "isAdvanced":true,
        "maxValue":30000,
        "minValue":500,
        "name":"Supervisory Report Timeout",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":2,
        "writeable":true
    },
    "7":{
        "defaultValue":10,
        "info":,
        "isAdvanced":true,
        "maxValue":15,
        "minValue":0,
        "name":"Z-Wave Sleep Timeout",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "8":{
        "defaultValue":3,
        "info":,
        "isAdvanced":true,
        "maxValue":10,
        "minValue":0,
        "name":"Siren duration",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "9":{
        "defaultValue":10,
        "info":,
        "isAdvanced":true,
        "maxValue":100,
        "minValue":0,
        "name":"QRBP switch threshold",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    },
    "10":{
        "defaultValue":50,
        "info":,
        "isAdvanced":true,
        "maxValue":255,
        "minValue":0,
        "name":"Logo LED ramp time",
        "noBulkSupport":false,
        "readable":true,
        "requiresReInclusion":false,
        "type":"number",
        "valueFormat":1,
        "valueSize":1,
        "writeable":true
    }
}
```

supportsAGI: True

supportsConfigCCV3: True

supportsZWavePlus: True

zWavePlusVersion: 2

marcus-j-davies, 2021-04-06:
Found it (I'm pretty sure)
Config params match

Very little (none) on Alliance database (from my basic search)

[Ring Alarm Outdoor Siren Zwave.pdf](https://github.com/zwave-js/node-zwave-js/files/6264795/Ring.Alarm.Outdoor.Siren.Zwave.pdf)

xhemp, 2021-04-06:
I have one of those and can get any further logs if needed.

marcus-j-davies, 2021-04-06:
could you share your node dump (specifically, the device ID's - just the siren will be fine), if they match with the issues title, then we're golden.

xhemp, 2021-04-06:
Sure thing! I added some of the configuration propertyname/description I got from the manual yesterday but here's the dump.

[ring_outdoor_siren.txt](https://github.com/zwave-js/node-zwave-js/files/6265984/ring_outdoor_siren.txt)

marcus-j-davies, 2021-04-06:
@blhoward2,

Device ID's Match - its Rings outdoor siren (couldn't locate this on Alliance for some reason)

xhemp, 2021-05-10:
It has been added already, i think @AlCalzone needs t close this one.

closed 2021-05-10 by AlCalzone