zwave-js/zwave-js #2258: Emulate devices for testing

blhoward2, 2021-04-07:
It would be helpful for testing downstream for the node to emulate various device classes and allow sending values through the actual library, including things like central scene notifications, basic sets, etc. Alarm values for devices would also be useful (door, water, etc.). This will lessen the need for actual devices to test downstream changes and ease development.

Ideally we would eventually emulate actual devices using data captured from the device, but as an initial step there should at least be a way to setup basic emulated devices of each class and to push through expected types of messages.

@raman325

blhoward2, 2021-04-07:
@billiaz @ahochsteger Any interest?

raman325, 2021-04-07:
See https://github.com/zwave-js/zwave-js-server/issues/171 for context. I'd like to add more test coverage to `zwave-js-server` (we have basically nothing now) so we need test data to work with. If `zwave-js` provides the data, presumably we can use that to build these tests

AlCalzone, 2021-04-07:
@raman325 my idea was that you can use the driver with a mocked serialport, like I do for end-to-end tests and send it the serial data that is received by devices. Enabling this will require a bit of work because right now that setup does not allow you to start from cache.

Basically the idea would then be:
- Create a driver instance with a mock serialport ✔
- Initialize that driver instance with a test network (one cached node) 🛠
- Send the driver data, handle/test events on your end ✔

raman325, 2021-04-07:
that's helpful, thanks! Full disclosure, I have been using this project to learn TypeScript, and tests are something I haven't worked with yet. Everything you said makes sense, next step for me is to figure out how to implement it 🙂

AlCalzone, 2021-04-07:
The tricky part is here:
https://github.com/zwave-js/node-zwave-js/blob/68294d7406bf6603b058ffeab1c1dc4949a5a7d3/packages/zwave-js/src/lib/driver/Driver.ts#L853-L863
skips the entire communication with the controller during unit tests. 
However that call is also responsible for populating the value DB from cache etc, so it will need to be refactored a bit.

AlCalzone, 2023-02-01:
This is actually supported since v10 and actively used in some integration tests I couldn't have done without it.

closed 2023-02-01 by AlCalzone