zwave-js/zwave-js #2261: Popp Z-Rain (POPE700168) sensor does not show whether it is currently raining or not

pascalmattes, 2021-04-07:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

The Popp Z-Rain sensor does not show whether it is currently raining or not. Therefore the device sends commands to the association group 2. More information can be found here: 
http://manuals-backend.z-wave.info/make.php?lang=EN&sku=POPE700168

**Device information**

Popp Z-Rain (POPE700168)
Node ID: 14

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [x] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [ ] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [x] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version 2021.3.4
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to Popp Z-Rain Device
2. Check configuration and see that there is now indication for current rain

**Additional context**

**Logfile:**

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6269677/zwavejs2mqtt-store.zip)

AlCalzone, 2021-04-07:
I'm guessing you meant Node 13, which reported this value (rain rate) via its lifeline group - I'd say that means it is raining:
```
09:42:41.231 CNTRLR   [Node 013] [~] [Multilevel Sensor] Rain rate: 0 => 18.1           [Endpoint 0]
09:42:41.233 SERIAL » [ACK]                                                                   (0x06)
09:42:41.235 DRIVER « [Node 013] [REQ] [BridgeApplicationCommand]
                      └─[MultilevelSensorCCReport]
                          type:  Rain rate
                          scale: Millimeter/hour
                          value: 18.1
```
Groups 2 and 3 are used to **control** other devices when it rains or not.

pascalmattes, 2021-04-07:
In other controllers, a "switch" is generated, which indicates the status whether it is raining or not. This is already set the first time the rocker is tilted. The rain rate is set only after a certain time, because there must be waited how often the rocker tilts in a certain time to see how much it rains.

AlCalzone, 2021-04-07:
Okay, that makes some sense - although Popp's choice to use Basic Set commands instead of a Binary Sensor to distinguish rain/no rain is suboptimal.
I'm hesitant to add too many associations by default, but you can add the controller to the node's groups 2 and 3 manually. Can you test that?
We'll probably have to edit the config file for the sensor to avoid hiding the Basic CC values though.

pascalmattes, 2021-04-07:
I have added the controller to group 2 and 3. Now I see following value in the config ui when tilting the rocker.

[image]

AlCalzone, 2021-04-07:
Does that stay after a restart?

pascalmattes, 2021-04-07:
Nope, after restart it is gone 🙃

AlCalzone, 2021-04-07:
Ok then we need to edit the config file.

pascalmattes, 2021-04-07:
Would be great. Is there any chance to get this into home assistant? (I have no idea how this works at the moment so sorry for that silly question)

AlCalzone, 2021-04-07:
You could probably map the file as a separate volume into the container - or wait for the next update.

PR #2346 (2021-04-14): fix(config): preserve Basic CC for Popp rain sensor
fixes: #2261
closed 2021-04-14 by AlCalzone