zwave-js/zwave-js #2264: [bug] Fibaro FGS214 Smart Module not detected and not working

ldymek, 2021-04-07:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt: 3.1.0
zwave-js: 7.1.0

### Describe the bug

a Fibaro FGS214 Smart Module can't be properly detected, only switch is visible and toggle it does not work. Can't find hardware id of switch in zwavejs2mqtt log nor in nodes dump:

```
"deviceId": "NaN-NaN-NaN",
        "manufacturer": "Unknown manufacturer 0xXXXX",
```
[nodes_dump.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6266843/nodes_dump.zip)

[nodes_dump.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6266845/nodes_dump.zip)

### To Reproduce

Steps to reproduce the behavior:

1. Pair FGS214
2. Device discovered as Unknown device unknown manufacturer
### Expected behavior

Device configured as in fgs214.json file

### Additional context

Tried to manually add it, without success (modified index.json pointing fgs214.json [this entry is missing]):

```
        {
                "manufacturerId": "0x010f",
                "manufacturer": "Fibargroup",
                "label": "FGS214",
                "productType": "0x0404",
                "productId": "0x1000",
                "firmwareVersion": {
                        "min": "0.0",
                        "max": "255.255"
                },
                "filename": "0x010f/fgs214.json"
        },
```

without success

AlCalzone, 2021-04-07:
@ldymek try editing the file the index points to, not the index itself

ldymek, 2021-04-08:
> @ldymek try editing the file the index points to, not the index itself.
> 
> FYI, the missing switch and toggle is unlikely to have anything to do with the missing/incorrect config file. For newer devices like this one, having a config is mostly cosmetical plus some UX improvements: https://zwave-js.github.io/node-zwave-js/#/config-files/overview

I've added entry to index.json as it is missing in current version - but fgs214.json is present. I compared the entry points in fgs214.json with OZW xml file - the only difference is in OZW product id equals to 0x4000, after change it to 0x1000 device is recognised and working properly (at OZW).

At zwave-js the entries are correct but can't force this to work - both in zwavejs and zwavejs2mqtt

blhoward2, 2021-04-08:
Index.json is built on startup. It doesn't work that way.

AlCalzone, 2021-04-08:
@ldymek there is a reason the index.json starts with this comment:
```
// This file is auto-generated. DO NOT edit it by hand if you don't know what you're doing!
```

Edit `fgs214.json` and add the missing productType/Id combination there. If you confirm that it worked, we can update the file in the repo.

ldymek, 2021-04-13:
I've updated zwavejs2mqtt to:

zwavejs2mqtt: 3.2.1
zwave-js: 7.2.2

and device is now properly recognised. I didn't touched any of json config files.

`{
  "id": 5,
  "deviceId": "271-4096-1028",
  "manufacturer": "FIBARO System",
  "manufacturerId": 271,
  "productType": 1028,
  "productId": 4096,
  "name": "De_Dietrich",
  "loc": "Boiler",
  "values": [
    {
      "id": "5-37-0-currentValue",
      "nodeId": 5,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "currentValue",
      "propertyName": "currentValue",
      "type": "boolean",
      "readable": true,
      "writeable": false,
      "label": "Current value",
      "stateless": false,
      "list": false,
      "value": true,
      "isCurrentValue": true,
      "targetValue": "37-0-targetValue",
      "lastUpdate": 1618304345577,
      "newValue": true
    },
    {
      "id": "5-37-0-targetValue",
      "nodeId": 5,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "boolean",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "list": false,
      "value": true,
      "lastUpdate": 1618304345579,
      "newValue": true
    },
    {
      "id": "5-91-0-slowRefresh",
      "nodeId": 5,
      "commandClass": 91,
      "commandClassName": "Central Scene",
      "endpoint": 0,
      "property": "slowRefresh",
      "propertyName": "slowRefresh",
      "type": "boolean",
      "readable": true,
      "writeable": true,
      "description": "When this is true, KeyHeldDown notifications are sent every 55s. When this is false, the notifications are sent every 200ms.",
      "label": "Send held down notifications at a slow rate",
      "stateless": false,
      "list": false,
[... 464 lines cut ...]
  "isListening": true,
  "inited": true,
  "hexId": "0x010f-0x0404-0x1000",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x010f:0x0404:0x1000:5.0",
  "productLabel": "FGS214",
  "productDescription": "Smart Module",
  "protocolVersion": 3,
  "zwavePlusVersion": 1,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 5,
  "nodeType": 1,
  "endpointsCount": 0,
  "isControllerNode": false,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 16,
    "specific": 1
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "De_Dietrich (Boiler)"
}`

Thank you for your quick help!

closed 2021-04-13 by ldymek