zwave-js/zwave-js #2267: Fibaro FGS221 local button 1 press not updated to controller

raymonvdm, 2021-04-07:
Today i made the update from version 1.0.0 to version zwavejs2mqtt: 3.1.0 (zwave-js: 7.1.0) but now i`m seem to be having issues with Controller / Lifeline updates on at least Fibaro FGS221 and FGS222

Button 1 is giving me absolutly radio silence

Button 2 is working when i press the local button
```
2021-04-07 00:14:01.416 INFO ZWAVE: Node 4: value updated: 37-2-currentValue true => false
2021-04-07 00:14:03.437 INFO ZWAVE: Node 4: value updated: 37-2-currentValue false => true
```
The issue occurs on at least 3 modules (which are in my reach to test) as soon as i start version 1.0.0 the Button 1 notify to the controller is working again.

AlCalzone, 2021-04-07:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

davyrolink, 2021-04-09:
I have the same issue with a FGS222, here are some logs of the specific node when the button was pressed:
```
2021-04-09T12:15:15.793Z SERIAL « 0x01090004000e032503ff26                                            (11 bytes)
2021-04-09T12:15:15.796Z CNTRLR   [Node 014] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2021-04-09T12:15:15.797Z CNTRLR   [Node 014] [+] [Binary Switch] targetValue: undefined             [Endpoint 0]
2021-04-09T12:15:15.798Z SERIAL » [ACK]                                                                   (0x06)
2021-04-09T12:15:15.801Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: true
2021-04-09T12:15:15.904Z SERIAL « 0x01090004000e032503ff26                                            (11 bytes)
2021-04-09T12:15:15.906Z CNTRLR   [Node 014] [~] [Binary Switch] currentValue: true => true         [Endpoint 0]
2021-04-09T12:15:15.907Z SERIAL » [ACK]                                                                   (0x06)
2021-04-09T12:15:15.910Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: true
2021-04-09T12:15:16.099Z SERIAL « 0x01160049840e10041001728670858e6025277a73ef256061                  (24 bytes)
2021-04-09T12:15:16.101Z SERIAL » [ACK]                                                                   (0x06)
2021-04-09T12:15:16.102Z DRIVER « [Node 014] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x840e10041001728670858e6025277a73ef2560
2021-04-09T12:15:16.106Z CNTRLR « [Node 014] Received updated node info
2021-04-09T12:15:17.905Z SERIAL « 0x01090004000e03250300d9                                            (11 bytes)
2021-04-09T12:15:17.909Z CNTRLR   [Node 014] [~] [Binary Switch] currentValue: true => false        [Endpoint 0]
2021-04-09T12:15:17.910Z SERIAL » [ACK]                                                                   (0x06)
2021-04-09T12:15:17.915Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
```

In home assistant the entity didn't show a "Turned on" event.

AlCalzone, 2021-04-09:
Sorry, but I need a little more context. Please re-interview the device (you can do that in the zwavejs2mqtt panel under "advanced") and post a complete log of that.

davyrolink, 2021-04-09:
@AlCalzone sorry, first time providing the log files..

Attached the logs of the re-interview and 2 button presses afterwards.
[zwavejs_356.log](https://github.com/zwave-js/node-zwave-js/files/6286524/zwavejs_356.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6286525/zwavejs2mqtt.log)

AlCalzone, 2021-04-09:
I mean it's nice that you try to keep the noise down but there's often a lot of context elsewhere in the logs for me :)

AlCalzone, 2021-04-09:
Ok the problem for FGS-222 is that the reports from endpoint 1 don't tell us they are from endpoint 1. I've added this case to https://github.com/zwave-js/node-zwave-js/issues/2286

I'll still need a log for FGS-221, but I guess that's the same.

fracarvic, 2021-04-11:
Upgraded to zwavejs2mqtt 3.2.1 (zwave-js 7.2.2) and all is working ok

AlCalzone, 2021-04-11:
Yeah, I've reverted the change that caused this for now.

closed 2021-04-12 by robertsLando

thekoma, 2021-12-05:
The problem has returned