zwave-js/zwave-js #2268: Aeotec ZW141 Shutter Control not recognized

embcla, 2021-04-07:
### 3.0.3 via HA addon 0.9.1

**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - npm install - npm run build )

zwavejs2mqtt version: 3.0.3 - 0.9.1
zwave-js version: <version here, you can find it on control panel>

### Describe the bug
Device ZW141 by Aeotec is not recognized by this, while HA's zwavejs does see it enough to at least recognize it as a shutter control.

### To Reproduce

Steps to reproduce the behavior:

1. Get a ZW141
2. Add it to your network
3. it will show as _unknown manufacturer 0x371 / unknown product 0x008d_
### Expected behavior

That it is properly recognized and enabled

### Additional context
Debug log below
```
{
  "id": 12,
  "deviceId": "881-141-3",
  "manufacturer": "Unknown manufacturer 0x0371",
  "manufacturerId": 881,
  "productType": 3,
  "productId": 141,
  "name": "Tapparella",
  "loc": "Cucina",
  "values": [
    {
      "id": "12-38-0-targetValue",
      "nodeId": 12,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "min": 0,
      "max": 99,
      "list": false,
      "value": 99,
      "lastUpdate": 1617707262428,
      "newValue": 99
    },
    {
      "id": "12-38-0-duration",
      "nodeId": 12,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
[... 534 lines cut ...]
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "hexId": "0x0371-0x0003-0x008d",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0371:0x0003:0x008d:0.0",
  "productLabel": "Unknown product 0x008d",
  "productDescription": "0x0003",
  "protocolVersion": 3,
  "zwavePlusVersion": 1,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 5,
  "nodeType": 1,
  "endpointsCount": 0,
  "isControllerNode": false,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 17,
    "specific": 7
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "Tapparella (Cucina)"
}
```

AlCalzone, 2021-04-07:
@blhoward2 We have this one under manufacturer `0x0086`, but not under `0x0371`.
https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x008d:0.0

Is that on your list for the next batch?

blhoward2, 2021-04-07:
Yes. It exists under both

embcla, 2021-04-07:
If you need anything, I can provide. Just let me know.
Thanks for working on this! :)

embcla, 2021-04-18:
Hey guys, I updated to v 7.2.4/3.4.0 and now I do see an xml definition for this device, but the driver is still not picking it up.
```
<Manufacturer id="0371" name="Aeotec Limited">
    <Product config="aeotec/zw141.xml" id="008d" name="ZW141 Nano Shutter" type="0003"/>
    <Product config="aeotec/zw141.xml" id="008d" name="ZW141 Nano Shutter" type="0103"/>
    <Product config="aeotec/zw141.xml" id="008d" name="ZW141 Nano Shutter" type="0203"/>
```
Do I have to do something for it to be picked up?

blhoward2, 2021-04-18:
That's under 0x0371. The file I recently uploaded is for an older version under 0x0086. The 0x0371 is still being worked on.

blhoward2, 2021-04-18:
That's from ozw? And it's for a different version of the device under 0x0371.

embcla, 2021-04-19:
I think that's from zwavw2mqtt, when the config file allowed defining a folder in the config folder where to save everything.

Just wanted to know how to track this, so I don't needlessly try to catch someone's attention :)

PR #2459 (2021-05-05): fix(config): cleanup and template Aeotec configurations (part 3)
Closes #1905. Closes #2268. Closes #2377.
closed 2021-05-05 by AlCalzone