zwave-js/zwave-js #2270: Thermostat Fan Mode CC Value no longer has a state value

raman325, 2021-04-07:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Thermostat Fan Mode values get discovered, but they have no state value. See below for an example:

```
{
  "endpoint": 0,
  "commandClass": 68,
  "commandClassName": "Thermostat Fan Mode",
  "property": "mode",
  "propertyName": "mode",
  "ccVersion": 1,
  "metadata": {
    "type": "number",
    "readable": true,
    "writeable": true,
    "label": "Thermostat fan mode",
    "min": 0,
    "max": 255,
    "states": {
      "0": "Auto low",
      "1": "Low"
    }
  }
}
```

**Device information**

Which device(s) is/are affected (make/model)?  Radio Thermostat CT100, Radio Thermostat CT101 (CT101 is user reported, my device is a CT100 Plus)
What are the node IDs? 26

**Did you change anything?**
- [ ] **Yes**: *(please describe)*
- [x] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *(specify application with versions)*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version 2021.4 (zwavejs2mqtt 3.2.0, zwave-js 7.1.1)
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Interview a Radio Thermostat thermostat
2. Observe that the thermostat has no state value for the discovered Thermostat Fan Mode CC Value

**Additional context**

Add any other context about the problem here.

**Logfile:**

[zwavejs_359.log](https://github.com/zwave-js/node-zwave-js/files/6274707/zwavejs_359.log)

AlCalzone, 2021-04-07:
Fix at #2272 - was probably noticed because of the mapping changes. The interview never worked correctly.

raman325, 2021-04-07:
that may explain the issues we were seeing with trying to set fan mode in HA. Thanks for the quick fix!

PR #2272 (2021-04-08): fix: use correct CC in Thermostat Fan Mode API
fixes: #2270
closed 2021-04-08 by AlCalzone