zwave-js/zwave-js #2282: [bug] HUSBZB-1 reported as Smart Temperature & Humidity Sensor

darootler, 2021-04-08:
### Version

zwavejs2mqtt version: 3.2.0
zwave-js version: 7.1.1
### Describe the bug

My z-wave stick is reported as [Smart Temperature & Humidity Sensor](https://devices.zwave-js.io/?jumpTo=0x0000:0x0003:0x0008:0.0)

### Expected behavior

Z-Wave stick reported as [HUSBZB-1](https://devices.zwave-js.io/?jumpTo=0x0000:0x0001:0x0001:0.0)

### Additional context

debug info:

```
{
  "id": 1,
  "deviceId": "0-8-3",
  "manufacturer": "Sigma Designs (Former Zensys)",
  "manufacturerId": 0,
  "productType": 3,
  "productId": 8,
  "name": "Büro Z-Wave USB Stick",
  "loc": "Büro",
  "values": [
    {
      "id": "1-114-0-manufacturerId",
      "nodeId": 1,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "manufacturerId",
      "propertyName": "manufacturerId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Manufacturer ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 0,
      "lastUpdate": 1617818136831,
      "newValue": 0
    },
    {
      "id": "1-114-0-productType",
      "nodeId": 1,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productType",
      "propertyName": "productType",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product type",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 3,
      "lastUpdate": 1617818136840,
      "newValue": 3
    },
    {
      "id": "1-114-0-productId",
      "nodeId": 1,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productId",
      "propertyName": "productId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false,
      "value": 8,
      "lastUpdate": 1617818136843,
      "newValue": 8
    },
    {
      "id": "1-134-0-libraryType",
      "nodeId": 1,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "libraryType",
      "propertyName": "libraryType",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Library type",
      "stateless": false,
      "list": false,
      "lastUpdate": 1617818136845
    },
    {
      "id": "1-134-0-protocolVersion",
      "nodeId": 1,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "protocolVersion",
      "propertyName": "protocolVersion",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Z-Wave protocol version",
      "stateless": false,
      "list": false,
      "lastUpdate": 1617818136848
    },
    {
      "id": "1-134-0-firmwareVersions",
      "nodeId": 1,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "firmwareVersions",
      "propertyName": "firmwareVersions",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Z-Wave chip firmware versions",
      "stateless": false,
      "list": false,
      "lastUpdate": 1617818136850
    }
  ],
  "groups": [],
  "neighbors": [
    11,
    12,
    17,
    18,
    19,
    2,
    20,
    21,
    23,
    28,
    4,
    6,
    7,
    8
  ],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "lastActive": 1617818136852,
  "supportsBeaming": true,
  "supportsSecurity": false,
  "isSecure": false,
  "keepAwake": false,
  "maxBaudRate": null,
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "hexId": "0x0000-0x0003-0x0008",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0000:0x0003:0x0008:0.0",
  "productLabel": "M417_9E",
  "productDescription": "Smart Temperature & Humidity Sensor",
  "protocolVersion": 3,
  "nodeType": 0,
  "endpointsCount": 0,
  "isControllerNode": true,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 2,
    "generic": 2,
    "specific": 7
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "Büro Z-Wave USB Stick (Büro)"
}
```

Here are two screenshots from the z-wave stick:

[image]
[image]

Let me know if more information is needed.

Regards
Richard

AlCalzone, 2021-04-08:
@robertsLando supported CCs have nothing to do with the device label :)

robertsLando, 2021-04-09:
Oh sorry I thought he means that the device was reporting humidity and temperature 😆 Didn't see the wrong label in the export

AlCalzone, 2021-04-23:
@darootler This one is really weird... The stick you shared photos of is not a HUSBZB-1, but a Silicon labs development stick - an "ACC-UZB3-E-STA" which is the UZB-3 stick (500-series) with an EU (868.42MHz) SAW filter, factory-flashed with the Static Controller library.
That one normally identifies itself as `0x0000-0x0001-0x0001`, which coincidentally is the same ID the HUSBZB-1 uses, because Nortek forgot to change it before putting their devices on the market.

There might be a couple of reasons which can all be related back to the device being a development tool, not an official consumer product. That information can be overwritten, but the necessary API call isn't implemented in zwave-js yet.

TL;DR: Not an incorrect config file, but a messed-up stick.

closed 2021-04-23 by AlCalzone