zwave-js/zwave-js #2285: [bug] FGK101 binary sensor stopped updating

paszczaq, 2021-04-09:
**Checklist:**

- [x] I am using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I am using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Fibaro Door/Window FGK101 (v.2.5) sensor doesnt change binary sensor status after update of the Z WaveJS to MQTT
Last working addon version 0.9.0
Non Working after update to 0.9.1 and 0.11.0
Currently non working:

zwavejs2mqtt: 3.2.0
zwave-js: 7.1.1

**Device information**
Device: Fibaro Door/Window FGK101 (v.2.5) 
Device ID: 271-4096-1792 (0x010f-0x0700-0x1000)

Affected Nodes ID: 18, 20, 21, 22

**Did you change anything?**
- [x] **Yes**: *Updating ZWaveJS2MQTT addon from 0.9.0 to 0.11.0*
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *0.9.0 version of zwavejs2mqtt addon
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x] `HomeAssistant` version core-2021.4.1 (Home Assistant OS 5.13)
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Update zwavejs2mqtt addon to ver. 0.11
2. Change binary sensor state (eg. open window)
3. No change in HA / ZwaveJS2MQTT panel and devices info - binary sensor doesnt change it state

**Additional context**

Add any other context about the problem here.

**Logfile:**

[zwavejs_356.zip](https://github.com/zwave-js/node-zwave-js/files/6285600/zwavejs_356.zip)

[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6285611/zwave_js_dump.zip)

AlCalzone, 2021-04-09:
Just to verify: Multilevel Sensor still works? At least the logs look like it.

paszczaq, 2021-04-09:
Yes, the Multilevel Sensor works - I have temperature sensor connected there and its working for those 4 devices. Example
node 22: 
[image]
You can see temperature drops when the door was opened :) - windy weather today :)

AlCalzone, 2021-04-09:
Thanks. Unfortunately, this is a symptom of a much larger issue for which we only had semi-working workarounds so far. Every time we tried to fix something that didn't work, we broke something else. You're going to have to wait for https://github.com/zwave-js/node-zwave-js/issues/2286 to be resolved.

AlCalzone, 2021-04-09:
Duplicate of #2286

closed 2021-04-09 by AlCalzone

capjay1, 2021-04-09:
Just want to add that the "Vision Security Motion Sensor (ZP3102)" exhibits similar behavior to the Fibaro Door/Window FGK101. Thanks.

AlCalzone, 2021-04-09:
FYI, I've rolled the changes back until #2286 is solved. Should be out later in version 7.2.1

bazmk2, 2021-04-17:
Also just to add that I think Everspring SM103 widow/door sensor exhibits the same behaviour.  Happy ro send more detail if needed

AlCalzone, 2021-04-18:
@bazmk2 please open a new issue and post a log where you actuate the device manually.