zwave-js/zwave-js #2290: Philio Tech PAN04-1B switch not reporting status in ZwaveJS

quizzical86, 2021-04-09:
@AlCalzone already explained on discord the input he needed for this issue. I'm now putting this same information in Github as I  assume it is more appropriate to log it here.

I have migrated a Philio Tech PAN04-1B ([link to zwave JS config DB](https://devices.zwave-js.io/?jumpTo=0x013c:0x0001:0x0003:0.0);  [link to manual](https://www.zwavetaiwan.com.tw/s/PAN04-manual-20181015-1.pdf)) from my old home automation solution to Home Assistant and ZwaveJS.

Trouble is that in Z-waveJS it doesn't report it's status when I manually click the light switch (it does when I turn on/off the light in Home Assistant). 

@AlCalzone and [this forum post](https://community.home-assistant.io/t/philo-pan04-problem/189800/2?u=quizzical) already hinted to the need to manually add associations so that group 2 and 3 point at the controller.

I have set an association to NodeID_1 for Load1 and Load2 using ZwaveJS2MQTT but that didn't solve the problem.

With these changes, the logs show [these reports](https://paste.ubuntu.com/p/tcPRSNpCky/) being sent when physically turning off one of the lights and [these reports](https://paste.ubuntu.com/p/5RwsZQBxKG/) when physically turning on that same light.

Here is the [homeid.json file](https://paste.ubuntu.com/p/YVqsdVx7Tb/)  and here is the [homeid.values.jsonl file](https://paste.ubuntu.com/p/TFy2j24k3N/).

This is all the information @AlCalzone asked for (and I produced it using the ZwaveJS2MQTT addon) but here is also the JSON dump which I produced using the ZwaveJS addon:
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6288045/zwave_js_dump.zip)

AlCalzone, 2021-04-09:
Hmm - seems like the device doesn't want to tell us **which** load was switched. I think the only way to get the actual status is to poll the values for endpoint 1/2 status whenever the value on endpoint 0 is changed.

To get these endpoint 0 values to exist, you'll need to edit the config file. Take the one I attached, and mount it as `/???/node_modules/@zwave-js/config/config/devices/0x013c/pan04.json` into the container (I am not sure what the first part of the path is in HA).
Afterwards, re-interview the device.

[pan04.json.txt](https://github.com/zwave-js/node-zwave-js/files/6288646/pan04.json.txt)

quizzical86, 2021-04-10:
@AlCalzone thank you.
As zwavejs doesn't allow re-interviewing the node I installed this file also in zwavejs2mqtt.
I reinterviewed the node using zwavejs2mqtt.

The change you did to the file is:
```
	"compat": {
		"preserveRootApplicationCCValueIDs": true
	}
```

I understand from what you write that this should somehow give me new values, but I didn't see any changes in what's available in zwavejs2mqtt.

Not sure you need it but just in case I attach here some logs (I have described for each report what action I took in terms of turning on or off a light):
[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/6289530/zwavejs.log)

I also attach instructions in case ever usefull for someone else how to get new device config files into zwavejs or zwavejs2mqtt (for me this was quite challenging):
[instructions to install zwave device config file into zwavejs or zwavejs2mqtt addon.txt](https://github.com/zwave-js/node-zwave-js/files/6289533/instructions.to.install.zwave.device.config.file.into.zwavejs.or.zwavejs2mqtt.addon.txt)

quizzical86, 2021-04-10:
I'm wondering whether the fact that it doesn't say which load was switched isn't related to this configuration parameter described in the manual:
[image]
And then there's also this configuration parameter:
[image]

AlCalzone, 2021-04-10:
Ohhh. I missed the last parameter there. That looks good, can you select the multi channel option?

quizzical86, 2021-04-10:
When we discussed yesterday on Discord I mentioned that z-wavejs and z-wavejs2mqtt do not have the configuration parameter for this set-up. 
I've tried to set this manually in home assistant in de developers tools panel using zwave_js.set_config_parameter with parameter 31, value 2 and bitmask 1, but I' not sure whether that is the correct way of doing it.

AlCalzone, 2021-04-10:
Leave out the bitmask, just set parameter 31 to value 2

quizzical86, 2021-04-10:
When I launch the following service call:
```
service: zwave_js.set_config_parameter
target:
  entity_id: switch.licht_kelder_beide
data:
  parameter: '31'
  value: '2'
```
The log shows the error:
"Unexpected error for call_service at pos 1: Configuration parameter with value ID 22-112-0-31 could not be found"

AlCalzone, 2021-04-10:
Not entirely sure about HA. You could try defining the new config parameter in your mounted file and try with that.

quizzical86, 2021-04-10:
OK I have guessed a bit how to do that and confirm that with this file it's working (also power consumption).
[pan04NEW.json.txt](https://github.com/zwave-js/node-zwave-js/files/6289927/pan04NEW.json.txt)
What is the next step now to get this fix in zwavejs for everyone?

AlCalzone, 2021-04-10:
@blhoward2 what do you think? Should we add something to our device files under metadata to mention known quirks?

I'm hesitant to change the configuration on users' devices by default, but we could at least mention it.

quizzical86, 2021-04-10:
This would for example also be usefull for the Aeotec Smart switch 6, where Basic CC is disabled by default. The result is that if a user pushes the little button on this power plug to turn it on, ZwaveJS won't get notified. I guess allmost all users will want to turn this on and mentioning it as a known quirk would be helpfull.

blhoward2, 2021-04-10:
What makes you believe this would apply to the Aeotec device? It doesn't have this parameter and we don't disable the Basic CC we map it usually. You should open your own issue so we can take a look.

quizzical86, 2021-04-10:
Perhaps I wasn't clear enough. I was just illustrating how AlCalzone's suggestion to have a place where users can read about known quirks can be usefull. This Philip Tech device clearly has a quirk that is worth flagging to the user, and I was trying to illustrate it is not the only case as the aeotec also has some quirks that are worth flagging to the user.

quizzical86, 2021-04-11:
OK well it was just a suggestion, in any case it is crystal clear that there's a crucial configuration parameter missing for this device and this configuration file adds it (https://github.com/zwave-js/node-zwave-js/files/6289927/pan04NEW.json.txt) so could we include this in zwaveJS so that at least the smart people (who can figure out they need this parameter without having read it in a 'known quirks' section) can use this z-wave device?

AlCalzone, 2021-04-11:
Yeah, that makes sense!

blhoward2, 2021-04-11:
Maybe I misread your post. I thought you meant to add the parameter to the Aeotec file.

AlCalzone, 2021-04-13:
@quizzical86 are you sure about the parameter number 11? The manual says this is parameter 31.
[image]

closed 2021-04-13 by AlCalzone

quizzical86, 2021-04-13:
No I'm not. This is a mistake on my side, it should indeed be parameter 31. However, what is very strange: I double checked and indeed I have configured it incorrectly with parameter 11 and yet this really solved my problem:

```
    {
      "id": "22-112-0-11",
      "nodeId": 22,
      "commandClass": 112,
      "commandClassName": "Configuration",
      "endpoint": 0,
      "property": 11,
      "propertyName": "Select Auto Report Type",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Select Auto Report Type",
      "default": 1,
      "stateless": false,
      "min": 0,
      "max": 255,
      "list": true,
      "allowManualEntry": false,
      "states": [
        {
          "text": "Basic type",
          "value": 1
        },
        {
          "text": "Multichannel type",
          "value": 2
        }
      ],
      "value": 2,
      "lastUpdate": 1618349712195,
      "newValue": 2
    },
```

I have no explanation how this is possible, this device is not even supposed to have a parameter 11.

AlCalzone, 2021-04-13:
WTF... Maybe it does have both parameters. Anyways - I've merged it with parameter 31. Once the update is available, can you please test if the device reacts to that param too?

quizzical86, 2021-04-13:
OK but my device which wasn't working is now working (with a non-existing parameter 11 set to "2").
Does that mean that to be able to test the new version you will be releasing I should:
1: remove the device from the network
2: factory reset it
3: add it back to the network and set the parameter 31 which you added
?

AlCalzone, 2021-04-14:
Yeah you can do that to make sure parameter 31 exists and is working.

quizzical86, 2021-04-14:
I removed the device from the network reset the device multiple times and then added it back.
Either my resets were unsuccesful or the reset does not restore the parameters to factory defaults, since some parameters I had set did not go back to default and in addition my device works perfectly which probably indicates the multichannel parameter is still set.
So I don't seem able to do tests whether using parameter 31 (as specified in the manual) also works; unless there's another way to undo the fact that I have changed the non-existing parameter 11.

quizzical86, 2021-04-14:
I knew I shouldn't have touched it while it was working perfectly :'-(
I wanted to test it for the good of others who use this device. So I downloaded the new file you created (the latest version is not yet available in home assistant) and installed it manually. Removed the z-wave device, did my best to reset it (not completely sure I succeeded but tried many many times) and then added it back.

So now the situation is that only Endpoint 3 (this is a virtual entity that controls both switches in the Z-wave device at the same time and can also show the sum of the wattage used by both lights) is updated.
So when I physically click the switch to turn on light 1 or light 2, this shows in this endpoint 3 (both in terms of the on/off status and the wattage) but does not show in Z-waveJS for the light in question.

Going back to the file I created (with the incorrect parameter 11, but which did solve my problem at the time) gives me the same result.

This is super frustrating as I had just gotten all my automations that are based on these lights working :-s
I have tried for many hours but can't find a way to get my device back to working condition :-(

AlCalzone, 2021-04-14:
What are your current configuration settings and associations?

quizzical86, 2021-04-14:
My heart skipped a beat I think when you posted this reply!
Because z-waveJS in home assistant doesn't support associations I had forgotten about that.
I fired up ZwaveJS2MQTT and set an association to NodeID_1 for Load1 and Load2.
Now it's working again :)
So if we add a 'known quirks' it should definitely include the need to set these associations for stupid people like me.

So I conclude that through some black magic this device accepts both parameter 11 and parameter 31 and the result is the same in both cases. 
And the more important conclusion is that your new configuration file for this device works well if the user subsequently implements the known quirks of changing the default value of parameter 31 and sets the associations for Load1 and Load2 (using ZwaveJS2MQTT).

AlCalzone, 2021-04-15:
> So I conclude that through some black magic this device accepts both parameter 11 and parameter 31 and the result is the same in both cases.

Might be that they realized their typo too late and subsequently changed the firmware to match 🤷🏻‍♂️

gratjetost, 2021-10-17:
Same Issue with the PAN06 switch from Philio. older version of the PAN04. Can this be fixed too? I cannot get it managed by myself.

blhoward2, 2021-10-17:
Don't comment on a closed issue. Open a new one and reference this issue number if need be.

PR #2337 (2021-04-13): fix(config): synchronize PAN04 configuration with manual
Pending confirmation from @quizzical86

fixes: #2290
(assuming the user sets parameter 31 correctly)