zwave-js/zwave-js #2300: RefreshAllNodes function

blhoward2, 2021-04-10:
Currently, you can refresh just the values of one node but there is no way to refresh all nodes. Because of recent changes to the startup procedure nodes don't seem to check the values on startup anymore (at least I think we used to do this). The result is that if the system crashes or goes down without saving the cached state, devices go out of sync with their actual state. There is no great way to fix this other than to force it to reinterview everything by savings the settings page. I'm unsure if you can even fix this in HA if you aren't running zwavejs2mqtt.

We should have a function to just refresh all values on all devices when necessary without reinterviewing, even though this function would generate a good deal of traffic. It's less traffic than reinterviewing everything.

/cc 
@robertsLando 
@raman325

raman325, 2021-04-10:
Maybe I am misunderstanding, but shouldn't we fix the root issue that requires a refresh of all nodes and all values instead of creating this workaround?

blhoward2, 2021-04-10:
Perhaps but I'm not sure that we can detect that the power cord was yanked out and that the states don't match. I'm also not sure it's a work around to be able to resync the whole network. I believe ozw had a similar function.

AlCalzone, 2021-04-10:
We have this for single nodes: https://zwave-js.github.io/node-zwave-js/#/api/node?id=refreshvalues

I wonder if it's worth adding another method that just loops through all nodes and calls this or if that is something applications can implement themselves.

robertsLando, 2021-04-11:
Humm we could do that just for config CC maybe?

AlCalzone, 2021-04-11:
That's https://zwave-js.github.io/node-zwave-js/#/api/node?id=refreshccvalues

blhoward2, 2021-04-11:
I swear I looked

robertsLando, 2021-04-12:
Not in the UI yet but there is already the mqtt support for it

AlCalzone, 2021-04-22:
I don't think there's a necessity to do this in the driver

closed 2021-04-22 by AlCalzone

blhoward2, 2021-04-22:
This may actually go hand-in-hand with the discussion on discord yesterday. The HA addons users don't have a way to invalidate the cache and a reinterview of all so if we added that I think it would supplant the need for this.

raman325, 2021-04-22:
Cache issues should be owned by the driver. If the cache needs to be invalidated, the driver should have the capability to do it, whether it's automatically or through a method. I'm not arguing that the capability is needed, but I don't think it makes sense for the HA addon or the HA integration to provide this capability directly