zwave-js/zwave-js #2319: Include wakeup instructions in config file metadata

cgarwood, 2021-04-11:
**Is your feature request related to a problem? Please describe.**
Config files contain instructions for inclusion/exclusion, but don't currently have an option to include wakeup instructions for battery devices.

**Describe the solution you'd like**
Add an additional `wakeup` key to the metadata portion of device configuration files, where wakeup instructions can be included. This will allow applications to show wakeup instructions if available for a battery node prior to performing a re-interview/refresh/etc.

blhoward2, 2021-04-14:
@AlCalzone further color on this. Some ZWA files include wakeup instructions but when I pulled them before they were terrible. A lot of "see manuals" and such. These would need to be hand added for the most part but it probably makes sense to include a place for people to do so.

AlCalzone, 2021-04-14:
> Or the UI could do that on its own where wakeup isn't defined.

Yeah, that's better IMO.

PR #2345 (2021-04-14): feat(config): parse wakeup instructions from device config
fixes: #2319
closed 2021-04-14 by AlCalzone