zwave-js/zwave-js #2325: sorry to ask this here... but could we do the same config change for https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0109/zd2102-5.json?

AlCalzone, 2021-04-12:
sorry to ask this here... but could we do the same config change for https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0109/zd2102-5.json?
it seems to have a similar issue

__Originally posted by @tofuSCHNITZEL in https://github.com/zwave-js/node-zwave-js/pull/2291#issuecomment-817619298__

tofuSCHNITZEL, 2021-04-12:
will try to test it with my home assistant setup if config change fixes or not.

tofuSCHNITZEL, 2021-04-12:
okay never mind... with version 7.1.1 it works as expected even without the enableBasicSetMapping as soon as 7.2.2 is available for ha users I will check if it still works

AlCalzone, 2021-04-12:
Ok let me know 👍🏻

tofuSCHNITZEL, 2021-04-13:
still working so you can go ahead and close this, sorry for the fuzz ;)

closed 2021-04-13 by AlCalzone