zwave-js/zwave-js #2328: Mssing parameters on device 0x0234-0x0003-0x0121

PoulKlemmensen, 2021-04-12:
Hi.

I have an stand-alone installation of zwavejs2mqtt and have several devices from Logic group.  On the ZDB5100 configuration file, parameters from 14-19 are missing for some reason?

Can you fix that or do I have to report this elsewhere?
Br.
Poul.

marcus-j-davies, 2021-04-12:
14-19 have not been setup.
https://devices.zwave-js.io/?jumpTo=0x0234:0x0003:0x0121:0.0

I would start by detailing what these parameters are (label, description, affect etc. etc.)
a link to their properties (i.e. a manual) will help.

or, you could try adding this in your self, and raising a PR.
https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide
https://zwave-js.github.io/node-zwave-js/#/config-files/file-format

PoulKlemmensen, 2021-04-12:
I have tried the guides mentioned above but could not get it to work. I have installed at separate version of your software and have altered the conf. files manually. 

The missing part of something like this : 
		"14": {
			"label": "Global Brightness.",
			"description": "Global broghtness.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 0,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"16": {
			"label": "Pushbutton 1 functionality",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 4,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Standard toggle-function",
					"value": 0
				},
				{
					"label": "Automatic turn off after the time has expired",
					"value": 1
				},
				{
					"label": "Automatic turn on after the time has expired",
					"value": 2
				},
				{
					"label": "Always turn off or dim down",
					"value": 3
				},
				{
					"label": "Always turn on or dim up",
					"value": 4
				}
			]
		},
		"17": {
			"label": "Timer value for pushbutton 1",
			"valueSize": 2,
			"minValue": 0,
			"maxValue": 43200,
			"defaultValue": 300,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"18": {
			"label": "Multi-Level Switch Set values for pushbutton 1.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 65535,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"19": {
			"label": "Binary Switch Set support for pushbutton 1",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 2,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": false,
			"options": [
				{
					"label": "control pushbutton indicator LEDs",
					"value": 0
				},
				{
					"label": "Controls LED's and state",
					"value": 1
				},
				{
					"label": "As activated the pushbutton",
					"value": 2
				}
			]
		},

PoulKlemmensen, 2021-04-12:
And if you are looking here, they are actually defined:
https://products.z-wavealliance.org/products/3399/configs

PoulKlemmensen, 2021-04-12:
I have asked before. Logic Group have also made a new device and I have created a config. file manually that works. Will you be able to create this one, or want does it take?

marcus-j-davies, 2021-04-12:
if you have a config, that you have written for a device that is currently not found on the [Device DB](https://devices.zwave-js.io) 
raise a PR to get it added to the main repo for distribution, it will then be apart of the standard config db, on a future release.
the PR will checked, and any changes/corrections required will be managed within the PR

blhoward2, 2021-04-12:
You asked like 5 hours ago? You can PR the change yourself ensuring it matches the style guide linked above, or we will eventually do it. It'll take some time.

PoulKlemmensen, 2021-04-12:
Thanks Marcus. Will try and do that.

And we are talking about two different requests. Device 0x0234-0x0003-0x0121 is missing parameters, Device 0x0234-0x0003-0x0125 is completely new.

marcus-j-davies, 2021-04-12:
As with most PR's, they are checked automatically (initially before being checked by someone here)  
one I thing I see a lot of is the titling of the PR not being to standards and the original author wondering why initial checks fail

so use **feat(config): add new device. some device Label** for the title/description.
the Z-Wave JS Bot may also comment (after the initial checks), he may suggest some things.

Roagert, 2021-04-12:
Pull Request is already there just needs review...
https://github.com/zwave-js/node-zwave-js/pull/2241

PoulKlemmensen, 2021-04-13:
Will this request also create a config. file for device : 0x0234-0x0003-0x0125 or do I create a new issue? I have a manual created config. file, but don't know how to include it and create a PR.

AlCalzone, 2021-04-13:
@PoulKlemmensen if you already have a device file, you could create a PR by uploading it:
[image]

Although it is probably better if you post it for @Roagert so he can incorporate it in his PR.

Roagert, 2021-04-13:
@PoulKlemmensen the parameters have been updated for this device, the parameters shouldnt be missing but you can verify the PR

AlCalzone, 2021-04-13:
@Roagert he was asking about another device which doesn't seem to be included in your PR yet:
> 0x0234-0x0003-0x0125

Roagert, 2021-04-13:
Ahh understood, i will check that.

Roagert, 2021-04-13:
@PoulKlemmensen Which devices is it that is missing ? Whats the product name ?

PoulKlemmensen, 2021-04-13:
Logic group, ZRB5120. Device ID : 0x0234-0x0003-0x0125. 
Like ZDB5100, but with two relays instead of dimmer. Lots of config. are similar between the two.

One example configuration could be : 

// Logic Group ZDB5100
// MATRIX Switch with Dimmer and Backlight
{
	"manufacturer": "Logic Group",
	"manufacturerId": "0x0234",
	"label": "ZRB5120",
	"description": "MATRIX Switch with dual relay and Backlight",
	"devices": [
		{
			"productType": "0x0003",
			"productId": "0x0125"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"supportsZWavePlus": true,
	"paramInformation": {
		"1[0x01]": {
			"label": "Pushbutton 1 controls the relay 1",
			"description": "Enabling this parameter lets pushbutton 1 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x02]": {
			"label": "Pushbutton 2 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 2 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x04]": {
			"label": "Pushbutton 3 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 3 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x08]": {
			"label": "Pushbutton 4 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 4 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"2[0x01]": {
			"label": "Pushbutton 1 controls the relay 2",
[... 799 lines cut ...]
		},
		"46": {
			"label": "ON indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 32512,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"47": {
			"label": "OFF indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 791621376,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		}
	}
}

Roagert, 2021-04-13:
@PoulKlemmensen Thanks, i thought it was this device. i will implement it later tonight. Should be fairly straight forward

AlCalzone, 2021-04-13:
@PoulKlemmensen please put code in code tags :)
[image]

PoulKlemmensen, 2021-04-13:
Thomas. Great and perfect !!

PoulKlemmensen, 2021-04-13:
`// Logic Group ZDB5100
// MATRIX Switch with Dimmer and Backlight
{
	"manufacturer": "Logic Group",
	"manufacturerId": "0x0234",
	"label": "ZRB5120",
	"description": "MATRIX Switch with dual relay and Backlight",
	"devices": [
		{
			"productType": "0x0003",
			"productId": "0x0125"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"supportsZWavePlus": true,
	"paramInformation": {
		"1[0x01]": {
			"label": "Pushbutton 1 controls the relay 1",
			"description": "Enabling this parameter lets pushbutton 1 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x02]": {
			"label": "Pushbutton 2 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 2 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x04]": {
			"label": "Pushbutton 3 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 3 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x08]": {
			"label": "Pushbutton 4 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 4 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"2[0x01]": {
			"label": "Pushbutton 1 controls the relay 2",
			"description": "Enabling this parameter lets pushbutton 1 control relay 2.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
[... 795 lines cut ...]
		"46": {
			"label": "ON indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 32512,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"47": {
			"label": "OFF indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 791621376,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		}
	}
}
`

AlCalzone, 2021-04-13:
Sorry, should have been clearer. Github is a bit weird:
1. Paste the code
2. select all of it
3. click the icon

This will create a code block instead of a really long inline code highlight.

PoulKlemmensen, 2021-04-13:
Better??. Sorry for that and THANKS for your patience

```
// Logic Group ZDB5100
// MATRIX Switch with Dimmer and Backlight
{
	"manufacturer": "Logic Group",
	"manufacturerId": "0x0234",
	"label": "ZRB5120",
	"description": "MATRIX Switch with dual relay and Backlight",
	"devices": [
		{
			"productType": "0x0003",
			"productId": "0x0125"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"supportsZWavePlus": true,
	"paramInformation": {
		"1[0x01]": {
			"label": "Pushbutton 1 controls the relay 1",
			"description": "Enabling this parameter lets pushbutton 1 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x02]": {
			"label": "Pushbutton 2 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 2 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x04]": {
			"label": "Pushbutton 3 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 3 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"1[0x08]": {
			"label": "Pushbutton 4 controls relay 1",
			"description": "Enabling this parameter lets pushbutton 4 control relay 1.",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"2[0x01]": {
			"label": "Pushbutton 1 controls the relay 2",
			"description": "Enabling this parameter lets pushbutton 1 control relay 2.",
			"valueSize": 1,
[... 799 lines cut ...]
			"label": "ON indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 32512,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		},
		"47": {
			"label": "OFF indication RGB-saturation for pushbutton 4.",
			"valueSize": 4,
			"minValue": 0,
			"maxValue": 4294967295,
			"defaultValue": 791621376,
			"unsigned": true,
			"readOnly": false,
			"writeOnly": false,
			"allowManualEntry": true
		}
	}
}

```

AlCalzone, 2021-04-13:
Much better

Roagert, 2021-04-13:
@PoulKlemmensen I am not able to se parameter 15 ? can you provide that ? and also the associations ?
i dont know if the "debug" tab works for this case ? 
[image]
and then look for "association"

PoulKlemmensen, 2021-04-13:
Parameter 15 is associations group security something like ZDB5100 (i assume, never used it).

Associations og groups (think that's what it is called in your GUI) is like ZDB5100 EXCEPT : 
ID 14 is Relay 1 basic report
ID 15 is Relay 2 basic report.

Roagert, 2021-04-13:
Can you share the text in that debug ?

PoulKlemmensen, 2021-04-13:
`{
  "id": 4,
  "deviceId": "564-293-3",
  "manufacturer": "Logic Group",
  "manufacturerId": 564,
  "productType": 3,
  "productId": 293,
  "name": "xxxx",
  "loc": "xxxx",
  "values": [
    {
      "id": "4-134-0-libraryType",
      "nodeId": 4,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "libraryType",
      "propertyName": "libraryType",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Library type",
      "stateless": false,
      "list": false,
      "value": 3,
      "lastUpdate": 1617949086080,
      "newValue": 3
    },
    {
      "id": "4-134-0-protocolVersion",
      "nodeId": 4,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "protocolVersion",
      "propertyName": "protocolVersion",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Z-Wave protocol version",
      "stateless": false,
      "list": false,
      "value": "6.4",
      "lastUpdate": 1617949086084,
      "newValue": "6.4"
    },
    {
      "id": "4-134-0-firmwareVersions",
      "nodeId": 4,
      "commandClass": 134,
      "commandClassName": "Version",
      "endpoint": 0,
      "property": "firmwareVersions",
      "propertyName": "firmwareVersions",
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Z-Wave chip firmware versions",
      "stateless": false,
      "list": false,
      "value": [
        "0.3"
      ],
      "lastUpdate": 1617949086089,
      "newValue": [
        "0.3"
      ]
    },
    {
      "id": "4-134-0-hardwareVersion",
[... 2785 lines cut ...]
  "isRouting": true,
  "isFrequentListening": false,
  "isListening": true,
  "inited": true,
  "hexId": "0x0234-0x0003-0x0125",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0234:0x0003:0x0125:0.0",
  "productLabel": "ZRB5120",
  "productDescription": "MATRIX Switch with dual relay and Backlight",
  "protocolVersion": 3,
  "zwavePlusVersion": 1,
  "zwavePlusNodeType": 0,
  "zwavePlusRoleType": 5,
  "nodeType": 1,
  "endpointsCount": 6,
  "isControllerNode": false,
  "dataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 24,
    "specific": 1
  },
  "status": "Alive",
  "interviewStage": "Complete",
  "_name": "zzzzzz"
}`

PoulKlemmensen, 2021-04-13:
Sorry for the format, it did a timeout twice.

PoulKlemmensen, 2021-04-13:
```
  "groups": [
    {
      "text": "Lifeline",
      "value": 1,
      "maxNodes": 5,
      "isLifeline": true,
      "multiChannel": true
    },
    {
      "text": "B1 Basic Report",
      "value": 2,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B1 Binary Set",
      "value": 3,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B1 Multilevel Set",
      "value": 4,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B2 Basic Report",
      "value": 5,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B2 Binary Set",
      "value": 6,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B2 Multilevel Set",
      "value": 7,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B3 Basic Report",
      "value": 8,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B3 Binary Set",
      "value": 9,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B3 Multilevel Set",
      "value": 10,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B4 Basic Report",
      "value": 11,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B4 Binary Set",
      "value": 12,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "B4 Multilevel Set",
      "value": 13,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "R1 Basic Report",
      "value": 14,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    },
    {
      "text": "R2 Basic Report",
      "value": 15,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": true
    }
  ],
```

Roagert, 2021-04-13:
So the first draft is implemented, i am missing the associations part for the device but i believe its correct.
I cannot find the manual from Logic Group so i cannot with 100% certainty implement the device. So if you have it i will gladly look it over and implement it @PoulKlemmensen

PoulKlemmensen, 2021-04-13:
The only documentation I have is the following. But I will be happy to try the configuration out in my installation.

`/*01*/ {"Relay1 key",          FORMAT_BIT_FIELD, SIZE_8BIT, 0, 0x0F, 0x01},     // bitfield of keys that controls the relay1 output // Btn 1 on/off
/*02*/ {"Relay2 key",          FORMAT_BIT_FIELD, SIZE_8BIT, 0, 0x0F, 0x02},     // bitfield of keys that controls the relay2 output // Btn 2 on/off
/*03*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0}, 
/*04*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},             
/*05*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},  
/*06*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0}, 
 
/*07*/ {"Central Scene",       FORMAT_ENUM, SIZE_8BIT, 0, 1, 1},                // Central Scene notifications
/*08*/ {"Double click option", FORMAT_ENUM, SIZE_8BIT, 0, 1, 1},                // Cleaning Mode
/*09*/ {"Not used",            FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 0, 255, 0},  // NOT USED
/*10*/ {"Enhanced color ctrl", FORMAT_ENUM, SIZE_8BIT, 0, 1, 0},                // Warm white field is used for enhanced color commands
/*11*/ {"Key debounce",        FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 5},
/*12*/ {"Press threshold",     FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 20},  // 20 == 200 milliseconds
/*13*/ {"Hold threshold",      FORMAT_UNSIGNED_INTEGER, SIZE_8BIT, 1, 255, 50},  // 50 == 500 milliseconds
/*14*/ {"Global brightness",   FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0xFFFFFF00},
/*15*/ {"AG security mode",    FORMAT_BIT_FIELD, SIZE_16BIT, 0, 0x1FFF, 0x1FFF},
 
/*16*/ {"Key 1 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*17*/ {"Key 1 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*18*/ {"Key 1 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*19*/ {"Key 1 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*20*/ {"LED 1 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*21*/ {"LED 1 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*22*/ {"LED 1 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*23*/ {"LED 1 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*24*/ {"Key 2 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*25*/ {"Key 2 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*26*/ {"Key 2 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*27*/ {"Key 2 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*28*/ {"LED 2 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*29*/ {"LED 2 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*30*/ {"LED 2 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*31*/ {"LED 2 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*32*/ {"Key 3 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*33*/ {"Key 3 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*34*/ {"Key 3 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*35*/ {"Key 3 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*36*/ {"LED 3 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*37*/ {"LED 3 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*38*/ {"LED 3 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*39*/ {"LED 3 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00},
 
/*40*/ {"Key 4 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 4, 0},
/*41*/ {"Key 4 timer",         FORMAT_UNSIGNED_INTEGER, SIZE_16BIT, 0, 43200, 300},
/*42*/ {"Key 4 value",         FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0x01FFFF00, 0x01FF0000},
/*43*/ {"Key 4 control",       FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*44*/ {"LED 4 mode",          FORMAT_ENUM, SIZE_8BIT, 0, 10, 7},
/*45*/ {"LED 4 color control", FORMAT_ENUM, SIZE_8BIT, 0, 2, 0},
/*46*/ {"LED 4 color on",      FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x00007F00},
/*47*/ {"LED 4 color off",     FORMAT_UNSIGNED_INTEGER, SIZE_32BIT, 0, 0xFFFFFFFF, 0x0F0F0F00}`

Roagert, 2021-04-13:
@PoulKlemmensen Please have a look in the PR i have uploaded the json file, if you want to try it out

PoulKlemmensen, 2021-04-13:
@Roagert. I have tried to implement the two files, ZDB5100.json and ZRB5120.json and they are working as expected. Thanks  for your nice work !!!

Regarding ZRB5120, you seems to be missing parameter 15, but I can understand that it is difficult without proper documentation.

The work done by dividing the parameters in parts, e.g. parameter 3-112-0-1 in bit 1, 2, 4 and 8 is great and makes configuration much more intuitive. But finding parameters is not easy when they are sorted in the way they are. 

I am sure, it has been discussed heavily, but what about sorting it by 3-112-0-1, 3-112-0-2, e.g. by parametername, meaning that parameters 1 - 47 is placed in sequence??

Roagert, 2021-04-13:
Is there a flaw in the ZDB5100?

Also yes pretty difficult but i guess its the same for parameter 15 i Can see the byte size is the same

PoulKlemmensen, 2021-04-13:
@Roagert 
With regards to ZRB5120 and parameter 15 compared to ZDB5100. There might be a change as the ZRB contains two relays instead of one dimmer. Wouldn't that mean, that there should be a parameter 8192 for association group 15 and then parameter 8191 is something different? E.g. 12.287 ??

With regards to ZDB5100, it seems to work and I have not found any issues. I don't think that the default values match, but I have not checkked.

Roagert, 2021-04-13:
Yes ofcourse there might be but it looks like it has the same byte size as the ZDB5100 but i would need the manual for me to update it.

@PoulKlemmensen Okay if you dont think the default values matches please let me know maybe i missed some from the user manual.

blhoward2, 2021-04-13:
> I am sure, it has been discussed heavily, but what about sorting it by 3-112-0-1, 3-112-0-2, e.g. by parametername, meaning that parameters 1 - 47 is placed in sequence??

We have no control over this. That's a decision each UI makes.

PR #2241 (2021-04-27): feat(config): template logic group configuration files
fixes: #2328
closed 2021-04-27 by AlCalzone