zwave-js/zwave-js #2334: [bug]Zooz Zen20 powerstrip not being queried

faspina, 2021-04-13:
### Version

zwavejs/zwavejs2mqtt:latest docker 

Log error for the Zen20 zooz powerstrip
2021-04-13T15:38:35.139Z CNTRLR [Node 056] Timed out while waiting for a response from the node
2021-04-13T15:38:35.154Z CNTRLR [Node 056] Interviewing Multi Channel... [Endpoint 1]
2021-04-13T15:38:35.155Z CNTRLR » [Node 056] Querying endpoint count for CommandClass Binary Switch...
2021-04-13T15:38:35.155Z CNTRLR [Node 056] Error during node interview: This virtual node (Endpoint 1) does not support the command GetV1!

Build/Run method

- [X ] Docker

zwavejs2mqtt version: 3.2.1
zwave-js version: 7.2.2

### Describe the bug

moveng to ZwaveJS all device queried except the Zoos Zen20 I get the error above in the log.

AlCalzone, 2021-04-13:
@faspina I need the `zwave-js` logfile, not the one from `zwavejs2mqtt`

faspina, 2021-04-13:
[zwavejs_1.log](https://github.com/zwave-js/node-zwave-js/files/6305157/zwavejs_1.log)

AlCalzone, 2021-04-13:
Ok thanks, I think I found it.

closed 2021-04-13 by AlCalzone