zwave-js/zwave-js #2338: PIR Motion sensor returns values of 0 or 255

Dave-Casey, 2021-04-14:
5. Also make sure to provide the necessary information, as described here:
I wanted to make sure I've provided the necessary information however this link appears to be broken:  
https://zwave-js.github.io/node-zwave-js/#/development/troubleshooting?id=providing-the-necessary-information-for-an-issue

[zwave_js_dump (1).json.zip](https://github.com/zwave-js/node-zwave-js/files/6307312/zwave_js_dump.1.json.zip)

**Checklist:**

- [x ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
- [x ] I have checked the troubleshooting section and my problem is **not** described there.
- [x ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

ZP3102 by Vision Security (PIR sold by Monoprice) - Motion is no longer a bool on/off or motion / no motion.  It is identified as a “basic” sensor and returns the value 0 or 255.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Did you change anything?**
- [x ] **Yes**: *updated to ZwaveJS to V0.1.18 *
- [ ] **No**

**Did this use to work before?**
- [ ] **Don't know**, this is a new device
- [ ] **No**, it never worked anywhere
- [x] **Yes**, in: *addon_core_zwave_js_0.1.13*
    
**How are you using `node-zwave-js`**
- [ ] `zwavejs2mqtt` (latest) docker image
- [ ] `zwavejs2mqtt` (dev) docker image
- [ ] `zwavejs2mqtt` Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] ioBroker.zwave2 adapter
- [x ] `HomeAssistant` Home Assistant OS 5.13
- [ ] Pkg
- [ ] Manual Docker build
    - `node-zwave-js` branch: 
    - `zwavejs2mqtt` branch: 
- [ ] `node-red-contrib-zwave-js`
    - Module version (double click node): 
- [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github)) 
- [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

[core_zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/6307326/core_zwave_js.log)

Photo64, 2021-04-14:
I am having a similar issue with the Vision Security Door/Window sensors. Prior to upgrading the binary_sensor returned on/off. Now it doesn't change, it is stuck at either on or off. The basic sensor now returns 0 or 255. Only difference is I am using Zwave-js2mqtt v0.11.0
Device Info:
ZD2102
by Vision Security
Firmware: 4.84 
Upon triggering open/close this is what is returned in the log
2021-04-13 15:47:25.456 INFO ZWAVE: Node 65: value updated: 32-0-currentValue 0 => 255
2021-04-13 15:47:25.475 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Sensor status 2 => 2
2021-04-13 15:47:38.816 INFO ZWAVE: Node 65: value updated: 32-0-currentValue 255 => 0
2021-04-13 15:47:38.836 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Sensor status 2 => 2

blhoward2, 2021-04-14:
In both cases the driver is just passing on the values. We don't setup the device entity so your issue is with the integration or zwavejs2mqtt depending on which you're using for discovery.

blhoward2, 2021-04-14:
I take back my first post. These both describe basic sets coming in, which is odd for this sort of device. Can you try reinterviewing them?

AlCalzone, 2021-04-14:
Didn't check out the log yet but I think they're sending Basic Set commands which we're no longer automatically mapping to a different CC. Instead they are treated as Basic Reports.

AlCalzone, 2021-04-14:
> 2021-04-13T23:50:07.617Z CNTRLR   [Node 071] treating BasicCC::Set as a report

Yep, that's it. @blhoward2 we need the `enableBasicSetMapping` compat flag for these devices now.

Photo64, 2021-04-14:
Just fyi, I have tried re-interviewing them but the binary_sensor never updates

AlCalzone, 2021-04-14:
Any chance you kept a log of that re-interview?

Photo64, 2021-04-14:
I just did another, here are the logs
2021-04-14 10:58:52.680 INFO ZWAVE: Node 65: interview started
2021-04-14 10:58:52.694 INFO ZWAVE: Success zwave api call refreshInfo { success: true, message: 'Success zwave api call', result: undefined }
2021-04-14 10:58:52.705 INFO ZWAVE: Node 65: interview stage PROTOCOLINFO completed
2021-04-14 10:58:53.644 INFO ZWAVE: Node 65: value added: 113-0-Home Security-Cover status => 3
2021-04-14 10:58:54.444 INFO ZWAVE: Node 65 is now awake
2021-04-14 10:58:54.456 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Cover status 3 => 3
2021-04-14 10:58:54.593 INFO ZWAVE: Node 65: interview stage NODEINFO completed
2021-04-14 10:58:54.666 INFO ZWAVE: Node 65: value added: 114-0-manufacturerId => 265
2021-04-14 10:58:54.667 INFO ZWAVE: Node 65: value added: 114-0-productType => 8193
2021-04-14 10:58:54.668 INFO ZWAVE: Node 65: value added: 114-0-productId => 258
2021-04-14 10:58:54.745 INFO ZWAVE: Node 65: value added: 134-0-libraryType => 6
2021-04-14 10:58:54.746 INFO ZWAVE: Node 65: value added: 134-0-protocolVersion => 3.52
2021-04-14 10:58:54.746 INFO ZWAVE: Node 65: value added: 134-0-firmwareVersions => 4.84
2021-04-14 10:58:54.767 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Cover status 3 => 3
2021-04-14 10:58:55.244 INFO ZWAVE: Node 65: value added: 128-0-level => 100
2021-04-14 10:58:55.245 INFO ZWAVE: Node 65: value added: 128-0-isLow => false
2021-04-14 10:58:55.323 INFO ZWAVE: Node 65: metadata updated: 132-0-wakeUpInterval
2021-04-14 10:58:55.364 INFO ZWAVE: Node 65: value added: 132-0-wakeUpInterval => 3600
2021-04-14 10:58:55.365 INFO ZWAVE: Node 65: value added: 132-0-controllerNodeId => 1
2021-04-14 10:58:55.459 INFO ZWAVE: Node 65: metadata updated: 48-0-Any
2021-04-14 10:58:55.460 INFO ZWAVE: Node 65: value added: 48-0-Any => true
2021-04-14 10:58:55.565 INFO ZWAVE: Node 65: value added: 113-0-Home Security-Sensor status => 2
2021-04-14 10:58:55.566 INFO ZWAVE: Node 65: interview stage COMMANDCLASSES completed
2021-04-14 10:58:55.567 INFO ZWAVE: Node 65: interview stage OVERWRITECONFIG completed
2021-04-14 10:58:55.582 INFO ZWAVE: Node 65: interview stage NEIGHBORS completed
2021-04-14 10:58:55.582 INFO ZWAVE: Node 65: interview stage COMPLETE completed
2021-04-14 10:58:55.588 INFO ZWAVE: Node 65: value added 65-48-0-Any => true
2021-04-14 10:58:55.588 INFO ZWAVE: Node 65: value added 65-113-0-Home Security-Cover status => 3
2021-04-14 10:58:55.589 INFO ZWAVE: Node 65: value added 65-113-0-Home Security-Sensor status => 2
2021-04-14 10:58:55.589 INFO ZWAVE: Node 65: value added 65-128-0-level => 100
2021-04-14 10:58:55.589 INFO ZWAVE: Node 65: value added 65-128-0-isLow => false
2021-04-14 10:58:55.593 INFO ZWAVE: Node 65: value added 65-114-0-manufacturerId => 265
2021-04-14 10:58:55.594 INFO ZWAVE: Node 65: value added 65-114-0-productType => 8193
2021-04-14 10:58:55.594 INFO ZWAVE: Node 65: value added 65-114-0-productId => 258
2021-04-14 10:58:55.594 INFO ZWAVE: Node 65: value added 65-134-0-libraryType => 6
2021-04-14 10:58:55.594 INFO ZWAVE: Node 65: value added 65-134-0-protocolVersion => 3.52
2021-04-14 10:58:55.594 INFO ZWAVE: Node 65: value added 65-134-0-firmwareVersions => 4.84
2021-04-14 10:58:55.595 INFO ZWAVE: Node 65: value added 65-132-0-wakeUpInterval => 3600
2021-04-14 10:58:55.595 INFO ZWAVE: Node 65: value added 65-132-0-controllerNodeId => 1
2021-04-14 10:58:55.595 INFO ZWAVE: Node 65 ready: Vision Security - ZD2102 (Door Window Sensor)
2021-04-14 10:58:55.601 INFO ZWAVE: Node 65: interview COMPLETED, all values are updated
2021-04-14 10:58:56.629 INFO ZWAVE: Node 65 is now asleep
2021-04-14 10:58:59.151 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Cover status 3 => 3
2021-04-14 10:59:00.301 INFO ZWAVE: Node 60: value updated: 50-0-value-66049 2.292 => 2.247
2021-04-14 10:59:01.350 INFO ZWAVE: Node 65 is now awake
2021-04-14 10:59:02.381 INFO ZWAVE: Node 65 is now asleep
2021-04-14 10:59:08.031 INFO ZWAVE: Node 65: value updated: 113-0-Home Security-Cover status 3 => 3

Dave-Casey, 2021-04-14:
@AlCalzone I am unaware of a method to re-interview using Zwave JS (without MQTT)?

blhoward2, 2021-04-14:
You can do it in the zwavejs2mqtt UI without using mqtt itself.

Photo64, 2021-04-14:
Interesting, since I re-interviewed the node earlier now both the binary_sensor and the basic sensor are stuck. binary_sensor continues to show "off" as it previously did but now the basic sensor is showing '255' when it should be '0'.

AlCalzone, 2021-04-14:
Let's do the change I mentioned above and see if that works.

closed 2021-04-14 by AlCalzone

Photo64, 2021-04-15:
> Let's do the change I mentioned above and see if that works.

Since this was merged and closed (with #2343) when would I see this update if I am using zwave-js2mqtt? Would it be with an update to zwave-js2mqtt addon or would it be in the next HA update in May?

AlCalzone, 2021-04-15:
You should see it with the next addon update **after** there's a new release of `node-zwave-js` and `zwavejs2mqtt` itself.

PR #2343 (2021-04-14): fix(config): enable Basic Set Mapping for ZP3102
fixes: #2338