zwave-js/zwave-js #2348: August Smart Lock gen 3 pairs and creates a Z-wave node but then does not discover any parameters or is modifiable.

liquidbear99, 2021-04-15:
I am happy to help in any way possible. When I pair the august smart lock 3 with z wave JS a node is created. It does not discover any parameters. I have tried factory resetting the lock and am only doing this as secure inclusion. It does tell me that the lock supports S2 but successfully paired and will be functional with S0. I know z wave JS does not support S2 yet but the august app says that it should still work on S0. Let me know how I can help diagnose the problem here.

AlCalzone, 2021-04-15:
You can start by not deleting the issue template and following it.

liquidbear99, 2021-04-15:
How can I access the template so I can follow it's format? I'm very new to GitHub and trying to learn.

AlCalzone, 2021-04-15:
You click <kbd>New issue</kbd> and then select **Bug report** from the list.
[image]

closed 2021-04-15 by liquidbear99