zwave-js/zwave-js #2351: Central scene events broken again?

Madelaide, 2021-04-15:
**Checklist:**

-   [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
-   [ ] I am **not** using ZWaveJS2MQTT. **Or:** a developer has told me to come here.
-   [ ] I have checked the troubleshooting section and my problem is **not** described there.
-   [ ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

**Device information**

Which device(s) is/are affected (make/model)?  
What are the node IDs?

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

Madelaide, 2021-04-15:
Scenes used to work after a brief few weeks of being broken. The old ticket is closed and I can’t write to it. 

When I press my wallmote buttons I see the action in the zw log but no trigger happens. The automation used to work. 

Did the fields change again?

ps: I’ll update ha details when my computer is available but using the latest public supervised version.

AlCalzone, 2021-04-15:
Without seeing a log that says otherwise, I'm gonna say "no":
[image]

AlCalzone, 2021-04-15:
Also, you forgot the most important parts of the checklist:
[image]

Since it works for me and it hasn't been confirmed that this issue is indeed caused by the driver and is not an integration/application problem, I'm gonna refer you to the HA repo and close this one.

closed 2021-04-15 by AlCalzone

Madelaide, 2021-04-16:
I didn't tick the first two because I am using both - why else would I be here?

the events show in the debug window
2021-04-16 15:46:55.615 INFO ZWAVE: Node 142: value notification: 91-0-scene-001 0
2021-04-16 15:47:01.586 INFO ZWAVE: Node 142: value notification: 91-0-scene-003 0

They don't show in HA>developer tools>events>zwave_js_event - nothing. So is this an HA issue or ZWJS issue?

If ha can't see anything, it points to zwjs not sending anything to HA.

Madelaide, 2021-04-16:
Never mind, I figured it out myself; the  zwave_js_event was changed to zwave_js_value_notification which broke all my automations ( and it doesn't match the event type listed in ha>developertools>events which lists the type as zwave_js_event.

AlCalzone, 2021-04-16:
> I didn't tick the first two because I am using both - why else would I be here?

I'm going to quote the issue template again - which you obviously skipped over and **didn't read** thoroughly. This is in place to avoid these kinds of issues that stem from user errors, or bugs in downstream applications, etc... Stuff that isn't caused by `node-zwave-js` and keeps me from investigating the issues that are.

---

  🚨🚨🚨 STOP! STOP! STOP! 🚨🚨🚨

  Before opening an issue, please read and follow these steps:

  1. Are you using HomeAssistant?
  👉🏻👉🏻👉🏻 **If yes, please open your issue at https://github.com/home-assistant/core/issues
  UNLESS a developer told you to come here.** 👈🏻👈🏻👈🏻

  2. Are you using ZWaveJS2MQTT?
  If yes, please open your issue at https://github.com/zwave-js/zwavejs2mqtt/issues
  UNLESS a developer told you to come here.

  3. Check the troubleshooting section if your problem is described there:
  https://zwave-js.github.io/node-zwave-js/#/development/troubleshooting

  4. Check the changelog if your problem was already fixed recently.
  https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md
  We cannot provide support if you are not using the latest version.

  🙏🏻🙏🏻🙏🏻 Thanks, now onto your issue:

Madelaide, 2021-04-16:
Of course i and probably every other zwave user appreciate your efforts this zwave library smokes the previous two attempts, but to be clear; "which you obviously skipped over and didn't read thoroughly" is close to libel.

I am a software QA Manager of a major automotive supplier, defining checklists and ensuring they are correct is something I do well.

The checklist I was presented with is as follows, not the version you claimed i did not read (which is actually a much better checklist)
"
Checklist:

  I am not using HomeAssistant. Or: a developer has told me to come here.
  I am not using ZWaveJS2MQTT. Or: a developer has told me to come here.
  I have checked the troubleshooting section and my problem is not described there.
  I have read the changelog and my problem was not mentioned there.
"

Not Clear, not helpful (no links or references) and definitely would fail an audit.

Now I found the problem myself, zwave_js_* event type name was changed, is that node-zwave-js or something else, who the hell can tell with no library prefixing of function names. At least the new checklist points reporters in the right direction.

And as far as auditing goes, the new checklist would be a good process improvement indicator.

cgarwood, 2021-04-16:
> Now I found the problem myself, zwave_js_* event type name was changed, is that node-zwave-js or something else, who the hell can tell with no library prefixing of function names. At least the new checklist points reporters in the right direction.

This was clearly called out as a breaking change in the Home Assistant release notes for 2021.4:
https://www.home-assistant.io/blog/2021/04/07/release-20214/#breaking-changes

AlCalzone, 2021-04-16:
@Madelaide

> The checklist I was presented with is as follows, not the version you claimed i did not read (which is actually a much better checklist)

These are the steps I go through when I open a bug report here:
1. Click <kbd>New issue</kbd>, see this:
[image]

2. Click <kbd>Get started</kbd> next to bug report, see this:
[image]

Are you saying that's not visible for you?

Madelaide, 2021-05-06:
It is visible now, it was not when  first posted this bug. As I mentioned, the version you quoted is much clearer, that is not the version I saw when I posted my issue.

Madelaide, 2021-05-06:
> This was clearly called out as a breaking change in the Home Assistant release notes for 2021.4:
> https://www.home-assistant.io/blog/2021/04/07/release-20214/#breaking-changes

...And so it is, I didn't see it, but I did try searching for that event type and nothing came up.

Is there a "single source of truth" for all release notes? new features, deprecated features where one can search.

AlCalzone, 2021-05-06:
> Is there a "single source of truth" for all release notes? new features, deprecated features where one can search.

You'll have to ask the HA devs about that. For the driver itself, there's a [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) - but what applications do with that and how it affects you as the end user may be different.